Koenkk/zigbee2mqtt #3993: zigbee2mqtt crashing. Could it be too many deviceS?

onlymejosh, 2020-07-27:
# Bug Report
I have tried to add 4 new bulbs to my network. `zigbee2mqtt` crashes and requires me to pull the usb and plug it back in again. Here are the related logs

```
winston:file: written true false
  mqttjs:client publish :: message `{"battery":100,"linkquality":60,"update_available":false,"occupancy":true,"temperature":19.5,"illuminance":24003,"illuminance_lux":250}` to topic `zigbee2mqtt/motion_sensor` +48s
  mqttjs:client publish :: qos 0 +0ms
  mqttjs:client MqttClient:publish: packet cmd: publish +0ms
  mqttjs:client _sendPacket :: (mqttjs_4ddf23d2) ::  start +1ms
  mqttjs:client sendPacket :: packet: {
  mqttjs:client   cmd: 'publish',
  mqttjs:client   topic: 'zigbee2mqtt/motion_sensor',
  mqttjs:client   payload: '{"battery":100,"linkquality":60,"update_available":false,"occupancy":true,"temperature":19.5,"illuminance":24003,"illuminance_lux":250}',
  mqttjs:client   qos: 0,
  mqttjs:client   retain: false,
  mqttjs:client   messageId: 26133,
  mqttjs:client   dup: false
  mqttjs:client } +0ms
  mqttjs:client sendPacket :: emitting `packetsend` +1ms
  mqttjs:client sendPacket :: writing to stream +0ms
  mqtt-packet:writeToStream generate called +48s
  mqtt-packet:writeToStream generate: packet.cmd: publish +0ms
  mqtt-packet:writeToStream publish: packet: { cmd: 'publish', topic: 'zigbee2mqtt/motion_sensor', payload: '{"battery":100,"linkquality":60,"update_available":false,"occupancy":true,"temperature":19.5,"illuminance":24003,"illuminance_lux":250}', qos: 0, retain: false, messageId: 26133, dup: false } +0ms
  mqtt-packet:writeToStream writeVarByteInt: writing to stream: <Buffer a2 01> +1ms
  mqtt-packet:writeToStream writeNumberCached: number: 25 +0ms
  mqtt-packet:writeToStream writeNumberCached: <Buffer 00 19> +0ms
  mqtt-packet:writeToStream publish: payload: '{"battery":100,"linkquality":60,"update_available":false,"occupancy":true,"temperature":19.5,"illuminance":24003,"illuminance_lux":250}' +1ms
  mqttjs:client sendPacket :: writeToStream result true +2ms
  mqttjs:client sendPacket :: invoking cb +0ms
  mqttjs:client _sendPacket :: (mqttjs_4ddf23d2) ::  end +0ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0x001788010862db89/2 1030(10, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +49s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":15072,"destendpoint":2,"srcendpoint":1,"clusterid":1030,"transid":10,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,120,11,10,0]}} +49s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,224,58,2,1,6,4,10,0,30,5,24,120,11,10,0,129] +49s
  serialport/stream _write 20 bytes of data +35ms
  serialport/binding-abstract write 20 bytes +121ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +35ms
  serialport/bindings/unixWrite Starting write 20 bytes offset 0 bytesToWrite 20 +49s
winston:file: logged 56543 debug 2020-07-27 20:28:57: Received Zigbee message from 'motion_sensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 2 with groupID 0

winston:file: logged 56765 info  2020-07-27 20:28:57: MQTT publish: topic 'zigbee2mqtt/motion_sensor', payload '{"battery":100,"linkquality":60,"update_available":false,"occupancy":true,"temperature":19.5,"illuminance":24003,"illuminance_lux":250}'

  serialport/stream _read reading { start: 1038, toRead: 64498 } +10ms
  serialport/binding-abstract read +10ms
  serialport/bindings/unixRead Starting read +45ms
  serialport/bindings/unixWrite write returned: wrote 20 bytes +9ms
  serialport/bindings/unixWrite Finished writing 20 bytes +0ms
  serialport/stream binding.write write finished +2ms
  serialport/bindings/unixRead Finished read 6 bytes +2ms
  serialport/stream binding.read finished { bytesRead: 6 } +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +49s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading { start: 1044, toRead: 64492 } +3ms
  serialport/binding-abstract read +5ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +53ms
  serialport/bindings/poller received "readable" +638ms
  serialport/bindings/unixRead Starting read +640ms
  serialport/bindings/unixRead Finished read 8 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 8 } +642ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,183,48,0,5] +643ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,183,48,0,5] +0ms
[... 186 lines cut ...]

## How to reproduce it (minimal and precise)
Try and interact with one of the new bulbs or when trying to add another one of the ` RS 229 T - Innr GU10` bulbs
```
  '0x588e81fffe21f46a'
  '0x588e81fffe233426'
  '0x588e81fffe232770'
```

Which are:
```
info  2020-07-27 21:07:11: 0x588e81fffe21f46a (0x588e81fffe21f46a): RS 229 T - Innr GU10 spot 350 lm, dimmable, white spectrum (Router)
info  2020-07-27 21:07:11: 0x588e81fffe233426 (0x588e81fffe233426): RS 229 T - Innr GU10 spot 350 lm, dimmable, white spectrum (Router)
info  2020-07-27 21:07:11: 0x588e81fffe232770 (0x588e81fffe232770): RS 229 T - Innr GU10 spot 350 lm, dimmable, white spectrum (Router)
```

## Debug Info
Zigbee2mqtt version:
```
"name": "zigbee2mqtt",
"version": "1.14.2",
```
Adapter hardware: CC2531
Adapter firmware version:
```info  2020-07-27 21:07:11: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'```

Koenkk, 2020-07-27:
The adapter crashes indeed, try switching to the source routing firmware which provides more stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

onlymejosh, 2020-07-28:
Does it correlate to the amount of devices I have? I'll switch the source out and try again. 

Is there anything else I can do? A different adapter etc?

Koenkk, 2020-07-28:
@onlymejosh a more powerful adapter will also solve it probably, e.g. https://www.zigbee2mqtt.io/information/supported_adapters.html#electrolama-zig-a-zig-ah-zzh

onlymejosh, 2020-07-28:
Great thanks -- Looks like the `Electrolama zig-a-zig-ah!` is out of stock at the moment but I have joined the waitlist. 

I had a look at the ConBee thread -- Although still in WIP what do you think of using that?

Koenkk, 2020-07-28:
It's still experimental, you can also check https://www.zigbee2mqtt.io/information/supported_adapters.html#slaeshs-cc2652rb-stick or https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-launchxl-cc26x2r1 (all same chip)

onlymejosh, 2020-07-28:
Thanks I did see all of those unfortunately they're all saying a while to deliver whereas the conbee is next day! I will hold out for the `Electrolama zig-a-zig-ah!`. 

For now if i re-flash the adapter with the firmware you mentioned, will I need to re-pair all of my devices?

Koenkk, 2020-07-29:
No you don't need to reflash: (https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices)

onlymejosh, 2020-07-29:
Thanks @Koenkk for all of your help. I flashed with the different firmware and it is working now. I'll keep an eye on both the conbee support and `Electrolama zig-a-zig-ah!`

closed 2020-07-29 by onlymejosh