Koenkk/zigbee2mqtt #3997: Salus SP600 OTA Update errors

derekoharrow, 2020-07-28:
I have a couple of Salus SP600 devices, which are recognised by z2m just fine.

But, I'm getting the following errors in the log file:

```
error 2020-07-28 11:17:53: Failed to call 'OTAUpdate' 'onZigbeeEvent' (TypeError: Cannot read property 'version' of undefined
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/salus.js:17:37)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/common.js:177:18)
    at async Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/common.js:168:23)
    at async OTAUpdate.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/otaUpdate.js:46:31)
    at async Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:360:21))
```

Not sure of the issue, but it looks like something to do with OTA update checks for the Salus device.

Version of Zigbee2Mqtt: 1.14.2
Coordinator version: 20190619

Koenkk, 2020-07-28:
Should be fixed, you will still see an error because there is no update available for your device but the message should be clear now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

hugalafutro, 2020-08-15:
I recreated my docker with latest-dev image, running 1.14.3-dev and I'm still seeing this error, although the wording changed. From the error it looks like it doesn't know the type of the plug (SP600), which it should, because it prints it out when the plug announces itself on the network. The error I'm getting now is:
`Zigbee2MQTT:error 2020-08-15 20:48:05: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for modelId 'undefined'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/ota/salus.js:15:5)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/ota/common.js:177:18)
    at async Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/ota/common.js:168:23)
    at async OTAUpdate.onZigbeeEvent (/app/lib/extension/otaUpdate.js:58:31)
    at async Controller.callExtensionMethod (/app/lib/controller.js:373:21))`

closed 2020-09-22 by github-actions[bot]

peter-dolkens, 2021-02-03:
I'm seeing the following when trying to run the update:

```
Info Updating 'Lounge Plug' to latest firmware
Info MQTT publish: topic 'home/zigbee2mqqt/bridge/log', payload '{"message":"Updating 'Lounge Plug' to latest firmware","meta":{"device":"Lounge Plug","status":"update_in_progress"},"type":"ota_update"}'
Info MQTT publish: topic 'home/zigbee2mqqt/Lounge Plug', payload '{"energy":0.61,"linkquality":81,"power":19.3,"state":"ON","update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'home/zigbee2mqqt/bridge/log', payload '{"message":"Update of 'Lounge Plug' failed (Cannot read property 'indexOf' of undefined)","meta":{"device":"Lounge Plug","status":"update_failed"},"type":"ota_update"}'
Info MQTT publish: topic 'home/zigbee2mqqt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Lounge Plug"},"error":"Update of 'Lounge Plug' failed (Cannot read property 'indexOf' of undefined)","status":"error","transaction":"w9nek-2"}'
Error Update of 'Lounge Plug' failed (Cannot read property 'indexOf' of undefined)
```

dr-harper, 2021-02-17:
I too can report the same issue as @peter-dolkens 

Running zigbee2mqtt v 1.17.1 via docker. Happy to help troubleshoot but not sure where to start looking.

mainsails, 2021-06-20:
Running 1.19.1 and report the same (post-stale comments) with the Salus SP600 :

```
Info Updating 'Office-Plug' to latest firmware
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Office-Plug' to latest firmware","meta":{"device":"Office-Plug","status":"update_in_progress"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/Office-Plug', payload '{"energy":0.01,"linkquality":165,"power":0.1,"state":"OFF","update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)","meta":{"device":"Office-Plug","status":"update_failed"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Office-Plug"},"error":"Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)","status":"error","transaction":"9e2dv-1"}'
Error Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)
```

lwfitzgerald, 2021-06-26:
I'm also seeing exactly the same issue on `1.19.1` when run against a new SP600.

I've previously updated these plugs with zigbee2mqtt so it looks like a bug has sneaked in here.

@Koenkk Would you mind re-opening this ticket? The github-actions bot doesn't seem to be re-opening it despite new comments.

Koenkk, 2021-06-26:
I've added additional debug logging, can you provide the debug log on the **dev** branch?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

mainsails, 2021-06-26:
Zigbee2MQTT version
1.19.1-dev commit: 519a6e9
```
debug 2021-06-26 19:08:38: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Office-Plug","transaction":"joqsb-2"}'
info  2021-06-26 19:08:38: Updating 'Office-Plug' to latest firmware
info  2021-06-26 19:08:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Office-Plug' to latest firmware","meta":{"device":"Office-Plug","status":"update_in_progress"},"type":"ota_update"}'
debug 2021-06-26 19:08:38: Received Zigbee message from 'Office-Plug', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20160331"}' from endpoint 9 with groupID 0
debug 2021-06-26 19:08:38: Updating to latest '0x001e5e09021619c0' (SP600)
debug 2021-06-26 19:08:38: Using endpoint '9'
debug 2021-06-26 19:08:39: Received Zigbee message from 'Office-Plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17387808,"imageType":38,"manufacturerCode":4216}' from endpoint 9 with groupID 0
debug 2021-06-26 19:08:39: Got OTA request '{"fieldControl":0,"manufacturerCode":4216,"imageType":38,"fileVersion":17387808}'
debug 2021-06-26 19:08:39: getNewImage for '0x001e5e09021619c0', meta {"fileVersion":33640736,"url":"https://eu.salusconnect.io/download/firmware/dd12520a-7100-4968-9483-d995a8d18d40/SP600_02015120.tar.gz"}
debug 2021-06-26 19:08:39: Update of 'Office-Plug' failed (TypeError: Cannot read property 'indexOf' of undefined)
info  2021-06-26 19:08:39: MQTT publish: topic 'zigbee2mqtt/Office-Plug', payload '{"energy":0.02,"linkquality":150,"power":0,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2021-06-26 19:08:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)","meta":{"device":"Office-Plug","status":"update_failed"},"type":"ota_update"}'
info  2021-06-26 19:08:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Office-Plug"},"error":"Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)","status":"error","transaction":"joqsb-2"}'
error 2021-06-26 19:08:39: Update of 'Office-Plug' failed (Cannot read property 'indexOf' of undefined)
debug 2021-06-26 19:08:41: Received Zigbee message from 'Office-Plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17387808,"imageType":38,"manufacturerCode":4216}' from endpoint 9 with groupID 0
debug 2021-06-26 19:08:44: Received Zigbee message from 'Office-Plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17387808,"imageType":38,"manufacturerCode":4216}' from endpoint 9 with groupID 0
debug 2021-06-26 19:08:47: Received Zigbee message from 'Office-Plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17387808,"imageType":38,"manufacturerCode":4216}' from endpoint 9 with groupID 0
debug 2021-06-26 19:08:50: Received Zigbee message from 'Office-Plug', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17387808,"imageType":38,"manufacturerCode":4216}' from endpoint 9 with groupID 0
```

lwfitzgerald, 2021-06-26:
Funny timing, I was also just trying this! I just patched the line into 1.19.1 though.

The log line you added wasn't that helpful, only showing the same message so I changed it to print the stack instead:
```
Update of 'sp600_plug' failed (TypeError: Cannot read property 'indexOf' of undefined at getNewImage (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:426:33) at processTicksAndRejections (internal/process/task_queues.js:95:5) at async Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:280:19) at async OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.js:219:21) at async Controller.callExtensionMethod (/app/lib/controller.js:386:21))
```

lwfitzgerald, 2021-06-26:
Raised a fix here - https://github.com/Koenkk/zigbee-herdsman-converters/pull/2728

Upgrading my firmware as we speak