Koenkk/zigbee2mqtt #3998: WXKG01LM device type specific configuration for long_release

SaturnusDJ, 2020-07-28:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L1096:

```
// After 4000 milliseconds of not reciving release we assume it will not happen.
```

This is about the `long_release` function of WXKG01LM.

After sniffing it, it looks like the long press function is not really a Zigbee message but code deciding if the button up message after a single press message is not received within a certain while, a `long` function will be triggered. But as can be read above, 4000 ms after the `long` it gives up waiting for the `long_release` (button up) message.

With the idea of connecting this device to a wall switch, the `long` function would turn the light on, and the `long_release` should turn the light off. However, after 4000 ms the button up message will not be translated to a `long_release` anymore.

Can this be turned into a device type specific configuration please?
So that the user can decide if the software should keep waiting for the `long_release`.

(I hope in the future my understanding of code will improve so that I can do pull requests for these kind of issues.)

Koenkk, 2020-07-28:
Implemented this (only for the `action` not `click` since that is considered legacy). 

Docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/WXKG01LM.md#device-type-specific-configuration

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-28 by Koenkk

SaturnusDJ, 2020-07-28:
Awesome! Will test once on master and let you know.

SaturnusDJ, 2020-09-18:
1.14.4

```
  '0x*':
    friendly_name: button
    hold_timeout: 0
    hold_timeout_expire: 720000
```
It does not work.

ciB89, 2020-11-20:
I was using the switch connected to a regular wall light switch to turn off my sonoff enabled lights. The long_release was triggered even after days. Now it won't trigger anymore after 400ms (or whatever else I set). Can't there be an option to disable the long_release_timeout option?

lsteelandt, 2021-05-09:
I am using a bi-stable switch "hardwared" connected to the **WXKG01LM** monostable switch (see photo).

Obviously using a bistable switch generate only "Hold" states. Hold in press state and hold in release state. 

When switching to the 'press' state (Switch ON), I want to avoid the 1000ms delay before receiving the "Hold" action.  
Trying to setup **hold_timeout** to 20 (20ms), but it doesn't work. 
As if the 1000ms delay is a minimum. 

Setting the hold_timeout_expire to 5000000 (almost 2 months), it works. I receive the 'release' action even I put the switch  to 'release' state after 10 minutes (I haven't verify after 1 month 😉 )

Is there a minimum of 1000ms considered on the delay to trigger an "hold" action ? 
If yes, is it possible to allow value like 20ms ?

Thanks for this awful work. 

[image]

SaturnusDJ, 2021-05-26:
@lsteelandt
Did you try with `legacy: false` for the device?
I remember something that this first gen Xiaomi button was always slow. I replaced it with the door/window switch (desolder the reed sensor, and solder the wires on those points) for near instant switching.