Koenkk/zigbee2mqtt #3999: Interview - node descriptor request failed for '0x000d6f0012e52153'

646413046, 2020-07-28:
device.js
	{
        zigbeeModel: ['041580'],
        model: 'IHW1215(AL)',
        description: 'Honyar curtain motor',
        supports: 'open, close',
        vendor: 'Honyar',
        fromZigbee: [],
        toZigbee: [],
    },

issue:
Interview failed

files:
debug.log
database.db

please help me check it
Thanks

646413046, 2020-07-28:
[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/4989200/debug.log)
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4989212/database.zip)

646413046, 2020-07-29:
new log:
> zigbee2mqtt@1.14.1 start /media/data/ubuntu/application/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-28 18:50:25: Logging to console and directory: '/media/data/ubuntu/application/zigbee2mqtt/data/log/2020-07-28.18-50-25' filename: log.txt
zigbee2mqtt:debug 2020-07-28 18:50:26: Can't load state from file /media/data/ubuntu/application/zigbee2mqtt/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-07-28 18:50:26: Starting zigbee2mqtt version 1.14.1 (commit #unknown)
zigbee2mqtt:info  2020-07-28 18:50:26: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-07-28 18:50:26: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/media/data/ubuntu/application/zigbee2mqtt/data/database.db","databaseBackupPath":"/media/data/ubuntu/application/zigbee2mqtt/data/database.db.backup","backupPath":"/media/data/ubuntu/application/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/media/data/ubuntu/application/zigbee2mqtt/data/database.db","databaseBackupPath":"/media/data/ubuntu/application/zigbee2mqtt/data/database.db.backup","backupPath":"/media/data/ubuntu/application/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +16ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +2ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +30ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +15ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +17ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +14ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}} +13ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8,0,0,99] +13ms
[... cut ...]
zigbee2mqtt:debug 2020-07-28 18:51:30: Received Zigbee message from '0x000d6f0012e52153', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-07-28 18:51:30: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-07-28 18:51:30: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +21ms

Koenkk, 2020-07-29:
Seems the device fails to respond to requests, try pairing the device close to the coordinator and keep it awake by pressing a button on it.

closed 2020-09-05 by github-actions[bot]