Koenkk/zigbee2mqtt #4000: unknow device but similar device in device.js

VladDrakul1986, 2020-07-28:
# Unknown Device
## What did you expect to happen
i try to add an new device a Paul Neumann Lamp and get following message:
```
info  2020-07-28 17:44:32: Device '0x00158d00044d879c' joined
info  2020-07-28 17:44:32: Starting interview of '0x00158d00044d879c'
info  2020-07-28 17:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00044d879c"}}'
info  2020-07-28 17:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00044d879c"}}'
info  2020-07-28 17:44:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00044d879c"}}'
info  2020-07-28 17:44:33: Successfully interviewed '0x00158d00044d879c', device has successfully been paired
warn  2020-07-28 17:44:33: Device '0x00158d00044d879c' with Zigbee model 'NLG-RGBW light' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
debug 2020-07-28 18:03:06: Received Zigbee message from '0x00158d00044d879c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4624,"imageType":4365,"fileVersion":1}' from endpoint 1 with groupID 0
```
according to this i check  the how to side and check according to the introduction the device.js and found
```
    {
        zigbeeModel: ['NLG-RGBW light '],
        model: '100.110.39',
        vendor: 'Paul Neuhaus',
        description: 'Q-FLAG LED Panel, Smart-Home RGBW',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        endpoint: (device) => {
            return {'default': 2};
        },
    },
```
Seems to be okay for my device but the model number is different my one 828303 and it is not a panel but a floor lamp.

## What did you expect to happen
That the lamp get recognized. Since there are tons of Paul Neuman lamps iam not sure how to make the solution for more people.

## How to reproduce it (minimal and precise)
pair the lamp
## Debug Info
Zigbee2mqtt version: 1.14.2 (commit #faaf3e4)
Adapter hardware: CC2531

Koenkk, 2020-07-28:
I don't understand why you get an unsupported error when the device is already supported as you mentioned?

MPM1107, 2020-07-28:
@Koenkk Looks like the definition in the converter has a space at the end of the string, and the device reported as unsupported doesn't. Seems kinda odd but that's probably the reason?

Koenkk, 2020-07-29:
@MPM1107 yes I think so, probably this device has been added in zigbee-shepherd times were string handling was not completely correct (leading to the spaces).

@VladDrakul1986 can you provide an url/link to this device?

VladDrakul1986, 2020-07-30:
Hi was kind of tricky to find an english site
https://www.illumination.co.uk/paul-neuhaus-q-riller-floor-lamp-843-17-k.html
this is the shop of the manufacturer
https://www.paul-neuhaus.de/shop/de/led-stehleuchte-smart-home-alexa-tauglich-100-111-57.html
there are tons of this lamp which are using the same controll.
Some have RGB like this one some doesn't have, thank you for your effort

Koenkk, 2020-07-30:
Device should be supported in latest dev branch now.
I've made the models/descriptions generic since we cannot separately identify them, please test & close the issue.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

VladDrakul1986, 2020-07-31:
the device gets recognized now
```
info  2020-07-31 20:38:16: Device 'WoziLampe' is supported, identified as: Paul Neuhaus Various RGBW lights (e.g. 100.110.39, 100.111.57) (NLG-RGBW light)
info  2020-07-31 20:38:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"WoziLampe","model":"NLG-RGBW light","vendor":"Paul Neuhaus","description":"Various RGBW lights (e.g. 100.110.39, 100.111.57)","supported":true}}'
debug 2020-07-31 20:38:28: Saving state to file /app/data/state.json
```
but when i try to set the light on or off
```
error 2020-07-31 20:51:30: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:173:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:376:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:268:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:409:12)
    at work (/app/node_modules/mqtt/lib/client.js:320:12)
    at Writable.writable._write (/app/node_modules/mqtt/lib/client.js:334:5))
```

and i see 2 properties in HomeAssistant switch and lqi but the rgb and brightness control is this related to your changes? If not this point is closed and i have to create a new one for this topic

Koenkk, 2020-08-01:
Can you provide some more debug logging of that exception?

VladDrakul1986, 2020-08-01:
Hi how i can do this? i already had set the logging level to debug.
No i add the herdsman Debug stuff into my docker command.
```
zigbee2mqtt_1  | Zigbee2MQTT:debug 2020-08-01 12:03:01: Received MQTT message on 'zigbee2mqtt/WoziLampe/set' with data '{"state": "ON"}'
zigbee2mqtt_1  | Zigbee2MQTT:error 2020-08-01 12:03:01: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined
zigbee2mqtt_1  |     at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:173:52)
zigbee2mqtt_1  |     at Controller.callExtensionMethod (/app/lib/controller.js:376:44)
zigbee2mqtt_1  |     at Controller.onMQTTMessage (/app/lib/controller.js:268:14)
zigbee2mqtt_1  |     at MQTT.emit (events.js:315:20)
zigbee2mqtt_1  |     at MQTT.onMessage (/app/lib/mqtt.js:95:14)
zigbee2mqtt_1  |     at MqttClient.emit (events.js:315:20)
zigbee2mqtt_1  |     at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1269:12)
zigbee2mqtt_1  |     at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:409:12)
zigbee2mqtt_1  |     at work (/app/node_modules/mqtt/lib/client.js:320:12)
zigbee2mqtt_1  |     at Writable.writable._write (/app/node_modules/mqtt/lib/client.js:334:5))
zigbee2mqtt_1  | 2020-08-01T10:03:44.198Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,101,177,0,86]
zigbee2mqtt_1  | 2020-08-01T10:03:44.198Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,101,177,0,86]
zigbee2mqtt_1  | 2020-08-01T10:03:44.198Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [101,177,0] - 86
zigbee2mqtt_1  | 2020-08-01T10:03:44.198Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":45413,"relaycount":0,"relaylist":[]}
zigbee2mqtt_1  | 2020-08-01T10:03:44.198Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-08-01T10:03:44.218Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,101,177,1,13,0,110,0,40,54,230,0,0,12,1,1,1,0,16,18,13,17,1,0,0,0,101,177,29,105]
zigbee2mqtt_1  | 2020-08-01T10:03:44.218Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,101,177,1,13,0,110,0,40,54,230,0,0,12,1,1,1,0,16,18,13,17,1,0,0,0,101,177,29,105]
zigbee2mqtt_1  | 2020-08-01T10:03:44.218Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,101,177,1,13,0,110,0,40,54,230,0,0,12,1,1,1,0,16,18,13,17,1,0,0,0,101,177,29] - 105
zigbee2mqtt_1  | 2020-08-01T10:03:44.219Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":45413,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":110,"securityuse":0,"timestamp":15087144,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,1,1,0,16,18,13,17,1,0,0,0]}}
zigbee2mqtt_1  | 2020-08-01T10:03:44.220Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4624,"imageType":4365,"fileVersion":1}},"address":45413,"endpoint":1,"linkquality":110,"groupID":0}'
zigbee2mqtt_1  | Zigbee2MQTT:debug 2020-08-01 12:03:44: Received Zigbee message from 'WoziLampe', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4624,"imageType":4365,"fileVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt_1  | 2020-08-01T10:03:44.222Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-08-01T10:03:44.223Z zigbee-herdsman:controller:endpoint CommandResponse 0x00158d00044d879c/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
zigbee2mqtt_1  | 2020-08-01T10:03:44.223Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":45413,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":5,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,5,2,152]}}
zigbee2mqtt_1  | 2020-08-01T10:03:44.224Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,101,177,1,1,25,0,5,0,30,4,25,5,2,152,127]
zigbee2mqtt_1  | 2020-08-01T10:03:44.235Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt_1  | 2020-08-01T10:03:44.235Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt_1  | 2020-08-01T10:03:44.235Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt_1  | 2020-08-01T10:03:44.235Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt_1  | 2020-08-01T10:03:44.235Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-08-01T10:03:44.242Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,5,195]
zigbee2mqtt_1  | 2020-08-01T10:03:44.242Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,5,195]
zigbee2mqtt_1  | 2020-08-01T10:03:44.242Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195
zigbee2mqtt_1  | 2020-08-01T10:03:44.242Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5}
[... cut ...] :0,"len":8,"data":{"type":"Buffer","data":[24,66,10,0,0,41,181,8]}}
zigbee2mqtt_1  | 2020-08-01T10:06:10.342Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":66,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2229}]},"address":52394,"endpoint":1,"linkquality":84,"groupID":0}
```

Koenkk, 2020-08-01:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

VladDrakul1986, 2020-08-02:
Thank you very much, it works now.
Via HomeAssistant i just can set only on and off but with MQTT i can control all properties as expected. So the rest seems to be a other problem

closed 2020-08-02 by VladDrakul1986