Koenkk/zigbee2mqtt #4001: Double action from Tuya switch

elfhack, 2020-07-28:
# Bug Report
I bought myself a couple of [these Lonsonho Tuya Switches](https://pl.aliexpress.com/item/4001095272467.html?spm=a2g0s.9042311.0.0.783a5c0fL7NFxN). They pair almost fine - zigbee2mqtt complains about not having a converter for the power state. However, what's more damning is that each click whichever it might be - single, double, hold, is being repeated twice.

## What did you expect to happen
Clicks are reported only once

## How to reproduce it (minimal and precise)
Pair device, click. 

I'm attaching debug + debug herdsman logs

```
2020-07-28T16:44:32.629Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,1,0,65,72,1,1,0,23,0,21,182,231,0,0,7,8,7,10,33,0,32,200,65,72,29,90]
2020-07-28T16:44:32.629Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,0,65,72,1,1,0,23,0,21,182,231,0,0,7,8,7,10,33,0,32,200,65,72,29,90]
2020-07-28T16:44:32.630Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,65,72,1,1,0,23,0,21,182,231,0,0,7,8,7,10,33,0,32,200,65,72,29] - 90
2020-07-28T16:44:32.630Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":23,"securityuse":0,"timestamp":15185429,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,7,10,33,0,32,200]}}
2020-07-28T16:44:32.631Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":200}]},"address":18497,"endpoint":1,"linkquality":23,"groupID":0}'
zigbee2mqtt:debug 2020-07-28 18:44:32: Received Zigbee message from 'TuyaSingle', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-28 18:44:32: No converter available for 'S9TSZGB_1' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":200}'
2020-07-28T16:44:32.632Z zigbee-herdsman:controller:endpoint DefaultResponse 0xec1bbdfffea528ff/1 1(10, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-07-28T16:44:32.633Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18497,"destendpoint":1,"srcendpoint":1,"clusterid":1,"transid":6,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,7,11,10,0]}}
2020-07-28T16:44:32.633Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,65,72,1,1,1,0,6,0,30,5,24,7,11,10,0,33]
2020-07-28T16:44:32.633Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-28T16:44:32.641Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-07-28T16:44:32.641Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-07-28T16:44:32.642Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-07-28T16:44:32.642Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-07-28T16:44:32.642Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-28T16:44:32.729Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,6,192]
2020-07-28T16:44:32.729Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,6,192]
2020-07-28T16:44:32.729Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,6] - 192
2020-07-28T16:44:32.729Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":6}
2020-07-28T16:44:32.729Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-28T16:44:32.934Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,65,72,1,1,0,23,0,208,185,231,0,0,4,1,8,253,0,65,72,29,175]
2020-07-28T16:44:32.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,65,72,1,1,0,23,0,208,185,231,0,0,4,1,8,253,0,65,72,29,175]
2020-07-28T16:44:32.935Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,65,72,1,1,0,23,0,208,185,231,0,0,4,1,8,253,0,65,72,29] - 175
2020-07-28T16:44:32.935Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":23,"securityuse":0,"timestamp":15186384,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,8,253,0]}}
2020-07-28T16:44:32.938Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,8,253,0]},"address":18497,"endpoint":1,"linkquality":23,"groupID":0}'
zigbee2mqtt:debug 2020-07-28 18:44:32: Received Zigbee message from 'TuyaSingle', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,8,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:44:32: MQTT publish: topic 'zigbee2mqtt/TuyaSingle', payload '{"linkquality":23,"action":"single"}'
2020-07-28T16:44:32.940Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt:info  2020-07-28 18:44:32: MQTT publish: topic 'zigbee2mqtt/TuyaSingle', payload '{"linkquality":23,"action":""}'
zigbee2mqtt:info  2020-07-28 18:44:32: MQTT publish: topic 'zigbee2mqtt/TuyaSingle/action', payload 'single'
2020-07-28T16:44:33.943Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,65,72,1,1,0,28,0,31,198,231,0,0,4,1,8,253,0,65,72,29,20]
2020-07-28T16:44:33.943Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,65,72,1,1,0,28,0,31,198,231,0,0,4,1,8,253,0,65,72,29,20]
2020-07-28T16:44:33.943Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,65,72,1,1,0,28,0,31,198,231,0,0,4,1,8,253,0,65,72,29] - 20
2020-07-28T16:44:33.943Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":28,"securityuse":0,"timestamp":15189535,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,8,253,0]}}
2020-07-28T16:44:33.944Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,8,253,0]},"address":18497,"endpoint":1,"linkquality":28,"groupID":0}'
zigbee2mqtt:debug 2020-07-28 18:44:33: Received Zigbee message from 'TuyaSingle', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,8,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:44:33: MQTT publish: topic 'zigbee2mqtt/TuyaSingle', payload '{"linkquality":28,"action":"single"}'
2020-07-28T16:44:33.945Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt:info  2020-07-28 18:44:33: MQTT publish: topic 'zigbee2mqtt/TuyaSingle', payload '{"linkquality":28,"action":""}'
zigbee2mqtt:info  2020-07-28 18:44:33: MQTT publish: topic 'zigbee2mqtt/TuyaSingle/action', payload 'single'
```

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-07-28:
Can you change the converters in `fromZigbee.js` to:

```js
ts0043_click: {
    cluster: 'genOnOff',
    type: 'raw',
    convert: (model, msg, publish, options, meta) => {
        if (hasAlreadyReceivedTransaction(msg)) return;
        const buttonMapping = {1: 'right', 2: 'middle', 3: 'left'};
        const clickMapping = {0: 'single', 1: 'double', 2: 'hold'};
        return {action: `${buttonMapping[msg.endpoint.ID]}_${clickMapping[msg.data[3]]}`};
    },
},
ts0042_click: {
    cluster: 'genOnOff',
    type: 'raw',
    convert: (model, msg, publish, options, meta) => {
        if (hasAlreadyReceivedTransaction(msg)) return;
        const buttonMapping = {1: 'left', 2: 'right'};
        const clickMapping = {0: 'single', 1: 'double', 2: 'hold'};
        return {action: `${buttonMapping[msg.endpoint.ID]}_${clickMapping[msg.data[3]]}`};
    },
},
ts0041_click: {
    cluster: 'genOnOff',
    type: 'raw',
    convert: (model, msg, publish, options, meta) => {
        if (hasAlreadyReceivedTransaction(msg)) return;
        const clickMapping = {0: 'single', 1: 'double', 2: 'hold'};
        return {action: `${clickMapping[msg.data[3]]}`};
    },
},
```

I hope that this fixes it, as otherwise I don't know how to identify the duplicates else, I'm a bit worried that the transaction sequence number always stays 0 and therefore only one `action` is provided and after that none.

elfhack, 2020-07-28:
Should I be using -dev for this? 
```
zigbee2mqtt:error 2020-07-28 21:35:02: Failed to call 'Receive' 'onZigbeeEvent' (ReferenceError: hasAlreadyReceivedTransaction is not defined
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:2982:13)
    at /app/lib/extension/receive.js:168:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/app/lib/controller.js:360:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:246:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:67:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:503:22))
2020-07-28T19:35:02.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2020-07-29:
Yes use dev (and also the `fromZigbee.js` that comes with it), then apply changes from https://github.com/Koenkk/zigbee2mqtt/issues/4001#issuecomment-665215701

elfhack, 2020-07-29:
No good, now no action gets published to mqtt at all :( 

```
2020-07-29T10:13:10.567Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,1,0,65,72,1,1,0,94,0,27,242,171,0,0,7,8,47,10,33,0,32,200,65,72,29,61]
2020-07-29T10:13:10.567Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,0,65,72,1,1,0,94,0,27,242,171,0,0,7,8,47,10,33,0,32,200,65,72,29,61]
2020-07-29T10:13:10.567Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,65,72,1,1,0,94,0,27,242,171,0,0,7,8,47,10,33,0,32,200,65,72,29] - 61
2020-07-29T10:13:10.568Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":11268635,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,47,10,33,0,32,200]}}
2020-07-29T10:13:10.569Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":47,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":200}]},"address":18497,"endpoint":1,"linkquality":94,"groupID":0}'
zigbee2mqtt:debug 2020-07-29 12:13:10: Received Zigbee message from 'SwitchKitchen', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 12:13:10: No converter available for 'S9TSZGB_1' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":200}'
2020-07-29T10:13:10.571Z zigbee-herdsman:controller:endpoint DefaultResponse 0xec1bbdfffea528ff/1 1(10, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-07-29T10:13:10.572Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18497,"destendpoint":1,"srcendpoint":1,"clusterid":1,"transid":41,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,47,11,10,0]}}
2020-07-29T10:13:10.572Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,65,72,1,1,1,0,41,0,30,5,24,47,11,10,0,38]
2020-07-29T10:13:10.572Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-29T10:13:10.580Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-07-29T10:13:10.580Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-07-29T10:13:10.580Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-07-29T10:13:10.580Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-07-29T10:13:10.581Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-29T10:13:10.668Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,41,239]
2020-07-29T10:13:10.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,41,239]
2020-07-29T10:13:10.668Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,41] - 239
2020-07-29T10:13:10.668Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":41}
2020-07-29T10:13:10.669Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-29T10:13:10.872Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,65,72,1,1,0,94,0,216,245,171,0,0,4,1,48,253,0,65,72,29,214]
2020-07-29T10:13:10.872Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,65,72,1,1,0,94,0,216,245,171,0,0,4,1,48,253,0,65,72,29,214]
2020-07-29T10:13:10.872Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,65,72,1,1,0,94,0,216,245,171,0,0,4,1,48,253,0,65,72,29] - 214
2020-07-29T10:13:10.872Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":11269592,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,48,253,0]}}
2020-07-29T10:13:10.874Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,48,253,0]},"address":18497,"endpoint":1,"linkquality":94,"groupID":0}'
zigbee2mqtt:debug 2020-07-29 12:13:10: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,48,253,0]}' from endpoint 1 with groupID 0
2020-07-29T10:13:10.875Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-29T10:13:11.880Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,65,72,1,1,0,94,0,41,2,172,0,0,4,1,48,253,0,65,72,29,215]
2020-07-29T10:13:11.881Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,65,72,1,1,0,94,0,41,2,172,0,0,4,1,48,253,0,65,72,29,215]
2020-07-29T10:13:11.881Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,65,72,1,1,0,94,0,41,2,172,0,0,4,1,48,253,0,65,72,29] - 215
2020-07-29T10:13:11.881Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":18497,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":11272745,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,48,253,0]}}
2020-07-29T10:13:11.881Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,48,253,0]},"address":18497,"endpoint":1,"linkquality":94,"groupID":0}'
zigbee2mqtt:debug 2020-07-29 12:13:11: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,48,253,0]}' from endpoint 1 with groupID 0
2020-07-29T10:13:11.882Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2020-07-29:
Can you provide the zigbee2mqtt debug logging (not herdsman debug logging) of the following sequence:
- Trigger action
- wait 5 seconds
- repeat this 10 times.

elfhack, 2020-07-29:
With or without the aforementioned fromZibee.js patch?

Koenkk, 2020-07-29:
Doesnt matter

elfhack, 2020-07-29:
Ok, as best as I could
```
zigbee2mqtt:debug 2020-07-29 13:26:19: Received Zigbee message from 'SwitchKitchen', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:19: No converter available for 'S9TSZGB_1' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":200}'
zigbee2mqtt:debug 2020-07-29 13:26:19: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,4,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:20: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,4,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:25: Received Zigbee message from 'SwitchKitchen', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:25: No converter available for 'S9TSZGB_1' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":200}'
zigbee2mqtt:debug 2020-07-29 13:26:25: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,6,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:26: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,6,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:26:29: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'raw'
zigbee2mqtt:info  2020-07-29 13:26:29: Starting network scan (includeRoutes 'false')
zigbee2mqtt:debug 2020-07-29 13:26:29: LQI succeeded for 'Coordinator'
zigbee2mqtt:debug 2020-07-29 13:26:30: LQI succeeded for 'Kitchentop'
zigbee2mqtt:info  2020-07-29 13:26:30: Network scan finished
zigbee2mqtt:info  2020-07-29 13:26:30: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00124b0018df5b1a","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null},{"ieeeAddr":"0x000b57fffe368edd","friendlyName":"Bedroom_Remote","type":"EndDevice","networkAddress":35886,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI remote control","lastSeen":1596017960455},{"ieeeAddr":"0x000b57fffe39eb73","friendlyName":"Couch_Remote","type":"EndDevice","networkAddress":22624,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI remote control","lastSeen":1596021199983},{"ieeeAddr":"0x000b57fffe396706","friendlyName":"Kitchen_Remote","type":"EndDevice","networkAddress":34924,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI remote control","lastSeen":1596021876471},{"ieeeAddr":"0x00124b001e728db4","friendlyName":"Kitchentop","type":"Router","networkAddress":3778,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1596021899527},{"ieeeAddr":"0xec1bbdfffea52627","friendlyName":"SwitchLivingRoom","type":"EndDevice","networkAddress":63294,"manufacturerName":"_TZ3000_oikiyf3b","modelID":"TS0042","lastSeen":1596011151152},{"ieeeAddr":"0xec1bbdfffea528ff","friendlyName":"SwitchKitchen","type":"EndDevice","networkAddress":18497,"manufacturerName":"_TZ3000_5cz0qu5f","modelID":"TS0041","lastSeen":1596021986392}],"links":[{"source":{"ieeeAddr":"0x00124b001e728db4","networkAddress":3778},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":21,"depth":1,"routes":[],"sourceIeeeAddr":"0x00124b001e728db4","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":3778,"lqi":21,"relationship":1},{"source":{"ieeeAddr":"0x000b57fffe368edd","networkAddress":35886},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":32,"depth":1,"routes":[],"sourceIeeeAddr":"0x000b57fffe368edd","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":35886,"lqi":32,"relationship":1},{"source":{"ieeeAddr":"0x000b57fffe39eb73","networkAddress":22624},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":62,"depth":1,"routes":[],"sourceIeeeAddr":"0x000b57fffe39eb73","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":22624,"lqi":62,"relationship":1},{"source":{"ieeeAddr":"0x000b57fffe396706","networkAddress":34924},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":67,"depth":1,"routes":[],"sourceIeeeAddr":"0x000b57fffe396706","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":34924,"lqi":67,"relationship":1},{"source":{"ieeeAddr":"0xec1bbdfffea52627","networkAddress":63294},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":92,"depth":1,"routes":[],"sourceIeeeAddr":"0xec1bbdfffea52627","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":63294,"lqi":92,"relationship":1},{"source":{"ieeeAddr":"0xec1bbdfffea528ff","networkAddress":18497},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":91,"depth":1,"routes":[],"sourceIeeeAddr":"0xec1bbdfffea528ff","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":18497,"lqi":91,"relationship":1},{"source":{"ieeeAddr":"0x0000000000000000","networkAddress":60258},"target":{"ieeeAddr":"0x00124b0018df5b1a","networkAddress":0},"linkquality":0,"depth":255,"routes":[],"sourceIeeeAddr":"0x0000000000000000","targetIeeeAddr":"0x00124b0018df5b1a","sourceNwkAddr":60258,"lqi":0,"relationship":3},{"source":{"ieeeAddr":"0x00124b0018df5b1a","networkAddres [... cut ...]
zigbee2mqtt:debug 2020-07-29 13:27:30: No converter available for 'S9TSZGB_1' with cluster 'genPowerCfg' and type 'attributeReport' and data '{"batteryPercentageRemaining":200}'
zigbee2mqtt:debug 2020-07-29 13:27:30: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,28,253,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-29 13:27:31: Received Zigbee message from 'SwitchKitchen', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,28,253,0]}' from endpoint 1 with groupID 0
```

Koenkk, 2020-07-29:
Thanks, found something, should be fixed now & also marked them as TuYa. 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

elfhack, 2020-07-30:
@Koenkk just tried the freshest latest-dev image, and indeed, clicks now work perfectly. What would I need to add to support battery reporting as well? (The genPowerCfg cluster)

Koenkk, 2020-07-30:
Done!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

elfhack, 2020-07-30:
Thank you for a job well done :) I'll test it out as soon as I get home (My daughter is about to be born!)

Koenkk, 2020-07-30:
(Almost) congratulations with your daughter!

The next z2m release will be on 12 August. 

I assume this can be closed now, but feel free to re-open if it doesn't work as expected.

closed 2020-07-30 by Koenkk

elfhack, 2020-08-03:
Thanks, and no need. It works perfectly now :)