Koenkk/zigbee2mqtt #4003: Error message in Z2M

Westcott1, 2020-07-28:
## Debug Info
Zigbee2mqtt version: 1.14.2 DEV
Adapter hardware: CC2530
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20200720}}'

This error message appears occasionally -

zigbee2mqtt:error 2020-07-28 19:00:42: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:384:32)
    at /opt/zigbee2mqtt/lib/extension/receive.js:168:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:348:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:234:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.js:67:52)
    at Controller.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:503:22)

Koenkk, 2020-07-28:
Can you provide more debug logging of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Westcott1, 2020-07-28:
OK will do.
I just have to uncomment it.

Westcott1, 2020-07-28:
zigbee2mqtt:info  2020-07-28 20:51:10: MQTT publish: topic 'zigbee2mqtt/Attic', payload '{"temperature":25.7,"linkquality":0,"humidity":37.2,"battery":100,"voltage":3000}'
zigbee2mqtt:debug 2020-07-28 20:51:10: Received Zigbee message from 'Attic', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":29}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-07-28 20:51:10: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:384:32)
    at /opt/zigbee2mqtt/lib/extension/receive.js:168:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:348:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:234:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.js:67:52)
    at Controller.emit (events.js:311:20)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:503:22))
zigbee2mqtt:debug 2020-07-28 20:51:18: Received Zigbee message from 'Attic', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-28 20:51:18: No converter available for 'TS0201' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":19,"appVersion":255}'
zigbee2mqtt:debug 2020-07-28 20:51:18: Received Zigbee message from 'Attic', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":65}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-28 20:51:18: No converter available for 'TS0201' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":19,"appVersion":65}'

Westcott1, 2020-07-28:
Attic (0xec1bbdfffe69d387): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)

Koenkk, 2020-07-29:
Thanks, fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-29 by Koenkk