Koenkk/zigbee2mqtt #4005: Aqara double switches with neutral error in logs

quarcko, 2020-07-29:
Hello,

Latest version of Z2M is used in HASSIO.

Again Aqara double wall switches (with neutral) and router function those,
are giving some warnings in logs of zigbee2mqtt. Previously i started a bug report: 
https://github.com/Koenkk/zigbee2mqtt/issues/3403

Which apparently got solved, but there is more:
zigbee2mqtt:info  2020-07-29 16:59:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f9a097', payload '{"power":0,"consumption":12.15,"temperature":29,"linkquality":84,"state_left":"OFF","state_right":"OFF","last_seen":"2020-07-29T16:59:42+03:00","click":""}'
(node:235) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'toString' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.14.2/lib/extension/homeassistant.js:1891:45)
    at EventBus.<anonymous> (/zigbee2mqtt-1.14.2/lib/extension/homeassistant.js:1814:63)
    at EventBus.emit (events.js:327:22)
    at EventBus.emit (/zigbee2mqtt-1.14.2/lib/eventBus.js:16:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.14.2/lib/controller.js:330:23)
(node:235) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5037)

I think it is again related to "click" data. in my case ALL double wall switches are sending this data as empty string.
and something fails, there is no check for empty string?

Koenkk, 2020-07-29:
Can you provide more debug logging of this issue?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

quarcko, 2020-07-29:
Here you go:

zigbee2mqtt:debug 2020-07-29 21:32:19: Received Zigbee message from '0x00158d0001bf9c83', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-29 21:32:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001bf9c83', payload '{"power":22.04,"consumption":35.97,"temperature":27,"linkquality":84,"state_left":"OFF","state_right":"OFF","last_seen":"2020-07-29T21:32:19+03:00"}'
zigbee2mqtt:info  2020-07-29 21:32:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001bf9c83', payload '{"power":22.04,"consumption":35.97,"temperature":27,"linkquality":84,"state_left":"OFF","state_right":"OFF","last_seen":"2020-07-29T21:32:19+03:00","click":""}'
(node:241) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'toString' of undefined
    at HomeAssistant.onPublishEntityState (/zigbee2mqtt-1.14.2/lib/extension/homeassistant.js:1891:45)
    at EventBus.<anonymous> (/zigbee2mqtt-1.14.2/lib/extension/homeassistant.js:1814:63)
    at EventBus.emit (events.js:327:22)
    at EventBus.emit (/zigbee2mqtt-1.14.2/lib/eventBus.js:16:15)
    at Controller.publishEntityState (/zigbee2mqtt-1.14.2/lib/controller.js:330:23)
(node:241) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

Koenkk, 2020-07-29:
Thanks, fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-29 by Koenkk

quarcko, 2020-07-29:
Thank you very much Koenkk!
I will wait for official release, as this is just mere warning that i wanted to report and be fixed,
really it does not break anything for me.

But still much appreciate great support!