Koenkk/zigbee2mqtt #4009: Zigbee2mqtt discover devices, but not integrate in HA

wilsonlspacheco, 2020-07-29:
# Bug Report
## What happened

I have HA installed in RasPi4 (Hassio / docker) with a CC2531 flashed with Z-Stack, which detects Zigbee devices, but does not integrate these devices in HA, either as devices or as entities. They also don't appear in Dev Tools. Although it has Zigbee2Mqtt / MosquittoBroker working well, detecting and communicating with Zigbee equipment. Interestingly, the Add-On Zigbee Network Map also doesn't represent anything, nor does the CC2531 cordinator. I was suspicious of the firmware, but I flashed several times, some with Z-Stack 1.2 and other times with Z-Stack 3.0 on the C2531 ... But it doesn't work.

[image]

[image]

[image]

**Example with Z-Stack 3.0 flashed on CC2531**

[image]

## What did you expect to happen

Zigbee devices be integrated HA after be discovered

## How to reproduce it (minimal and precise)

I cant find the problem, seems all is working fine, but devices are not integrating

## Debug Info
Zigbee2mqtt version: zigbee2mqtt-1.14.2
Adapter hardware: CC2531 
Adapter firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-07-29T21:55:57: PM2 log: Launching in no daemon mode
2020-07-29T21:55:57: PM2 log: App [npm:0] starting in -fork mode-
2020-07-29T21:55:57: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.2 start /zigbee2mqtt-1.14.2
> node index.js
zigbee2mqtt:info  2020-07-29 21:56:01: Logging to console and directory: '/share/zigbee2mqtt/log/2020-07-29.21-56-01' filename: log.txt
zigbee2mqtt:info  2020-07-29 21:56:02: Starting zigbee2mqtt version 1.14.2 (commit #unknown)
zigbee2mqtt:info  2020-07-29 21:56:02: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-07-29 21:56:04: zigbee-herdsman started
zigbee2mqtt:info  2020-07-29 21:56:04: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-07-29 21:56:04: Currently 2 devices are joined:
zigbee2mqtt:info  2020-07-29 21:56:04: 0x04cf8cdf3c763c2b (0x04cf8cdf3c763c2b): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
zigbee2mqtt:info  2020-07-29 21:56:04: 0x00158d000275c971 (0x00158d000275c971): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
zigbee2mqtt:warn  2020-07-29 21:56:04: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-07-29 21:56:04: Allowing new devices to join.
zigbee2mqtt:warn  2020-07-29 21:56:04: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-07-29 21:56:04: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-07-29 21:56:04: Connecting to MQTT server at mqtt://core-mosquitto:1883
zigbee2mqtt:info  2020-07-29 21:56:04: Connected to MQTT server
zigbee2mqtt:info  2020-07-29 21:56:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Koenkk, 2020-07-30:
Can you provide some more logging of the zigbee2mqtt startup? (do you see zigbee2mqtt publish to `homeassistant/*`?

hista, 2020-08-09:
I may have a similar issue (I was hesitating opening another issue but identical, or posting here. If you prefer I open a new one don't hesitate to tell :-)

As OP, my zigbee devices are detected in zigbee2mqtt, which pushes their information to mqtt server… and then I lose their trace, they don't integrate in HA (hassio on a RaspberryPi3).

cat /usr/share/hassio/share/zigbee2mqtt/configuration.yaml: 
https://framabin.org/p/?7303f5f5785a0791#p4xGEtVir0Kfz5CjIqbq0lS6/UlcwxCydGVOQxkC7dQ=

The zigbe2mqtt logs:
https://framabin.org/p/?70aa5dd430394e44#N970wvWuSrSLNR6ommqyblenQcbxFr2z7ULCtQcO6Pw=

The core_mosquitto logs:
https://framabin.org/p/?b1eb858d27523bbf#MkYDT6xQrKRuSYFm1/fPHO+7ROB1kijRscMD7TEAKwg=

The core_mosquitto configuration file:
https://framabin.org/p/?f9f199ace004eb8b#Tuwl0R7X7UsU8fT/xwfxRJ1GQA47vkoNx3GwEW96zfc=

And finally my HA configuration file:
cat /usr/share/hassio/homeassistant/configuration.yaml: 
https://framabin.org/p/?64e822cedd6dad1e#39Q89E8T90aYGvOaFsV07BwcfrPDI8IPq7RzQSNUUzI=

I suspect I missed an easy trick, but I can't see which one :-/

Don't hesitate to tell if you need further information (or a separate ticket)!

Koenkk, 2020-08-09:
@hista you've specified a user/password for MQTT in the zigbee2mqtt configuration.yaml, but didn't for the Home Assistant confguration.yaml, perhaps Home Assistant cannot connect to the MQTT server because of that.

wilsonlspacheco, 2020-08-10:
> homeassistant

wilsonlspacheco, 2020-08-10:
> an you provide some more logging of the zigbee2mqtt startup? (do you see zigbee2mqtt publish to `homeassistant/*`?

Yes

> 2020-08-10T20:12:47: PM2 log: Launching in no daemon mode
> 2020-08-10T20:12:48: PM2 log: App [npm:0] starting in -fork mode-
> 2020-08-10T20:12:48: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.2 start /zigbee2mqtt-1.14.2
> node index.js
zigbee2mqtt:info  2020-08-10 20:13:14: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-10.20-13-13' filename: log.txt
zigbee2mqtt:info  2020-08-10 20:13:16: Starting zigbee2mqtt version 1.14.2 (commit #unknown)
zigbee2mqtt:info  2020-08-10 20:13:16: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-08-10 20:13:21: zigbee-herdsman started
zigbee2mqtt:info  2020-08-10 20:13:21: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-08-10 20:13:21: Currently 2 devices are joined:
zigbee2mqtt:info  2020-08-10 20:13:21: 0x04cf8cdf3c763c2b (0x04cf8cdf3c763c2b): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
zigbee2mqtt:info  2020-08-10 20:13:21: 0x00158d000275c971 (0x00158d000275c971): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
zigbee2mqtt:warn  2020-08-10 20:13:21: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-08-10 20:13:21: Allowing new devices to join.
zigbee2mqtt:warn  2020-08-10 20:13:21: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-08-10 20:13:21: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-08-10 20:13:21: Connecting to MQTT server at mqtt://core-mosquitto:1883
zigbee2mqtt:info  2020-08-10 20:13:22: Connected to MQTT server
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c763c2b', payload '{"state":"ON","power":0,"voltage":227,"current":0,"consumption":0.29,"temperature":31,"linkquality":92}'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'zigbee2mqtt/0x00158d000275c971', payload '{"battery":100,"voltage":3025,"illuminance":138,"illuminance_lux":138,"linkquality":94,"occupancy":true}'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.2","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c763c2b/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b/set","state_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","name":"0x04cf8cdf3c763c2b_switch","unique_id":"0x04cf8cdf3c763c2b_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c763c2b"],"name":"0x04cf8cdf3c763c2b","sw_version":"Zigbee2mqtt 1.14.2","model":"Mi power plug ZigBee EU (ZNCZ04LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c763c2b/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:flash","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","name":"0x04cf8cdf3c763c2b_power","unique_id":"0x04cf8cdf3c763c2b_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c763c2b"],"name":"0x04cf8cdf3c763c2b","sw_version":"Zigbee2mqtt 1.14.2","model":"Mi power plug ZigBee EU (ZNCZ04LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-10 20:13:22: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c763c2b/consumption/config', payload '{"unit_of_measurement":"kWh","value_template":"{{ value_json.consumption }}","icon":"mdi:flash","state_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","name":"0x04cf8cdf3c763c2b_consumption","unique_id":"0x04cf8cdf3c763c2b_consumption_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c763c2b"],"name":"0x04cf8cdf3c763c2b","sw_version":"Zigbee2mqtt 1.14.2","model":"Mi power plug ZigBee EU (ZNCZ04LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-10 20:13:23: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c763c2b/current/config', payload '{"unit_of_measurement":"A","icon":"mdi:current-ac","value_template":"{{ value_json.current }}","state_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c763c2b","name":"0x04cf8cdf3c763c2b_current","unique_id":"0x04cf8cdf3c763c2b_current_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c763c2b"],"name":"0x04cf8cdf3c763c2b","sw_version":"Zigbee2mqtt 1.14.2","model":"Mi power plug ZigBee EU (ZNCZ04LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-10 20:13:23: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c763c2b/voltage/config', payload '{"unit_of_measurement":"V","icon":"mdi:alpha-v","value_template":"{{ value_json.voltage }}","state_topic":"zigbee2mqtt/0x04cf [... cut ...]
zigbee2mqtt:info  2020-08-10 20:13:23: MQTT publish: topic 'homeassistant/sensor/0x00158d000275c971/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d000275c971","json_attributes_topic":"zigbee2mqtt/0x00158d000275c971","name":"0x00158d000275c971_linkquality","unique_id":"0x00158d000275c971_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000275c971"],"name":"0x00158d000275c971","sw_version":"Zigbee2mqtt 1.14.2","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'`

closed 2020-09-30 by github-actions[bot]