Koenkk/zigbee2mqtt #4013: How to add NLG-remote (Paul Neuhaus Q-Remote)

Korschan, 2020-07-30:
Hi Koenkk,

since the NLG-Remote (Paul Neuhaus Q-Remote) is supported, I wanted to pair my Remote with C2531 - but it does not work.
I succesfully paired all my Paul Neuhaus Lights, but I dont see any announcements about my remote.

Tried to reset it, but it didnt help.

Thanks,
K.

arodus, 2020-07-31:
Hi Korshan, I also just added all my Paul Neuhaus lights which was easy. Adding my Q-Remotes was a huge pain: There are a lot different models of the remote existing which are looking exactly the same (well one model has the on/off buttons switched).

I have 4 different models here, 2 were working out of the box, one required adding the model to zigbee2mqtt (https://github.com/Koenkk/zigbee-herdsman-converters/pull/1445) and for the last model I was unable to connect it to zigbee2mqtt (yet).

So if you don't get any warnings like `Zigbee model 'x' is NOT supported,` it should work using these steps:
1. Permit new devices in zigbee2mqtt
2. Reset the remote (pressing the reset button until the led flashes 4 times and then 10 times fast)
3. Hold the remote near to the controller (< 10cm)
4. press the on button for min 5 secs until the LED is flashing again

When pressing `1-3` and `off` afterwards all lights should turn off (no idea how to change that 🙈 ). If it's not working repeat steps 1-4 a few times.

Korschan, 2020-08-02:
Hi arodus,

thanks, I successfully paired my NLG-remote!
I also have another Q-remote, which I was not able to pair.
It's this one: https://zigbeealliance.org/de/zigbee_products/q-Fernbedienung/

Did the reset, did the pair, but nothing to see in the logs.

arodus, 2020-08-13:
I'm glad pairing the NLG-Remote worked!

I think I have the same one lying around somewhere. But never tried that one as I have way too many of the other ones. But might try this one too. Given I manage to find it somewhere.

arodus, 2020-08-19:
Hi @Korschan, finally found the same remote you have, also got it working after a few tried.

```
zigbee2mqtt:info  2020-08-19 08:02:58: Successfully interviewed '0x00158d000256f504', device has successfully been paired
zigbee2mqtt:info  2020-08-19 08:02:58: Device '0x00158d000256f504' is supported, identified as: Paul Neuhaus Q-REMOTE (100.462.31)
```

What I did:
Resetting the remote: Holding the `W` button until the status light flashes then release the `W` button and press on/off within a few seconds. The status LED should flash again to confirm the reset. When pressing +/- afterwards just one of the group inidicator LEDs should turn on but not the status LED.

After that I tried to things, not to sure which one actually initiated the pairing:
1) Holding the power button until the status LED flashes 
2) Holding one of the group buttons until the status LED flashes

Hope you also have success pairing your remote.

closed 2020-10-06 by github-actions[bot]

DavidFFerreira, 2021-11-24:
hi I have a q-Remote and its not paring with z2m =( its the E0040044
https://zigbeealliance.org/zigbee_products/radio-remote-control-rgbcct-e0040044/
[image]

DavidFFerreira, 2021-11-24:
https://zigbeealliance.org/zigbee_products/radio-remote-control-rgbcct-e0040044/

DavidFFerreira, 2021-11-24:
@Koenkk can u help-me?
[image]

DavidFFerreira, 2021-11-25:
@arodus my remote is the same yes?