Koenkk/zigbee2mqtt #4016: Change Home Assistant Module's Birth and LWT Messges To Home Assistant's New Default

danieledwardgeorgehitchcock, 2020-07-30:
Home Assistant (as-of v0.113) now publishes by default to Birth and Last Will and Testament topics, see [here](https://www.home-assistant.io/docs/mqtt/birth_will/). This differs slightly from what is recommended in the documentation i.e. 'hass/status' vs 'homeassistant/status'.

If the plugin and documentation were to change to the new topic, it would avoid the extra configuration required at Home Assistant's end.

Koenkk, 2020-07-30:
This has already been addressed in https://github.com/Koenkk/zigbee2mqtt/issues/3918 / https://github.com/Koenkk/zigbee2mqtt/commit/a845d35b8ef9a4363930bb1ce825dd03bea21523 / https://github.com/Koenkk/zigbee2mqtt.io/commit/5bc56e904a7bd7e14d54569322e9fa6b7f6c0fe2

By default zigbee2mqtt subscribes to `homeassistant/status` now, documentation has also been updated: https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery

closed 2020-07-30 by Koenkk

danieledwardgeorgehitchcock, 2020-07-31:
I’m still seeing `hass/status` in the documentation, hence the confusion.

Thanks.

Koenkk, 2020-07-31:
That's true indeed, but it's not needed anymore stated in: "# birth_message and will_message is not required anymore for Home Assistant 0.113 >="