Koenkk/zigbee2mqtt #4019: Pairing Not Compatible Yale YRD221 RL Lock

hendmele, 2020-07-31:
I am trying to pair Yale YRD221 RL Lock (as far as I know, it is the same as YRD221, but with a different mechanical lock).

As this is not a compatible device, I am running Zigbee2MQTT on Windows to make it easier to troubleshoot than on my rpi.

Device pairs succefully and gives the following messages:

> zigbee2mqtt:info  2020-07-30 19:27:28: Device '0x000d6f0010fa0350' joined
zigbee2mqtt:info  2020-07-30 19:27:28: Starting interview of '0x000d6f0010fa0350'
zigbee2mqtt:info  2020-07-30 19:27:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f0010fa0350"}}'
zigbee2mqtt:info  2020-07-30 19:27:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f0010fa0350"}}'
zigbee2mqtt:info  2020-07-30 19:27:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x000d6f0010fa0350"}}'
zigbee2mqtt:info  2020-07-30 19:27:46: Successfully interviewed '0x000d6f0010fa0350', device has successfully been paired
zigbee2mqtt:warn  2020-07-30 19:27:46: Device '0x000d6f0010fa0350' with Zigbee model 'NO LOCK PRESENT' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-07-30 19:27:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f0010fa0350","supported":false}}'

I just think it is odd that device name says 'NO LOCK PRESENT', but couldn't find anything about that.

One of the times I paired, I was with 'debug' mode on, so I have a more complete log here ->https://pastebin.com/fyezMV9q

Problem is that after this I am stuck.
I tried going thorugh the 'How to support new devices' wiki with no luck
I added the following code to my devices.js but I got nothing. No events when opening/closing the lock.

> {
        zigbeeModel: ['NO LOCK PRESENT'],
        model: 'NO LOCK PRESENT',
        vendor: 'Yale',
        description: 'deadbolt lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [],
	toZigbee: [],	
    }

Then I searched for other Yale locks on devices.js files and tried using their 'configuration' parameter, also no luck.

I am stuck.
Can anyone point me on any direction?
I am a little bit worried with this 'NO LOCK PRESENT' model that the lock is spitting out, it appears that the zigbee module is not connected to the lock itself. Problem is that to pair, I have to go through the lock menu, not directly on the zigbee module, and it is pairing in less than 2 seconds, every time.

hendmele, 2020-07-31:
Hi.

Adding some more information I found out today.

I copied this configuration from a compatible device and just changed the zigbeeModel parameter

> {
        zigbeeModel: ['NO LOCK PRESENT'],
        model: 'YRD226HA2619',
        vendor: 'Yale',
        description: 'Assure lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.generic_lock],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

It recognized so I just let it sit for a while.

After 20 minutes this message appeared on the log:

> zigbee2mqtt:debug 2020-07-31 10:08:23: Received Zigbee message from '0x000d6f0010fa0350', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0

Then after almost 1 hour, the same message:

> zigbee2mqtt:debug 2020-07-31 11:15:37: Received Zigbee message from '0x000d6f0010fa0350', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0

I opened the door between this two messages and left it open, but the 'lockState' doesn't seem to have changed.

hendmele, 2020-07-31:
Well, another progress.

Found this one model on devices.js with a comment saying it could work for YRD221RL (my model)
So I duplicated it adding "NO LOCK PRESENT" as the model.

I can now send MQTT lock/unlock commands and it will work. I just don't receive anything from the lock.

> {
        zigbeeModel: ['NO LOCK PRESENT'],
        // The zigbee module card indicate that the module will work on YRD 221 and YRD 221RL also
        model: 'YRL-221RL',
        vendor: 'Yale',
        description: 'Real living keyless leveler lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery_not_divided],
        toZigbee: [tz.generic_lock],
		// fromZigbee: [],
		// toZigbee: [],
        meta: {configureKey: 6},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },

Koenkk, 2020-08-01:
Do you mean you are not receiving a status update when locking/unlocking from the lock itself?

Did the device configured successfully? You can force it with https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

hendmele, 2020-08-01:
Yes @Koenkk I am not receiving a status update when I am locking/unlocking from the lock itself.
In fact apparently I am not receiving any status update from the lock, even with debug mode on.
I am supposed to receive a battery status, for example, right?

I have forced the configuration again and received a successfull message, but didn't help.

> zigbee2mqtt:debug 2020-08-01 10:49:56: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data '0x000d6f0010fa0350'
zigbee2mqtt:info  2020-08-01 10:49:56: Configuring '0x000d6f0010fa0350'
zigbee2mqtt:info  2020-08-01 10:50:03: Successfully configured '0x000d6f0010fa0350'

hendmele, 2020-08-01:
If I publush this do MQTT I get the status back.

> zigbee2mqtt/0x000d6f0010fa0350/get

Koenkk, 2020-08-02:
Are you sure that `fz.lock` is in the `fromZigbee` list because that should handle: 

```
zigbee2mqtt:debug 2020-08-01 11:02:28: Received Zigbee message from '0x000d6f0010fa0350', type 'readResponse', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-01 11:02:28: No converter available for 'YRL-221RL' with cluster 'closuresDoorLock' and type 'readResponse' and data '{"lockState":1}'
```

hendmele, 2020-08-03:
Hi, I think I disabled it for debugging purposes.

This lock has a separated Zigbee module and I just found out that the Zigbee module the store sent me is not 100% compatible with my lock, this is why I am not receiving any nor correct values from the lock itself.

I am waiting for the new one to arrive.
When I receive it, I'll update this issue with the working template for Yale YRD-221RL lock.

hendmele, 2020-08-11:
Received the new Yale zigbee module today and installed.
Zigbee2mqtt started working right out of the box although my lock model is not on the supported devices list.

Turns out that although my lock model on its box is Yale YRD 221 RL the zigbee model is Yale YRD220/240 TSDB. Since this last module was already supported, it works without problems apparently (still waiting for the battery level to update).

Is there a way for this to be added to the docs? It may help other to know that YRD 221 RL model is supported by zigbee2mqtt already.

closed 2020-08-11 by hendmele

Koenkk, 2020-08-11:
@hendmele didn't they just sent a zigbee module meant for a different lock? The zigbee model explicitly mentions YRD220/240 TSDB

hendmele, 2020-08-11:
> @hendmele didn't they just sent a zigbee module meant for a different lock? The zigbee model explicitly mentions YRD220/240 TSDB

The zigbee module is the same for many models. My module shows it is compatible with YRD 221, YRD 221 RL an YRL 220L (3 models branded in Brazil).

When you plug the module to the lock, it is supposed to report the lock model.  For some reason, lock YRD221 RL reports model YRD220/240 TSDB. Probably they are in fact the same model (visually they look the same), and YRD221RL is just a localized code branded here in Brazil.

Links for these two products below.

YRD 220 -> https://www.asklocksmith247.com/yale-yrd220-z-wave-touchscreen-deadbolt/
yrd 221RL -> https://www.yale.com.br/pt-BR/produtos/fechaduras-digitais-e-biometricas/yrd-221-rl/yrd-221-rl/

Koenkk, 2020-08-12:
I've changed the model to `YRD220/YRD221` in the dev branch, hopefully that clarifies it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

hendmele, 2020-08-12:
Looks good, thanks.

Now I am going to start looking about the other commands available on Yale locks, like adding/removing PIN codes.
If you have already something under development that could be a start, let me know.

Koenkk, 2020-08-13:
Check https://github.com/Koenkk/zigbee-herdsman-converters/pull/1357