Koenkk/zigbee2mqtt #4020: Re-support Zigbee Led Controller (TUYA - Smartlife)

wolverinevn, 2020-07-31:
[Link to product](https://www.aliexpress.com/item/4000976025529.html?spm=a2g0o.productlist.0.0.5d7d1c31nH9vOs&algo_pvid=c9779b47-9a15-4396-9889-362a4fe624cf&algo_expid=c9779b47-9a15-4396-9889-362a4fe624cf-15&btsid=0ab6f81e15961926740947134e2903&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

It was add by me under Zigbee model TS0503A.
Document found at: https://www.zigbee2mqtt.io/devices/TYZS1L.html

Issue, so far as I know:
+ Device only work with HS color mode
+ Able to turn it off with genOnOff: offWithEffect
+ Able to turn it on with genLevelCtrl: moveToLevelWithOnOff
+ Able to change color with lightingColorCtrl: moveToHue & moveToSaturation _seperately_
+ moveToHueAndSaturation does not work
+ Can't turn it on with any genOnOff command
+ No brightness control so far

My codes to make it partly work:

#### fromZigbee #####
```
    tuya_ledstrip: {
        cluster: 'lightingColorCtrl',
        type: ['attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            /*
            if (msg.data.hasOwnProperty('61440')) {
                const property = getProperty('state', msg, model);
                return {[property]: msg.data['61440'] === 1 ? 'ON' : 'OFF'};
            }
            */
            
            const result = {};
				
			if (msg.data['61441']) {
                result.brightness = msg.data['61441'];
            }
            
            result.color = {};
            
            if (msg.data.hasOwnProperty('currentHue')) {
                result.color.h = precisionRound((msg.data['currentHue'] * 360) / 254, 0);
            }
            
            if (msg.data['currentSaturation']) {
                result.color.s = precisionRound(msg.data['currentSaturation'] / 2.54, 0);
            }
                
            return result;
        },
```
#### toZigbee ####
```
    tuya_on_off: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            if (value.toLowerCase() === 'off') {
                await entity.command('genOnOff', 'offWithEffect', {effectid: 0x01, effectvariant: 0x01}, getOptions(meta.mapped, entity));
            } else {
                const payload = {level: Number(value), transtime: 0};

                await entity.command('genLevelCtrl', 'moveToLevelWithOnOff', payload, getOptions(meta.mapped, entity));
                //await entity.command('genOnOff', 'onWithTimedOff', {ctrlbits: 0, ontime: 1, offwaittime: 1}, getOptions(meta.mapped, entity));
            }
                return {state: {state: value.toUpperCase()}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genOnOff', ['onOff']);
        },
    },
    tuya_light_brightness: {
        key: ['brightness', 'brightness_percent'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'brightness_percent') {
                value = Math.round(Number(value) * 2.55).toString();
            }
        	//const payload = {level: Number(value), transtime: getTransition(entity, key, meta).time};
        	const payload = {level: Number(value), transtime: 0};

            await entity.command('genLevelCtrl', 'moveToLevelWithOnOff', payload, getOptions(meta.mapped, entity));
            return {state: {brightness: Number(value)}, readAfterWriteTime: payload.transtime * 100};
            
        },
        convertGet: async (entity, key, meta) => {
            //await entity.read('genLevelCtrl', ['currentLevel']);
        },
    },

    tuya_color: {
        key: ['color'],
        convertSet: async (entity, key, value, meta) => {
            const hue = {};
            const saturation = {};
            
            hue.hue = Math.round((value.h * 254) / 360);
            saturation.saturation = Math.round(value.s * 2.54);
            
            hue.transtime = saturation.transtime = 0;
            hue.direction = 0;
            console.log(hue, saturation);

            await entity.command('lightingColorCtrl', 'moveToHue', hue, {}, getOptions(meta.mapped, entity));
            await entity.command('lightingColorCtrl', 'moveToSaturation', saturation, {}, getOptions(meta.mapped, entity));        
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('lightingColorCtrl', ['currenthHue','currentSaturation']);
        },

    },
```
#### device.js ####
```
{
    zigbeeModel: ['TS0503A'],
    model: 'TYZS1L',
    vendor: 'TuYa',
    description: 'Zigbee LED Strip controller HSB',
    fromZigbee: [fz.on_off,fz.tuya_ledstrip],
    toZigbee: [tz.tuya_on_off,tz.tuya_color,tz.tuya_light_brightness,tz.ignore_transition, tz.ignore_rate],
};
```
#### db and logs ####
```
{"id":2,"type":"Router","ieeeAddr":"0x086bd7fffed356f9","nwkAddr":37736,"manufId":4098,"manufName":"_TZ3000_c5drg1wn","powerSource":"Mains (single phase)","modelId":"TS0503A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503A","manufacturerName":"_TZ3000_c5drg1wn","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0}},"lightingColorCtrl":{"attributes":{"61440":1,"61441":255,"currentHue":0,"currentSaturation":254}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1596076607439}
```
log:
```
debug 2020-07-29 12:18:37: Received Zigbee message from '0x086bd7fffed356f9', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-07-29 12:19:27: Received Zigbee message from '0x086bd7fffed356f9', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2020-07-29 12:19:28: Received Zigbee message from '0x086bd7fffed356f9', type 'attributeReport', cluster 'lightingColorCtrl', data '{"61440":1}' from endpoint 1 with groupID 0
debug 2020-07-29 12:19:28: Received Zigbee message from '0x086bd7fffed356f9', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":0}' from endpoint 1 with groupID 0
debug 2020-07-29 12:19:28: Received Zigbee message from '0x086bd7fffed356f9', type 'attributeReport', cluster 'lightingColorCtrl', data '{"61441":255,"currentHue":0,"currentSaturation":254}' from endpoint 1 with groupID 0
debug 2020-07-29 12:19:28: Received Zigbee message from '0x086bd7fffed356f9', type 'raw', cluster 'lightingColorCtrl', data '{"type":"Buffer","data":[24,8,10,3,240,72,10,0,0,14,13,46,224,232,204,0,0]}' from endpoint 1 with groupID 0
```

So sorry for not testing it carefully at first submission (I have some others wifi controllers and a lot works to do so confusing, I guess)!
And so sorry for very long post.

Glad to make pull request if you need it.

Have a nice weekend! Great project.

Koenkk, 2020-08-01:
Could you make a PR?

wolverinevn, 2020-08-01:
> Could you make a PR?

Done.

closed 2020-08-03 by wolverinevn