Koenkk/zigbee2mqtt #4026: feature proposal: hue compensation / calibration

andreasbrett, 2020-08-02:
Not sure if this would be beneficial to a lot of people but how about adding a feature that will allow compensating different color renditions of our lights. I happen to have different brands in my collection of zigbee lights/lamps and each render colors differently. Unfortunately it's not a constant shift, so simply adding or subtracting some degrees of hue did not work. To compensate for these complex shifts I did the following:

- dialed in different sample colors on a lamp I chose to be the **reference lamp** (the one that matched colors on my color picker the best)
- dialed in colors that matched the resulting color of my reference lamp and noted that particular hue
- built a hue compensation map for each light (**hueIn -> hueOut**)
- built some JS code within the ruleset of my smarthome software to compensate the hue and roughly get

This JS code is running simple linear interpolation on an incoming hsb value based on a hueIn->hueOut map. Of course this is not perfect and will still result in small shifts but my lights are much closer now and it's really nice to actually get the color I picked in my smarthome app's color picker.

I think this hue compensation / calibration feature would be better suited to be built right into z2m to compensate in the last step right before sending the value to the device.

andreasbrett, 2020-08-02:
Here's some example code on that really basic (ranged) linear interpolation that works quite well though.

```
const mapLamp1 = {
	28: 45,
	89: 109,
	184: 203,
	334: 318
}

const interpolateHue = (value, conversionMap) => {
    let leftIn = 0, rightIn = 359
    for (let valueIn of Object.keys(conversionMap)) {
        if (valueIn <= value) leftIn = valueIn
        else {
            rightIn = valueIn
            break
        }
    }

    const proportion = 1 - (rightIn - value) / (rightIn - leftIn)
    const leftOut = conversionMap[leftIn] || 0
    const rightOut = conversionMap[rightIn] || 359
    return Math.round(leftOut + proportion * (rightOut - leftOut))
}
```

Koenkk, 2020-08-03:
I'm fine in adding such, you can access the device options in the converter (https://github.com/Koenkk/zigbee-herdsman-converters/blob/b9c0935da0cfc05f2d701a5524d239c0b117316c/converters/toZigbee.js#L541) via e.g. `meta.options.hue_saturation_correction`. When making a PR also make a PR to update the docs, updating one page is enough (e.g. https://www.zigbee2mqtt.io/devices/BY_185_C.html#device-type-specific-configuration), I will make sure it will land on all bulb pages.

andreasbrett, 2020-08-03:
@Koenkk Thanks, great to hear that this is fine with you. I added PRs a couple of minutes ago.

Koenkk, 2020-08-04:
Merged, assuming this can be closed now.

closed 2020-08-04 by Koenkk