Koenkk/zigbee2mqtt #4027: Future Request: Bind only specific cluster

jd1900, 2020-08-02:
It would be great to be able to bind  only to specific cluster when binding to a group or device.

For instance, if I have a remote which is able to use the OnOff cluster and regulate intensity, I may want to just use the OnOff, and don't bind the other cluster/s.

Is it possible right now?
Otherwise, could you implement it?

sjorge, 2020-08-02:
This is already implemented on the current dev using the new api, you can optionally specify an array of clusters in the payload.

jd1900, 2020-08-02:
> This is already implemented on the current dev using the new api, you can optionally specify an array of clusters in the payload.
> 
> [946ebe1](https://github.com/Koenkk/zigbee2mqtt/commit/946ebe16b6a375af179f248b578d9e48a950e279)

Great! 
Thank you for the information. I guess we'll need some time until getting into the main branch and update the docs.

closed 2020-08-02 by jd1900