Koenkk/zigbee2mqtt #4030: Failure of configuration during pairing Philips SML002

TilmanK, 2020-08-03:
# Bug Report
## What happened
I tried to pair a Philips SML002, activated paring, resetted the device and pairing started instantly:

`info  2020-08-03 17:46:27: Device '0x0017880106444d2d' joined
info  2020-08-03 17:46:27: Starting interview of '0x0017880106444d2d'
info  2020-08-03 17:46:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880106444d2d"}}'
info  2020-08-03 17:46:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880106444d2d"}}'
debug 2020-08-03 17:46:27: Device '0x0017880106444d2d' announced itself
info  2020-08-03 17:46:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106444d2d"}}'`

But then it seems like some message from genBasic can't be parsed?

`Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"modelId":"SML002"}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:42: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SML002"}'
debug 2020-08-03 17:46:42: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Philips"}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:42: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"Philips"}'
debug 2020-08-03 17:46:43: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:43: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
debug 2020-08-03 17:46:43: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:43: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
debug 2020-08-03 17:46:44: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:44: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":2}'
debug 2020-08-03 17:46:44: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:44: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":1}'
debug 2020-08-03 17:46:45: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:45: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2020-08-03 17:46:46: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190219"}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:46: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20190219"}'
debug 2020-08-03 17:46:46: Received Zigbee message from '0x0017880106444d2d', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"6.1.1.27575"}' from endpoint 1 with groupID 0
debug 2020-08-03 17:46:46: No converter available for '9290019758' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"6.1.1.27575"}'`

In the end pairing is successful:
`info  2020-08-03 17:46:47: Successfully interviewed '0x0017880106444d2d', device has successfully been paired
info  2020-08-03 17:46:47: Device '0x0017880106444d2d' is supported, identified as: Philips Hue motion outdoor sensor (9290019758)`

But configuration fails:
`error 2020-08-03 17:46:57: Failed to configure '0x0017880106444d2d', attempt 1 (Error: Bind 0x0017880106444d2d/2 genPowerCfg from '0x00124b001caabcf6/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))`

The firmware of the motion sensor is up to date. Configuring the device fails also on manual tries.

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200417
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5016993/log.txt)

TilmanK, 2020-08-03:
Ah, yeah, and I forgot to mention: The device isn't sending anything...

TilmanK, 2020-08-03:
And... I found a second SML002 still boxed (I tend to buy things on storage seemingly...).

I get the same converter "failures", but configuration works instantly and the device sends updates. Any advice?

Koenkk, 2020-08-03:
The no converter available messages are not errors, it just means that this message will not be converted to MQTT, which is fine as the information in it is not interesting. 

Can you try moving the device closer to the coordinator and right before executing the manual configure wakeup the device by pressing a button on it?

TilmanK, 2020-08-04:
The solution was simple. Once I exchanged the batteries, the sensor started to work as expected 👍

closed 2020-08-04 by TilmanK