Koenkk/zigbee2mqtt #4033: How to read actual transition time for Ikea Tradfri?

M0ebiu5, 2020-08-04:
The "get" command does not show the actual transition time for Tradfri bulbs.
Here is an example result:
`{"state":"OFF","linkquality":55,"brightness":1,"update_available":false}`
Even if i specify "transition" in the payload, it doesnt work...
How can i get that info?

NoWhere2Go19, 2020-08-06:
The transition time can be set in the payload (working for me on all light devices):
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset

M0ebiu5, 2020-08-06:
@NoWhere2Go19 you misread my question - i dont want to set it, i want to read (get, retrieve) the actual transition time of the lamp. Usually a device also allows to read it's current settings, but not in this case it seems.

NoWhere2Go19, 2020-08-07:
Bulbs have a transition of 0. Otherwise you would be annoyed if it has a default from the manufacturer. Therefore you can put it in the payload on the set. 
Additionally I have never seen a zigbee bulb that exposes its current in a channel. There might be one, but I haven't come across one and I'm using Osram, tradfi and innr

M0ebiu5, 2020-08-07:
at: NomenEstOmen: There was never the question about the default values of the transition time - it is about the get method of z2m and why it is inconsistent to the set function.

I think the get method should provide all setable properties - if the bulb does not send it, this should seamlessly be added by z2m (the tradfri bulbs even remember the transition time after power off, but i dont have the ikea gateway to check, if they have a possibility to read back the transition time setting).

elfhack, 2020-08-13:
@M0ebiu5 What we're trying to say is that all currently known bulbs have a default 0 transition time. You could make a case for yourself saying that you want to know what you've set in the config (eg set a transition: 2 for a specific bulb, and want to see it reflected in the payload). What's your usecase for this?

M0ebiu5, 2020-08-14:
@elfhack 
The ikea bulbs have a property "transition time". 
This property can only be set not read. 
This is inconsistant. 
There are many use cases, especially for these bulbs, because they remember the transition time over a power cycle. So, what would be the best way to get the current transition time, if not asking the device itself?

closed 2020-09-21 by github-actions[bot]