Koenkk/zigbee2mqtt #4036: IKEA E1744 SYMFONISK sound controller - pairing ok but stopped working

Lipown, 2020-08-04:
# Bug Report
After latest zigbee2mqtt homeassistant addon update IKEA E1744 SYMFONISK sound controller stopped working. Re-pairing log is without issues but useless.

## What did you expect to happen
IKEA E1744 SYMFONISK sound to send mqtt messages as always.

## How to reproduce it (minimal and precise)
Try to re-pair the controler and you can see the pairing is ok in the log but after any physicall action on the controller you will not see output in mqtt.

## Debug Info
Zigbee2mqtt version: 20190608
Adapter hardware: CC2531
Adapter firmware version:

flexyneat, 2020-08-05:
I've got the exact same setup as OP and experienced same issues. I even applied the available ota update to the Ikea sound controller but that didn't help. It just sends and update about 5mins apart saying it's online but button clicks and rotation messages are not published. I even tried replacing the batteries and re-pairing but no dice.

I had a spare brand new sound controller. I was able to set it up with z2m and it works fine publishing messages as I rotate and click.

I've got no idea why the first one stopped working, I only remember I started after upgrading to z2m 1.14.2

flexyneat, 2020-08-15:
Just updated to z2m 1.14.3 and the new ikea volume controller that has been working for the last week just stopped working. Just like the first one, it no longer publishes actions, it only send and online update once in a while.

Lipown, 2020-08-17:
I somehow managed to get device working but without brightness values properly changing. Then I disconnedted the device from network and tried to join again without any succes.

Koenkk, 2020-08-17:
Can you provide the debug logging when expecting actions from the device?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Lipown, 2020-08-18:
Today I made it pair again (yesterday it was not possible) but the rotatation weirdly changes values. When turning left, it sometimes report right movement or stick on the values. When on right it seems ok.

https://pastebin.com/x3QbFdeh

Koenkk, 2020-08-18:
@Lipown I recommend to disable the legacy integration: https://www.zigbee2mqtt.io/devices/E1744.html#legacy-integration

Lipown, 2020-08-19:
I have it disabled.
[image]

Koenkk, 2020-08-19:
Your log shows `0xccccccfffebe46eb` while you added `legacy: false` to `0x00158d000320b56a`

Lipown, 2020-08-19:
Ha, thanks! However it's almost the same. Brightness values now stays on 255 and brightness_move_down and up are random when rotating the encoder.

Koenkk, 2020-08-20:
Brightness will indeed not be updated anymore with in the new integration. The issue is really strange, it looks like the device is sending the wrong messages to Zigbee2MQTT? Have you already tried downgrading, does this fix the problem?

Lipown, 2020-08-20:
Oh thats a pitty cause my volume control is based on value from brightness. I need to find new way

I did not tried downgrade I will check how to do that.

axwell, 2020-09-16:
i have the same issue, did anybody solved this ?

flexyneat, 2020-09-16:
Still stuck with it. I noticed whenever I replace the battery it starts working again, but after 2-3 weeks its back to not working again.
Just ordered a new zigbee stick, will change battery after setting up my new zigbee mesh

ChillXXL, 2020-09-16:
Same issues here that it loses connection. I have two devices and can happen that one keeps connected (also for a short while) but the other one drops connection suddenly after a couple of minutes after pairing. 
Only repair can solve this for a while but unfortunately this is not workable. 

I also tried to merge them in a group, (which I read would solve the drop of the pairing) but didn't solve.
I also recognize that if it works turning it 'up' , it sometimes incidentally signals a 'down' and visa versa.

I hope maintaining a signal can be solved also. If I can test somethings out, please let me know.

ChillXXL, 2020-09-27:
> new zigbee stick.

Which one(s) have you tried out?

flexyneat, 2020-11-29:
Anybody still having this issue? I never got mine resolved before my cc2531 stick got bad. I ordered a better zigbe stick from slaesh on to die.com since September and it’s yet to deliver

philipforget, 2020-12-17:
Yeah, I've still got this issue myself. The devices works for ~1 week at a time before having to manually unpair it and repair it several times. I dont have this issue with other IKEA switches, there's just something about the IKEA E1744 that seems to be problematic. I'll keep digging.

Imperial-Guard, 2021-01-21:
Facing same issues here, I will add legacy: false and monitor it for the upcoming time.

jeffgoh, 2021-02-06:
> Zigbee2mqtt version: 1.14.2
> Adapter hardware: CC2531
> Adapter firmware version: 20190608

My CC2531 came from iTEAD pre-flashed - 20180507 - seems a bit old, but works for everything else so far except the IKEA E1744. This is on zigbee2mqtt 1.17.0 - updating now to z2m 1.17.1 - but the firmware update process seems a bit daunting - especially since I do not have a spare coordinator in case the flash process bricks my current CC2531 stick. Will start looking for a better option than CC2531 - it has served it's purpose in testing out zigbee on the cheap.

closed 2021-03-16 by github-actions[bot]

snowgumrd, 2021-11-09:
+1 for the same issue here. paired fine but no mqtt messages from actions.
I tried pairing it closer to the coordinator and for a brief moment I saw messages.

Hypothesis is that the repeaters don't carry the messages, or the co-ordinator doesn't understand it when it's via a repeater?

epicentrm, 2021-11-09:
the same issue, no messages from actions

WouterKr, 2021-11-26:
Hi, having the same issue. Bought one of these recently but besides pairing it does not do anything. 
Do not see any messages appearing in the debug log.

gohlas, 2021-12-12:
I just bough one of these and I'm not even able to pair it. 
When I try to pair it does not show anything in the logs, but the blinking led lit meaning that is in pairing mode.
Any tips?

henripalmroth, 2021-12-12:
Having the same issue as well. Conbee 2 stick and latest zigbee2mqtt.

stefancvetkovic, 2021-12-13:
+1 
Im using usb zigbee stick cc2651 and latest zigbee2mqtt docker container.
Pairing ok, but in the logs I cant see nothing when I click/rotate.

lcx, 2021-12-22:
I'll join in on this with a +1

makefu, 2022-01-05:
i am facing the same issue

elbarsal, 2022-01-17:
Maybe a bit more information - for me it paired, and worked nicely - in the map it showed directly connected to the controller. However - just yesterday it stopped working, and now it shows routing through a router (another Sonoff USB 3.0 stick flashed as a router).

I don't know how to retrieve any debugging information from the router - does anybody know if that is possible?

al3x1337, 2022-01-18:
I have the same issue now with Zigbee2Mqtt Edge and a cc2652p
I've force deleted and repaired but it does not detect any actions. Had it working previously.

Marcussjogren, 2022-01-19:
I also have the same issue, came here looking for a solution.

Ikea Symfonisk which pairs good, I can see battery status, signal strength etc. It also shows that it exposes "actions, like button", so the unit seems OK in that sense.

I have tried adding the legacy-attribute with no luck.

makefu, 2022-01-23:
Re-Pairing a couple of times close to the coordinator helped in my case, i assume that the messages get lost when sent via a bridge (e.g. any light bulb, AC Switch or gateway which may be part of your network).

lukaszzyla, 2022-01-27:
same issue here. also battery indicator behaviour is strange. i put a new bateery inside and soon after it shows 16%. also not possible to update. readding helps for a short while..

Marcussjogren, 2022-01-27:
Hi again,

it seems like I got my issue resolved yesterday but I’m not 100% sure of why.
But I _think_ the resolution was to do the following:

- Unplug the USB-stick for a few minutes
- Plug it back in
- Restart Zigbee2MQTT
- Pair the Symfonisk very close to the controller.

I now receive clicks and rotations again.

adammartin1981, 2023-01-11:
Maybe try a diff mqtt channel? (bit late to the party I know) - I was running on 11 - with intermittant issues, drop outs etc - changed to 15 - and touch wood, so far no issues.

It might be that mqtt is clashing with the wifi - and the E1744's small battery can't 'shout' loud enough over the wifi if they're on similar channels?

machv, 2023-01-11:
I had the same issue recently and surprisingly in my situation helped pairing E1744 to another ZigBee network and then back again to my primary one, since then (~2 weeks ago) it works fine.

iw1gfv, 2023-07-09:
I had the same issue 
I solved with a rework with hot air in the metal box on the second pcb

I also re soldered all the components on the main pcb.

Semmu, 2024-07-25:
hello everyone,

im facing this issue as well.

i have not been using my E1744 rotary controller for a while and now it does not seem to work correctly. i can pair it to my network without issues, it even reports healthcheck pings periodically, but no other messages are received by it no matter what i do. so the device is useless unfortunately.

* my zigbee network is on channel 11
* i repaired it countless times, near and far of my zigbee controller
* i tried it with legacy mode enabled and disabled as well
* i enabled debug logging for zigbee2mqtt and it shows nothing when clicking and rotating the button
* i even disassembled it and checked if it has any hardware issues, but i dont see anything problematic hardware-wise

here are the images i took of the PCB, maybe your eagle eyes will notice something i did not:

[image]
[image]
[image]

i would appreciate any tips if you have!

Semmu, 2024-07-25:
i just realized zigbee2mqtt offers an OTA update for the device, but when i try to update the firmware the button does not respond to the OTA request. any tips regarding this?

Semmu, 2024-07-25:
sorry for the spam, but in the end i managed to solve my problem! sharing the details in case anyone else faces the same issue:

* i replaced the battery within the button to a brand new one
* i had to try the OTA update multiple times and move the button very close to the controller for it to start working
* after it finally managed to update the firmware (took almost 2hrs) it started working fully again! i receive all the button press and rotation messages.

i expected this to be a hardware issue, as everything related to the rotary encoder was missing, but for some reason it was a firmware problem.

hope it helps anyone!