Koenkk/zigbee2mqtt #4037: [Q] BW-SHP13 Eu Plug as Router and Xiaomi Sensor Compatibility ?

fluffymadness, 2020-08-04:
Not a bug but more of a general compatibility question.
Not really sure where else I could post this

https://www.zigbee2mqtt.io/information/supported_devices.html#blitzwolf-1
Basically I'm interested in buying some of these because basically they are a very good deal:
- 16A
- Zigbee 3.0
- Compactly Built
- Cheap

The thing is I have mainly Xiaomi Devices, and there are some known incompatibilities with sensors and certain zigbee routers.
Some routers seem to send some kind of wakeup requests and Xiaomi sensors will drop off the network then.
Described in the docs where there is also a link to some compatible routers.
https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html

Does someone already have some of these plugs and can confirm that there is 
a) some kind of connection shown in the zigbee mesh (sensor not floating in the mesh, i have xiaomi buttons that are floating in the mesh, no matter what I do, but they are still working. I think they are probably connected to my Heiman Zigbee plug)

b) that sensors don't drop off the network after some time when paired with a BW-SHP13

Maybe we could make a general issue for Xiaomi Router Compatibility, where the content could be put in the zigbee2mqtt docs sooner or later.
I think it would probably help a lot of people from buying incompatible hardware for their mesh.

closed 2020-09-11 by github-actions[bot]

pedroke, 2022-04-21:
@fluffymadness Have you found if the BWSHP13 is compatible as a router for Xiaomi end devices? I also miss somewhere the compatibility list which would help to decide which router-end-device combinations to chose.. Thanks.

fluffymadness, 2022-04-24:
@pedroke I have them in my network with aqara buttons and aqara motion sensor connected to them. Haven't noticed any issues of dropped devices so far

pedroke, 2022-04-24:
@fluffymadness Thanks for info. I ordered one for tests, so I'm looking forward :) Btw I started discussion on this topic here https://github.com/Koenkk/zigbee2mqtt/discussions/12237#discussioncomment-2622137
Maybe we'll get some up to date list of compatible devices.. So far it looks promissing.