Koenkk/zigbee2mqtt #4038: Danalock is not working properly after it is connected to two CC2531 routers

elbueno222, 2020-08-04:
# Bug Report
## What happened

I have been using Danalock for 3 months now and it has been working fine being connected to one of my CC2531 routers (specially after I changed the firmware of my CC2531 coordinator to source_routing), but last week I switched off that router for an hour or so, and after I switched the router on again I realized Danalock had connected to the same router and also to another one. From that moment I am having intermittent issues where Danalock stops sending messages with the lock status and I have to manually send the "zigbee2mqtt/bridge/configure" message to fix it. And even this command sometimes fails, this is one example of the message I received when the command failed (I sent commands configure and get and both commands failed):

```
zigbee2mqtt:error 2020-08-03 23:33:27: Failed to configure 'DEVICE_ID_REMOVED', attempt 9 (Error: Bind DEVICE_ID_REMOVED/1 closuresDoorLock from '0x00124b0018e26fba/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

zigbee2mqtt:error 2020-08-03 23:33:50: Publish 'get' 'state' to 'DEVICE_ID_REMOVED' failed: 'Error: Read DEVICE_ID_REMOVED/1 closuresDoorLock(["lockState"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 53636 - 1 - 142 - 257 - 1 after 10000ms)'
zigbee2mqtt:info  2020-08-03 23:33:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'get' 'state' to 'DEVICE_ID_REMOVED' failed: 'Error: Read DEVICE_ID_REMOVED/1 closuresDoorLock([\"lockState\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 53636 - 1 - 142 - 257 - 1 after 10000ms)'","meta":{"friendly_name":"DEVICE_ID_REMOVED"}}'
```

This is the zigbee map where I can see Danalock connected to two routers (as mentioned above, before it was connected just to one of them and it was working fine):

[image]

How can I fix this issue? Is there any way to avoid devices to connect to two routers at the same time?

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_20190619

sjorge, 2020-08-04:
That looks normal, zigbee is a mesh so each device will connect to multiple nearest neighbors. If you generate a net map including route you will see which path back to the coordinator it uses.

elbueno222, 2020-08-05:
I am not sure if an endpoint connected to two routers is normal (I guess it is, because as you can see in the picture I have another device connected to two routers and it is working fine), but in this case the device is not working as expected, and even the "zigbee2mqtt/bridge/configure" message that never failed before (when the device was connected to one router) is failing now (it is not failing every time, it fails just sometimes so it is like a trial and error)

Koenkk, 2020-08-05:
Can you provide a sniff of the failing configure while sniffing at the location of the danalock? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

elbueno222, 2020-08-06:
> Can you provide a sniff of the failing configure while sniffing at the location of the danalock? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I am trying to sniff the messages but there are too many, is there a way to filter the messages? I am using zbee.sec.key to filter the messages but I am still seeing hundreds of messages per minute so I can't find the specific message from the failure, is there any other field I can use to filter?

elbueno222, 2020-08-06:
> Can you provide a sniff of the failing configure while sniffing at the location of the danalock? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Ok, it won't be easy as the error happens every 1-2 days at random times but I will try. Should I filter the messages using another filter apart from zbee.sec.key?

elbueno222, 2020-08-06:
> Can you provide a sniff of the failing configure while sniffing at the location of the danalock? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I got lucky, the error just happened and I was able to sniff 5 different errors (4 of them after sending message `zigbee2mqtt/bridge/configure` and 1 more after sending `zigbee2mqtt/[device_id]/get`). In all 5 cases I cannot see any message sent by Danalock (I use the filter `(zbee_nwk.src64 == [Danalock_address])`, but I don't really know what I should be looking for, how can I send you the logs?

Koenkk, 2020-08-08:
@elbueno222 on telegram @koenkk

elbueno222, 2020-08-09:
> @elbueno222 on telegram @Koenkk

Done

Koenkk, 2020-08-09:
I've investigated your sniff, the problem is the following. In Zigbee the parent (= router in this case, could also be the coordinator itself) is responsible for holding the message for the end device, when the end device does a "Data request" to the parent it should send any messages that it holds.

Zigbee2MQTT sends the bind request to the lock with the destination being 0x8444 (this should be the parent of the lock)

[image]

However the lock considers 0x96f8 as it's parent, not 0x8444!

[image]

This is similar to https://github.com/Koenkk/zigbee2mqtt/issues/3870 and I've posted a possible fix here: https://github.com/Koenkk/zigbee2mqtt/issues/3870#issuecomment-669801601 , please let me know if this fixes it for you (you can skip the adapter firmware part).

elbueno222, 2020-08-09:
I'm running zigbee2mqtt Hass.io Add-on, can I follow those instructions or are they just valid for bare metal?

Koenkk, 2020-08-10:
@elbueno222 since there are a lot of changes (to many files) I would recommend waiting until wednesday. After the z2m release I will merge these change so they will be available in the edge addon after this.

elbueno222, 2020-08-10:
Sure, more than happy to wait, just let me know when the changes are available in the edge version and I will test it.

Koenkk, 2020-08-12:
Fix will be in the edge addon +- 2 hours from now

elbueno222, 2020-08-12:
Edge version enabled, I will let you know if the issue happens again during the next days

elbueno222, 2020-08-13:
Unfortunately after uninstalling and installing the edge version to get the latest changes the error happened again. I will try to sniff the messages when the next error happens.

elbueno222, 2020-08-14:
The error has happened again, but I haven't been able to prepare the sniff because the behaviour is different. Previously just after the error happened once I was able to reproduce the error sending the configure message as many times as I needed during some time (at least 30 minutes or so until the error was fixed automatically and Danalock started to respond to messages), but now even the configure message reached the 10 seconds timeout and failed, the device was actually fixed around 13 seconds later. I know that because my automation requests the lock status 10 seconds after sending the configure message (**22:48:37** in this case), and this time Danalock replied with the status at 22:48:50 (13 seconds later). This is the zigbee2mqtt-edge log:

```
Zigbee2MQTT:info  2020-08-14 **22:48:27**: Configuring 'DANALOCK_ID'
Zigbee2MQTT:error 2020-08-14 **22:48:37**: Failed to configure 'DANALOCK_ID', attempt 3 (Error: Bind DANALOCK_ID/1 closuresDoorLock from '0x00124b0018e26fba/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
Zigbee2MQTT:info  2020-08-14 **22:48:50**: MQTT publish: topic 'zigbee2mqtt/DANALOCK_ID', payload '{"linkquality":55,"state":"UNLOCK","lock_state":"unlocked","battery":89}'
```

I have disabled my automation (the one that sends the configure message and request the lock status every time Danalock is over an hour without sending the status) to try and sniff the messages next time the error happens.

elbueno222, 2020-08-16:
@Koenkk the error happened again but causing a worse situation as Danalock has not responded to messages during the last 12 hours, I have sent you the sniff on Telegram

Koenkk, 2020-08-18:
@elbueno222 checked your sniff again. 

As the first requests fails, Zigbee2MQTT tries to discover the route again:

[image]

However as can be seen, 2 devices respond to it! The old `0xd814` and new `0x96f8`. At this point you have to be lucky for the coordinator to pick the correct one.

@ptvoinfo do you know if child ageing is enabled for the router firmwares? This should prevent such issues.

elbueno222, 2020-08-19:
I guess this issue would be solved if the device was connected to the coordinator just through one router, but why is the device connected to the coordinator through two different routers at the same time? is this expected in a zigbee network? If the end device is holding information about the router/coordinator it is connected to, and it can hold information about just one router/coordinator, shouldn´t the zigbee network remove the old connections (if that is possible) and keep just one route or path from every device to the coordinator to avoid this type of issue?

Koenkk, 2020-08-19:
@elbueno222 and end device can only be connected to one router (two connections is impossible). The problem is that 2 routers think it's connected to it, leading to a wrong route. Therefore my question about child ageing in https://github.com/Koenkk/zigbee2mqtt/issues/4038#issuecomment-675694507

elbueno222, 2020-08-19:
> @elbueno222 and end device can only be connected to one router (two connections is impossible). The problem is that 2 routers think it's connected to it, leading to a wrong route. Therefore my question about child ageing in [#4038 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4038#issuecomment-675694507)

Thank you for the clarification, it makes complete sense

elbueno222, 2020-08-24:
@Koenkk, I asked @ptvoinfo about this on his website (https://ptvo.info/cc2531-based-router-firmware-136/) and according to his comment (Posted at 12:15 August 21, 2020) he seems to believe the path is stored on the coordinator side, do you think the routers are also storing the information about the end devices connected to them?

I guess as a workaround I could reset the routers and/or coordinator so the problematic double connection disappears, but I have seen other people having issues with different devices due to these double connections, and I would like to find a solution to the issue so nobody else has this problem in the future.

Koenkk, 2020-08-24:
@elbueno222 the path is stored on the router side, you can verify it by shortly turning of the "invalid" parent router. After that I expect it to work.

elbueno222, 2020-08-24:
@Koenkk when the coordinator see two routers pointing to the same end device, is there any way for the coordinator to know what router is sending the wrong path (i.e. by checking last successful connection with the device or something like that) in order to ignore/remove that path?

Koenkk, 2020-08-25:
@elbueno222 I was thinking something similar, but z-stack does not expose such commands externally. This still would be a hack rather then a fix, since other devices in the network still have 2 routes to chose between. The correct solution would be to enable child aging for the routers.

@ptvoinfo I've checked this and it can easily be done, just change `uint8 zgChildAgingEnable = FALSE;` to `uint8 zgChildAgingEnable = TRUE;` in `Components/stack/sys/ZGlobals.c`.

ptvoinfo, 2020-08-26:
@Koenkk What about `zgNwkParentInformation` and other related options in the same file?

Koenkk, 2020-08-26:
@ptvoinfo I don't think you need to change anything else.

@elbueno222 do you have Xiaomi devices on your network? (if yes, are they connected to the CC2530 router?)

elbueno222, 2020-08-26:
My routers are CC2531 (not CC2530) but yes, most of my devices (21 out of 24) are Xiaomi, and many of them are connected to the routers (one of them is actually connected to both routers now, but for some reason it doesn´t cause issues as with the Danalock):

[image]

ptvoinfo, 2020-08-26:
@Koenkk I saw that the default aging time is 8 minutes. Is it ok for the most end-devices?

ptvoinfo, 2020-08-26:
@elbueno222 Please, try the following firmware. I did the changes suggested by @Koenkk 
https://ptvo.info/download/cc2530_1.2.2a.44539_firmware_20200826.zip

Koenkk, 2020-08-26:
@ptvoinfo we need to test! Since @elbueno222 has a lot of Xiaomi devices, it could be that they are kicked of the network now, let's see what happens.

elbueno222, 2020-08-26:
Sure, I will install the firmware in both routers and will let you know if it fix the "double path" issue and also if it produces any new issues with my devices.

Just FYI, with the current firmware all my Xiaomi devices work properly, except a Xiaomi leak sensor that sometimes is disconnected from the network (around once a month or so)

ptvoinfo, 2020-08-26:
@elbueno222 Please, note that the map may contain duplicate path for up to 256 minutes (about 4 hours).

elbueno222, 2020-08-26:
Thanks @ptvoinfo I am aware of that.

Just as a quick note, writing the original router-cc2531-std.hex firmware the process reported 12176 lines read and 96 pages:

´´´
  ID = b524.
  reading line 12170.
  file loaded (12176 lines read).
writing page  96/ 96.
verifying page  96/ 96.
 flash OK.
´´´

And with this new firmware it reported 12073 lines read and 95 pages:

´´´
  ID = b524.
  reading line 12070.
  file loaded (12073 lines read).
writing page  95/ 95.
verifying page  95/ 95.
 flash OK.
´´´

elbueno222, 2020-08-26:
I have bad news, after flashing the new firmware in the two CC2531 they cannot join the network. I have enabled the option to allow new devices to join, but they are both doing short fast blinks (one per second) and they are not discovered by the network.

elbueno222, 2020-08-27:
Yesterday I reconnected all the devices to the network (most of the required to hold the reset button for 5 seconds, of course I didn´t do anything to the devices that were connected to the other Ikea router I have). I checked the zigbee map and around half of the devices connected to each router, so every device was connected to one of the routers and it looked fine (no double connections, but that was expected as the double connections happens when I turn off one of the routers).

Unfortunately today all the Xiaomi devices appear as not connected to the network, but actually they work (I guess they can still find the path to the coordinator through the routers, even the routers think they are not connected through them?)

[image]

If the "timeout" is 256 minutes that shouldn´t have happened, as I believe most of the Xiaomi devices send data every hour or so. I have executed this command to check the last time every device sent data:

```
{%- set ns = namespace(time=0,device="") %}
{% for state in states.sensor if 'linkquality' in state.entity_id -%}
  {%- set temp = (((as_timestamp (now()) - as_timestamp (state.last_updated)) / 60) | round(0)) %}
  {{state.name}} sent data {{temp}} minutes ago 
{%- endfor %}
```

And this is the result:

```
  0x680ae2fffe3f6571_linkquality sent data 21 minutes ago
  Aqara Cube Linkquality sent data 172 minutes ago
  Bedroom Button Linkquality sent data 42 minutes ago
  Bedrom Window1 Sensor Linkquality sent data 35 minutes ago
  CC2531 Router1 Linkquality sent data 2 minutes ago
  CC2531 Router2 Linkquality sent data 49 minutes ago
  Cloakroom Motion Sensor Linkquality sent data 9 minutes ago
  Danalock Main Door Linkquality sent data 62 minutes ago
  Downstairs Motion Sensor Linkquality sent data 49 minutes ago
  Garden Door Sensor Linkquality sent data 59 minutes ago
  Honeywell Smoke Sensor Linkquality sent data 246 minutes ago
  Hot Water Thermostat Linkquality sent data 6 minutes ago
  Ikea Blind Linkquality sent data 586 minutes ago
  Ikea Open Closer Button Linkquality sent data 1008 minutes ago
  Kitchen Button Linkquality sent data 50 minutes ago
  Kitchen Garden Sensor Linkquality sent data 726 minutes ago
  Kitchen Motion Sensor Linkquality sent data 27 minutes ago
  Kitchen Temperature Sensor Linkquality sent data 6 minutes ago
  Kitchen Window1 Sensor Linkquality sent data 32 minutes ago
  Kitchen Window2 Sensor Linkquality sent data 605 minutes ago
  Leak Sensor Linkquality sent data 40 minutes ago
  Living Room Button Linkquality sent data 713 minutes ago
  Living Room Temperature Sensor Linkquality sent data 41 minutes ago
  Living Room Window1 Sensor Linkquality sent data 18 minutes ago
  Living Room Window2 Sensor Linkquality sent data 24 minutes ago
  Main Door Sensor Linkquality sent data 4 minutes ago
  Upstairs Sensor Linkquality sent data 1 minutes ago
```

As you can see most of the devices has sent data in the last hour, so why have they been disconnected?

ptvoinfo, 2020-08-27:
FYI: my Xiaomi Door sensors may sleep 6 hours and more.

elbueno222, 2020-08-27:
From Xiaomi I have door sensors, motion sensors, one leak sensor and one smoke sensor. I agree that some devices may sleep over 4 hours sometimes, but I don´t think every single device I have was sleeping for more than 4 hours overnight, but all of them were disconnected in the morning. Actually they are still disconnected as I haven't pressed the reset button in any of them, but as I mentioned before they are working.

elbueno222, 2020-08-28:
Same story today just with a small difference, 2 of my Xiaomi devices (one door sensor and the leak sensor) stayed connected overnight, but all the other Xiaomi devices got disconnected.

This was the map yesterday after I reconnected most of the devices:

[image]

And this is the map today (I have marked the 2 Xiaomi devices still connected):

[image]

@Koenkk it looks like this fix doesn´t work for Xiaomi devices, is there any other option? Ideally the fix should just affect the devices with double connection to avoid impacting devices connected just to one router.

ptvoinfo, 2020-08-28:
@elbueno222 Please, try another version. I've increased the timeout to 1024 minutes. 
https://ptvo.info/download/cc2531_1.2.2a.44539_firmware_20200828.zip

Koenkk, 2020-08-28:
@elbueno222 if they Xiaomi devices still work (in terms of sending data) I wouldn't worry to much.

elbueno222, 2020-08-28:
> @elbueno222 if they Xiaomi devices still work (in terms of sending data) I wouldn't worry to much.

They work, but they appear as disconnected in the zigbee map, what means none of the routers think they are connected to them, right? I actually don´t understand why they work

Koenkk, 2020-08-28:
@elbueno222 that's not what it means, it means that the parent is unknown, this does not mean they can't send data anymore! The problem we are trying to fix is that a router considers an end device  it's child while the child has already moved on to a different parent. To determine this the router now has a timeout, if the child hasn't checked in for an X amount of time it will not consider it a child anymore (the original router firmware had no timeout). Since Xiaomi devices sleep for very long time (longer than the timeout), none of the routers considers them as a child anymore. Knowing the parent of an end device is crucial when sending data to it, but since no data is send to Xiaomi  devices (besides while pairing), this doesn't matter.

elbueno222, 2020-08-28:
@Koenkk but then this could cause a problem for other devices that need to receive data in case they don´t send data for 4 hours (i.e. if Danalock doesn´t send data for 4 hours and the router doesn´t consider it a child anymore, it will be impossible to lock/unlock the door as the coordinator won´t be able to send data to Danalock). In that case I think a longer timeout as @ptvoinfo has suggested would be better, what do you think?

Koenkk, 2020-08-28:
@elbueno222 if a device would really sleep for 4 hours, you will only be able to send data when it is awake (once every 4 hours). This would of course not make sense for a lock. A router cannot "wakeup" a child/end device.

End devices that are "controllable" (battery powered locks, curtains, etc.) constantly check their parent if new data is available for them. You can see this in the sniffed traffic as a "Data request". 

The way sending data to battery powered/end devices works is that the coordinator sends the message to the router (parent) of the end device. The router then holds the message until the end device does a "Data request", when it does it will send the message to the end device.

elbueno222, 2020-08-28:
That make sense, in that case the only issue of this fix would be for the Xiaomi devices to appear as disconnected in the zigbee map as none of the routers will consider them their children, but they would still work as the devices will keep the information of the router they are connected to.

ptvoinfo, 2020-08-28:
@Koenkk If I'll release the router firmware to public, we'll get a lot of reports where devices are unconnected on the map :) Please, consider to keep the last known parent on the server side for the map only.

Koenkk, 2020-08-28:
But how to know the last parent?

ptvoinfo, 2020-08-28:
@Koenkk Build the map every 4 hours?

Koenkk, 2020-08-28:
@ptvoinfo regularly executing a network scan it not possible, it will put a lot of stress on the network making the network unusable for some time, not nice if you just wanted to turn on a bulb.

I think it's better to provide a good explanation why it happens and what the reason for it is (I can add this to the zigbee2mqtt docs) as this is just how Zigbee works. Note that many users already experience this since devices like Hue, OSRAM etc already have this timeout.

elbueno222, 2020-08-28:
I just found an unexpected behavior, I still have the latest firmware with timeout = 1024 (cc2531_1.2.2a.44539_firmware_20200828.zip) in both CC2531 routers, but to perform a better test I disconnected the Ikea router so all the devices are connected to the CC2531 routers with the new firmware, and for some reason Danalock gets disconnected after one hour and I cannot send lock/unlock commands. After it was disconnected the first time I reconnected it, and it got disconnected again.

This is the map after I disconnected the Ikea router with all devices connected to the CC2531 routers (danalock in a blue circle):

[image]

And this is aprox. one hour later:

[image]

Also the other device that was connected to both routers (Ikea blind) is now connected to one router after just one hour, when it should have been connected to both router for 1024 minutes (17 hours).

Something is not working as expected with the new option, are you guys sure the timeout is define in minutes? (i.e. if it was defined in seconds, that might explain this behavior).

sjorge, 2020-08-28:
> I think it's better to provide a good explanation why it happens and what the reason for it is (I can add this to the zigbee2mqtt docs) as this is just how Zigbee works. Note that many users already experience this since devices like Hue, OSRAM etc already have this timeout.

If this is going to be documented, will this allow us to rentable child aging in the coordinator firmware’s too? IIRC it is currently disabled for this.

Koenkk, 2020-08-28:
@elbueno222 but does the lock still work as expected?

@sjorge when I've tested this with CC2652 it caused the Xiaomi devices to be completely removed from the network (not sending data anymore), however when this happens to the coordinator Zigbee2MQTT is able to recover it, so enabling this won't fix anything (as far as I know now).

elbueno222, 2020-08-28:
@Koenkk Danalock doesn´t work as expected, after it was disconnected it doesn´t send the lock status every hour as usual, and if I send lock/unlock commands they can´t reach the lock (I can see the timeout error in the log), it is completely disconnected from the network.

Koenkk, 2020-08-29:
could you sniff the traffic again (close to the lock)?

elbueno222, 2020-08-29:
Done, I just sent you the sniff

Koenkk, 2020-08-29:
@ptvoinfo can you try changing `uint8 zgNwkParentInformation = NWK_PARENT_INFO_UNDEFINED;` to `uint8 zgNwkParentInformation = NWK_PARENT_INFO_MAC_DATA_POLL;`

ptvoinfo, 2020-08-31:
@Koenkk, @elbueno222 Done. Could you please try? 
https://ptvo.info/download/cc2531_1.2.2a.44539_firmware_20200831.zip

elbueno222, 2020-08-31:
I flashed both routers with the new firmware but now they are not connecting to the network. This already happened before so I will keep trying to connect them, but there might be some issue in the router firmware as sometimes they can't connect to the network and I have to reset them multiple times until they finally connect

elbueno222, 2020-08-31:
@Koenkk @ptvoinfo unfortunately with the latest firmware Danalock still gets disconnected in the zigbee map and it cannot received messages. I have created an automation to know how long it takes for Danalock to disconnect by request the lock status every minute, and the lock stops responding 10 minutes after it has joined the network (I have tested twice, and in both times it took 10 minutes to stop responding).

@Koenkk something that I don't understand is why Danalock gets disconnected after 10 minutes but the rest of my devices stay connected. I thought the reason could be Danalock is an "active" device (i.e. it is sending data request every 5 seconds) but another of my devices is an ikea blind which is also an "active" device (sending data request every second), and this ikea blind doesn't get disconnected with the new firmware, why do you think this happens?

[29 more replies]

closed 2020-09-29 by elbueno222