Koenkk/zigbee2mqtt #4039: conbee II adapter, unusual link quality

zhapwey, 2020-08-04:
I have been testing the conbee II and z2m adapter for a few days, the result so far has been very good, all devices continue to function properly.

I only find one problem, the link quality reports of the devices are wrong and I receive from all the devices the same link quality

Checking the network map of the devices, the link quality is not really real either but it varies somewhat more logically, do you know what it can be?

zigbee2mqtt/0x00158d0002bfe954 {"battery":97,"temperature":29.45,"humidity":48.77,"pressure":936.1,"linkquality":176}
zigbee2mqtt/0x00158d00032b68fe {"battery":100,"contact":true,"linkquality":176}
zigbee2mqtt/0x00158d000313e324 {"battery":100,"illuminance":7,"illuminance_lux":7,"linkquality":176,"occupancy":false}
zigbee2mqtt/0x04cf8cdf3c764b41 {"state":"ON","power":0,"current":0,"consumption":1.25,"temperature":33,"linkquality":176}
zigbee2mqtt/0xccccccfffe7c9356 {"occupied_heating_setpoint":28,"linkquality":176,"pi_heating_demand":7,"operation":"heating","local_temperature":30.99,"battery":74,"danfoss_trv_mode":1}
zigbee2mqtt/0x000d6ffffe009262 {"update_available":true,"linkquality":175,"battery":34}

Of these devices some are 2 meters from the coordinator and others at almost 50, all work correctly, but their link quality cannot be the same.

I am quite happy with the result and scope of this device, I hope they continue with their development because it seems like great hardware.

## Debug Info
Zigbee2mqtt version:1.14.2
Adapter hardware: conbee II
Adapter firmware version: deCONZ_ConBeeII_0x264a0700.bin.GCF

bkupidura, 2020-08-19:
Hey,
Probably im seeing same behavior.

ConbeeII firmware: 26580700
Z2M: 1.14.3

Im using node-red to get avg lqi from network, suddenly without any good reason my sensors starts reporting same lqi (~5).

Network works stable, everything is fine except this bizarre lqi readings.

Will observe and confirm if everything else is fine.

[image]
[image]

ChrisHae, 2020-09-02:
I'm quite busy right now but will have a look into this problem (if it lies in the deconz zigbee-herdsman code) when I have time.

bkupidura, 2020-09-02:
As a update, looks like sensors reports same LQI as other sensors. This works a snow ball.
Sensor_A reports LQI 100, Sensor_B reports wrong LQI copied from Sensor_A, Sensor_C reports wrong LQI copied from sensor_B, ...

In my environment they are going up and down...

[image]

andriej, 2020-09-02:
Same here (conbee2 + z2m latest stable)

thackel, 2020-09-06:
I also experience the same odd behavior or devices added to a Conbee2.
I am migrating from a CC2531 to a Conbee2 and having two containers with Zigbee2mqtt (1.14.4-dev (commit #c9094e9)) running.

The link quality for devices paired to the conbee2 seem to be summarized instead being an absolute value:

[image]

The two (blue/yellow) lines are two Xioami Temperature sensors paired to the Conbee2, the other values are from devices paired to a CC2531 where the values are OK.

`bash
grep linkquality log.txt 
info  2020-09-05 18:36:00: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '39'
info  2020-09-05 18:36:00: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '38'
info  2020-09-05 19:02:48: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '49'
info  2020-09-05 19:09:25: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '51'
info  2020-09-05 19:09:25: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '51'
info  2020-09-05 19:09:25: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '51'
info  2020-09-05 19:13:11: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '53'
info  2020-09-05 19:13:11: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '53'
info  2020-09-05 19:13:11: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '53'
info  2020-09-05 19:19:08: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '54'
info  2020-09-05 19:44:34: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '62'
info  2020-09-05 19:44:34: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '62'
info  2020-09-05 19:44:34: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '62'
info  2020-09-05 19:54:40: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '65'
info  2020-09-05 19:54:40: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '65'
info  2020-09-05 19:54:40: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '65'
info  2020-09-05 19:58:21: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '66'
info  2020-09-05 20:13:00: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '70'
info  2020-09-05 20:13:00: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '70'
info  2020-09-05 20:13:01: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '70'
info  2020-09-05 20:14:51: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '70'
info  2020-09-05 20:38:01: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '77'
info  2020-09-05 20:38:01: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '77'
info  2020-09-05 20:38:01: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '77'
info  2020-09-05 20:46:39: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '79'
info  2020-09-05 20:46:39: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '79'
info  2020-09-05 20:46:39: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '79'
info  2020-09-05 20:53:44: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '81'
info  2020-09-05 21:11:35: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '87'
info  2020-09-05 21:20:21: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '89'
info  2020-09-05 21:20:21: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '89'
info  2020-09-05 21:20:21: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '89'
info  2020-09-05 21:49:18: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '98'
info  2020-09-05 21:50:54: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '98'
info  2020-09-05 21:50:54: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '98'
info  2020-09-05 21:50:54: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '98'
info  2020-09-05 22:08:19: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '103'
info  2020-09-05 22:42:00: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '113'
info  2020-09-05 23:07:03: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '120'
info  2020-09-05 23:36:03: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '128'
info  2020-09-05 23:50:33: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '132'
info  2020-09-05 23:50:33: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '132'
info  2020-09-05 23:50:33: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '132'
info  2020-09-06 00:04:17: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '136'
info  2020-09-06 00:30:36: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '144'
info  2020-09-06 00:54:49: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '150'
info  2020-09-06 00:54:49: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '150'
info  2020-09-06 00:54:49: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '150'
info  2020-09-06 01:04:02: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '153'
info  2020-09-06 01:16:37: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '157'
info  2020-09-06 01:16:37: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '157'
info  2020-09-06 01:16:37: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '157'
[... cut ...]
info  2020-09-06 07:14:54: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '3'
info  2020-09-06 07:28:52: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '7'
info  2020-09-06 07:47:02: MQTT publish: topic 'zigbee2mqtt/temp04/linkquality', payload '12'
info  2020-09-06 07:58:40: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '15'
info  2020-09-06 07:58:40: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '15'
info  2020-09-06 07:58:40: MQTT publish: topic 'zigbee2mqtt/temp02/linkquality', payload '15'
`

ChrisHae, 2020-09-15:
It seems this is a firmware bug. It is beeing researched by developers right now. This error only appears in the lqi values of incoming data indications. A workaround is to use lqi requests. They show the correct lqi values. With zigbee2mqtt you can start a network scan with: zigbee2mqtt/bridge/networkmap. It sends lqi requests to all devices so this can take a while depending on your network size.

liviomarti, 2020-10-13:
I'm having the same issue.
Any news about if it's really an issue of the Conbee2 firmware?
If that's the case, are the reported values within Deconz also wrong?

Conbee2 Firmware: 26660700
Z2M: 1.15.0 (ed8b4e5)

s3frank, 2020-10-14:
yeah thanks Koen for tracking down this existing bug report. I agree with you this is identical to what I reported.
Let me know how I can help with testing.

Frank

ChrisHae, 2020-10-14:
@liviomarti deCONZ uses LQI requests to display LQI values. These are the correct values. You can also display them with z2mqtt when you start a network scan.

s3frank, 2020-10-14:
ok, not sure I understand where you are saying the bug is then.
deconz is using LQI requests to the firmware on the Conbee II stick I guess.
Does that mean that z2mqtt uses some other call or logic for the same
unless network scanning?

ChrisHae, 2020-10-14:
The bug is in the FW. and only appears on the LQI value that is attached to the received data indications. That means when devices send messages to the Coordinator. Somehow the wrong values are attached to the data packets. When starting a network scan the Coordinator sends LQI requests to all devices and they respond with the correct values.

s3frank, 2020-10-31:
ok I think I understand it now. Can you point me to where I can track and apply updated FW once it is out? I know where to find firmware but not how to track when this bug is fixed. Is it even known / reported with the conbee II folks?

ChrisHae, 2020-11-03:
deCONZ Firmware and Software changelogs are documented here:https://github.com/dresden-elektronik/deconz-rest-plugin/releases

Updateing FW is descriped here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
This manual also describes how to update deCONZ software. Look for "Flash deCONZ firmware" for the update FW steps.

kwetnico, 2020-11-23:
> deCONZ Firmware and Software changelogs are documented here:https://github.com/dresden-elektronik/deconz-rest-plugin/releases
>
> Updateing FW is descriped here: [...]

Hello @ChrisHae do you have an idea to find a solution on firmware ConbeeII with LQI ?

s3frank, 2020-11-23:
I updated last week but no fix yet

-FF

Sent from my mobile, powered by Frank's thumbs!

kwetnico, 2020-11-23:
> I updated last week but no fix yet
> […](#)
> -FF Sent from my mobile, powered by Frank's thumbs!

Me too

kwetnico, 2020-11-23:
> I updated last week but no fix yet
> […](#)
Me too

kwetnico, 2020-11-28:
> deCONZ Firmware and Software changelogs are documented here:https://github.com/dresden-elektronik/deconz-rest-plugin/releases
>
> Updateing FW is descriped here: [...]

Hello no news ?

zhapwey, 2020-11-30:
No, the linkquality problem remains the same, I have also noticed that the network map also has many lost links, I imagine that @ChrisHae  is quite busy.

ChrisHae, 2020-12-01:
LQI problem is not in my hands and will be fixed with future firmware release. Network map has lost links when enddevices are sleeping.

andriej, 2020-12-01:
@ChrisHae is there any roadmap on which release it might be 'fixed'? 
I didn't see changelog for the latest release of conbee 2 firmware (from 18/11) yet too

ChrisHae, 2020-12-01:
The issue is known by developers but not sheduled yet as far as I know. Perhaps ask the devs in the appropriate device issues discord channel could help. https://discord.com/channels/494922323518947329/692306044495593544

kwetnico, 2020-12-01:
> LQI problem is not in my hands and will be fixed with future firmware release. Network map has lost links when enddevices are sleeping.

Thank you for your answer, it's a shame for the links since we know that the xiami peripherals are sleeping. Despite this, there is no such problem with a CC2531 key.
Waiting for new firmware with LQI correction.

spaxxilein, 2020-12-01:
> @ChrisHae is there any roadmap on which release it might be 'fixed'?
> I didn't see changelog for the latest release of conbee 2 firmware (from 18/11) yet too

There is actually a new firmware released yesterday:

http://deconz.dresden-elektronik.de/deconz-firmware/

0x26680700

I will try if that changes anything regarding signal quality later when i am at home.

kwetnico, 2020-12-02:
> There is actually a new firmware released yesterday:
>
> http://deconz.dresden-elektronik.de/deconz-firmware/
>
> 0x26680700
>
> I will try if that changes anything regarding signal quality later when i am [...]

with the new firmware always the same problems

kwetnico, 2020-12-02:
> The issue is known by developers but not sheduled yet as far as I know. Perhaps ask the devs in the appropriate device issues discord channel could help. https://discord.com/channels/494922323518947329/692306044495593544

I cannot connect to the discord room; he tells me he doesn't exist ..

kwetnico, 2020-12-05:
> The issue is known by developers but not sheduled yet as far as I know. Perhaps ask the devs in the appropriate device issues discord channel could help. https://discord.com/channels/494922323518947329/692306044495593544

Since the last update, the problem with LQI is still present but the links on the map are not displayed on all devices but it is better with the latest firmware version of November 30
[image]

kwetnico, 2021-01-05:
Always problem with LQI for conbee 2 with z2m

m0wlheld, 2021-01-19:
Same for the Raspee II with most recent firmware ([0x26690700](https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26690700.bin.GCF), Dec. 19th 2020). LQI increases to 255, then drops off to zero.

m0wlheld, 2021-01-19:
Is there an issue report at deCONZ/Dresden Elektronik repository to track? Couldn't find none ...

kwetnico, 2021-01-19:
> Is there an issue report at deCONZ/Dresden Elektronik repository to track? Couldn't find none ...

Me too :(

zhapwey, 2021-01-19:
the lqi problem in individual messages remains the same. but i noticed that the netwokmap shows all the links.
what I don't know if the link quality in the networkmap is reliable. but if so, the coverage of conbee II is far superior to that of the texas red plate

andriej, 2021-01-19:
It is not. I'm testing few of the sticks and methods, right now have cc2652p on the same mesh that I had with conbee and LQI values seems 'reliable' - those from Conbee were not at all.

m0wlheld, 2021-01-19:
Cannot comment on this, as the Raspbee II is the first and only bridge device I own. Currently waiting for a CC2652RB from slaeh's, but DHL is giving me a hard time ...

kwetnico, 2021-01-19:
I have a cc2531 which works well, the LQI and the card are good but the device limit and the range are much better with the Conbee2 under z2m

s3frank, 2021-01-19:
I don't really care to be honest. I have put a bunch of ikea repeaters
throughout my home and then connected the rest up.
Haven't had a link problem anywhere with 54 devices across my house now and
response times are subsecond in the worst case scenario.

So while it annoys me that these values are wrong I don't really care for
day to day use.

andriej, 2021-01-19:
While you have quite big network with bunch of repeaters, there are still users that would like to debug their connectivity problems and please don't cut them out of the stick just because you don't have problems. :)

s3frank, 2021-01-19:
Sorry didn't mean to do that. Just wanted to share that with the IKEA
repeaters the link quality has been very stable for me with the conbee II.
I believe I was the original reporter of the issue btw, I do care and bugs
needs to be fixed especially in a commercially supported firmware like the
conbee II has.

MattWestb, 2021-01-19:
@m0wlheld  Good dialog with the CornBee "support" but i think you need little more luck to getting the confirmed bug fixes. And also sad is that the second gens coordinators is not having touchlink implanted and that is also "on the roadmap without ETA" after many months users requests, that is little low for one premium product (first gen is having it).

The positive thing its one good radio and ceramic antenna if not having decent routers in the network.
The radio can being little too "aggressive" so end device dont like being connected to routers in the near of them and is losing network connection then the coordinator is offline (Light link devices is working then the coordinator is offline but not if sleeping end devices is being connected to one offline coordinator).

And i agree with @s3frank then its one commercial product and not one 5 € CC2531 from china.

Comparing with Sonoff was cooking one new (from the factory) signed firmware for their zigbee bridge in 1 - 2 working days after being requested from the community (4 times the last 6 months) and the unsigned one from the community devs was done in some hours.
The Z2M community devs is also very good then its possible cooking fixed firmware for our TI coordinators (but can being very problematic some times with old and very closed SDK).

craigmcgowan, 2021-01-25:
I have just realised this is my issue too with latest firmware, fresh install etc.

[image]
[image]

kwetnico, 2021-01-25:
Or can we see if the company is looking for the problem of LQI to offer us a correction, because we do not know

> I have just realised this is my issue too with latest firmware, fresh install etc.

Where can we see if the company is looking for the problem of LQI to offer us a correction, because we do not know

MattWestb, 2021-01-25:
@m0wlheld wos gently asking the compamy https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4234.

kwetnico, 2021-01-25:
> @m0wlheld wos gently asking the compamy [dresden-elektronik/deconz-rest-plugin#4234](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4234).

Thanks wait and see

m0wlheld, 2021-01-25:
"It's already on the internal issue list." . To be honest, that LQI display is the pain-point that drives me away from Conbee/Raspbee II, but the fact that I need to re-pair all devices keeps me from switching to [slae.sh's CC2652RB](https://slae.sh/projects/cc2652/) stick, already laying on my desk.

s3frank, 2021-01-25:
I'm still a bit confused as to why Dresden isn't fixing this with a level
of basic business urgency.
I am not looking for hotfix type stuff, but it seems like they ship a
product with an inferior implementation in this part of the firmware and
their fix is to say that it shouldn't really be used as it's unreliable
anyway? That's like buying a car with a broken speedometer and the dealers
say to ignore it as it wasn't super accurate...just stick your face out the
window to get a better sense of things while driving....

I think I'm going to join the conversation @m0wlheld
<https://github.com/m0wlheld> started with them...

m0wlheld, 2021-01-25:
They should release a new firmware soon (monthly cycle), maybe we're lucky.

MattWestb, 2021-01-25:
@m0wlheld Sadly they is not converting tool for the CornBee sticks to other coordinators.
ZHA have doing it for some (TI and EZSP) coordinators and having one dialogue with Z2M to trying doing one "universal" backup / restore between coordinators that is working with both Z2M and ZHA co possible moving the coordinator between systems and changing coordinators without losing all devices.

@s3frank Its not only the firmware in our coordinators that is one mess in that company if you is reading little more on ther git issues . . .

docpayce, 2021-02-20:
Welp, that's a back to zig-a-zig-ah! for me. Goodbye Conbee II, has been a nice time.

ASNNetworks, 2021-02-20:
> Welp, that's a back to zig-a-zig-ah! for me. Goodbye Conbee II, has been a nice time.

Just out of curiosity, what made you switch from zig-a-zig-ah! to Conbee II in the first place? Does that have some issues as well?

docpayce, 2021-02-20:
> Just out of curiosity, what made you switch from zig-a-zig-ah! to Conbee II in the first place? Does that have some issues as well?

No, not really. I use zigbee2mqtt via a plugin for loxberry (https://www.loxwiki.eu/display/LOXBERRY/LoxBerry) and within a update of that plugin, the zig-a-zig-ah! stopped working due to some hickups not caused be zigbee2mqtt nor by the zig-a-zig-ah! firmware, but in that plugin (changing network keys but using old backups caused herdsman to not start I think).
The correct solution would have been to either change panID or similar, the quick and dirty solution was to switch the stick. XD
Came back to bite me. ;)

thackel, 2021-03-23:
Still open on dresden elektronic side -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4234

kwetnico, 2021-04-24:
> Still open on dresden elektronic side -> [dresden-elektronik/deconz-rest-plugin#4234](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4234)

yes still in progress and no solution for the moment

closed 2021-05-02 by github-actions[bot]

MattWestb, 2021-05-14:
New firmware for CornBee II https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4234#issuecomment-840646221
But not for CornBee I and RaspBee I/II but  that was expected from DE :-(

ChrisHae, 2021-05-25:
Please check out the next zigbee2mqtt release in dev-branch. The false lqi values should be fixed there.

s3frank, 2021-05-25:
ok does that mean the issue was NOT with the conbee stick?
Or is it a combination of?

ChrisHae, 2021-05-25:
> ok does that mean the issue was NOT with the conbee stick? Or is it a combination of?
> […](#)
> -- Best regards,
> -FF

seems so... my apologies

s3frank, 2021-05-25:
Which one is it? Combination or only on the zigbee2mqtt code?
No need to apologize, we all make mistakes. Besides, I think zigbee2mqtt is
an amazing piece of art!

ChrisHae, 2021-05-25:
Udating z2mqtt is enough to get it work.

gpiszczek, 2021-05-26:
@ChrisHae which commit (or PR) contains fix for the issue? I could not find it myself.

ChrisHae, 2021-05-26:
It is in the zigbee-herdsman repository. I started a github action to build a new release there so @Koenn will be notified to integrate it in the z2mqtt-dev branch and make a new release.

[13 more replies]