Koenkk/zigbee2mqtt #4040: TUYA like temperature humidity sensor cannot be paired

AivarasAukselis, 2020-08-04:
## What happened
Recently I bought a temperature humidity sensor. https://www.amazon.nl/gp/product/B089K1NSZ9/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
It looks like TUYA TS0201 clone or something
https://www.zigbee2mqtt.io/devices/TS0201.html

However when I connect to zigbee2mqtt, it does not work. z2ma shows that it is functioning properly but the sensor readings are not published to mqtt. Furthermore z2ma reports model and make as unknown:
_Zigbee Id is 0xec1bbdfffea227d0.
Address on Zigbee Network is 0x5CC2.
Link quality is 202/255.
Model is unknown by unknown.
Hardware version is unknown.
Firmware version is unknown.
This device type is EndDevice_

If I view z2mqtt logs during the pairing of the device I see that the values from the device are received successfully but this warning is also printed:
`Received message from unsupported device with Zigbee model 'undefined'`

## What did you expect to happen
Device pair successfully and the readings from the device published to mqtt.

## How to reproduce it (minimal and precise)
Pair said device to zigbee2mqtt network

## Debug Info
https://paste.ubuntu.com/p/Yfp2RNHdYz/

Zigbee2mqtt version:
Adapter hardware: **CC2531**
Adapter firmware version:
`{'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}}`

AivarasAukselis, 2020-08-04:
I have removed the device from zigbee2mqtt and installed dev version. However, the issue is the same. When repairing the device from new, the logs show more information:
https://paste.ubuntu.com/p/G4KnW7NVb5/

paveldn, 2020-08-09:
It helped me with the same device to remove manually records from database.db state.json, and database.db.backup. After such cleanup pairing was successful.

AivarasAukselis, 2020-08-09:
@paveldn I tried that but it still did not work. However, what worked for me is manually adding `manufName` and `modelId` to database.db. making the entry as follows

> {"id":8,"type":"EndDevice","ieeeAddr":"0xec1bbdfffea227d0","nwkAddr":34506,"manufId":4098,**"manufName":"TuYa"**,"powerSource":"Battery",**"modelId":"TS0201"**,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2780}},"genBasic":{"attributes":{"powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"msRelativeHumidity":{"attributes":{"measuredValue":4550}},"genPowerCfg":{"attributes":{"batteryVoltage":32,"batteryPercentageRemaining":200}}},"binds":[]}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1596574360431}

Why were these entries not added automatically? is it an issue with the device, bridge or my setup? I see that `manufld` and `devId` was added automatically. Is it something unique to the manufacturers and devices? perhaps devices settings could be bound to these ids in case the devices does not report text manufacturer and model?

closed 2020-09-16 by github-actions[bot]

wimpie007, 2020-11-05:
manually adding manufName and modelId fixed it for me also.
Strange...

loglux, 2020-11-13:
I have this issue. it can't pair

Sampledelic, 2020-11-21:
I have the same issue after updating to the latest HA, Zigbee2MQTT and upgrading my C26x2R1.
One of these have broken the recognition of my Tuya device, I assume its the C26x2R1 firmware as I had to reconnect all my ZigBee devices.

7system7, 2022-10-01:
I have the same issue.

floresek, 2023-01-14:
I have the same problem with:
https://expo.tuya.com/product/338587

My dongle is:
https://sonoff.tech/product/gateway-and-sensors/sonoff-zigbee-3-0-usb-dongle-plus-e/
pi@raspberrypi:~ $ find /dev/serial
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221029114026-if00

What can I do to discover the device?