Koenkk/zigbee2mqtt #4046: LLKZMK11LM sending the same message several times in the same second

McGiverGim, 2020-08-05:
# Bug Report
## What happened
I have at this moment three LLKZMK11LM. Two of them work perfectly, but the third is sending the same message several times before changing some state some seconds later. Here is some partial log showing the problem:
```txt
...
zigbee2mqtt:info  2020-08-05 09:42:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":35.77,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:42:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":35.77,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:42:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":35.77,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:42:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":35.77,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:42:25: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
...
zigbee2mqtt:info  2020-08-05 09:44:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:44:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:44:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:44:22: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0.04,"temperature":43,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-05 09:44:34: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0,"consumption":0,"temperature":44,"state_l1":"OFF"}'
...
```
The only difference I can thing between this relay and the other two, is that the other two have the interlock enabled and this other not.
I suppose I can fix this with a `debounce` configuration, but maybe there is a bug somewhere that is better to fix.

I attach the complete log. The Aqara relays all of them start with `rele_`
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/5026956/log.zip)

## What did you expect to happen
To only send one message at some moment if there is no change in the attributes.

## How to reproduce it (minimal and precise)
Difficult to say, I don't know why it only happens in one of my units.

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version: zStack 12.20190608

Koenkk, 2020-08-05:
- First make sure to check the latest dev first (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- If that doesn't fix it, can you provide the debug log of this issue?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

McGiverGim, 2020-08-06:
I'm under Home Assistant. How can I switch to edge without loosing the actual configuration? If I uninstall the "standard" version and install the edge one the configuration will remain?

McGiverGim, 2020-08-06:
This is the relevant log in `debug` mode but with stable version, if it helps:
```txt
zigbee2mqtt:debug 2020-08-06 08:56:15: Received Zigbee message from 'presencia_cocina', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":26}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 08:56:15: Received Zigbee message from 'presencia_cocina', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/presencia_cocina', payload '{"battery":100,"voltage":3025,"illuminance":26,"illuminance_lux":26,"linkquality":86,"occupancy":true}'
zigbee2mqtt:debug 2020-08-06 08:56:15: Received MQTT message on 'zigbee2mqtt/rele_cocina/l2/set' with data 'OFF'
zigbee2mqtt:debug 2020-08-06 08:56:15: Publishing 'set' 'state' to 'rele_cocina'
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
zigbee2mqtt:debug 2020-08-06 08:56:15: Received MQTT message on 'zigbee2mqtt/rele_cocina/l1/set' with data 'OFF'
zigbee2mqtt:debug 2020-08-06 08:56:15: Publishing 'set' 'state' to 'rele_cocina'
zigbee2mqtt:debug 2020-08-06 08:56:15: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440624,"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-06 08:56:16: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
zigbee2mqtt:debug 2020-08-06 08:56:16: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440625,"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:16: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
```
This seems to happen only when there is a change in the presence in the same room. Maybe Home Assistant is sending some action to the relay and this produces a message?

Koenkk, 2020-08-07:
The following part of the log is relevant:

```

Koenkk, 2020-08-07:
The following part of the log is relevant:

```
zigbee2mqtt:debug 2020-08-06 08:56:15: Received MQTT message on 'zigbee2mqtt/rele_cocina/l1/set' with data 'OFF'
zigbee2mqtt:debug 2020-08-06 08:56:15: Publishing 'set' 'state' to 'rele_cocina'
zigbee2mqtt:debug 2020-08-06 08:56:15: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440624,"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
zigbee2mqtt:info  2020-08-06 08:56:16: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
zigbee2mqtt:debug 2020-08-06 08:56:16: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440625,"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:16: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
```

In this log there are 3 published state messages which are all explainable:
1. Zigbee2MQTT receives a request to turn of l1, this succeeds so Zigbee2MQTT publishes `{"state_l1": "OFF"}`, relevant log:
```
zigbee2mqtt:debug 2020-08-06 08:56:15: Received MQTT message on 'zigbee2mqtt/rele_cocina/l1/set' with data 'OFF'
zigbee2mqtt:debug 2020-08-06 08:56:15: Publishing 'set' 'state' to 'rele_cocina'
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
```
2. Device reports the state of l2 (endpoint 2) is in the off state, relevant log:
```
zigbee2mqtt:debug 2020-08-06 08:56:15: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440624,"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:15: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":65,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
```
3. Device reports the state of l1 (endpoint 1) is in the off state, relevant log:
```
zigbee2mqtt:debug 2020-08-06 08:56:16: Received Zigbee message from 'rele_cocina', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440625,"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-06 08:56:16: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"state_l2":"OFF","linkquality":68,"power":0.03,"consumption":0.02,"temperature":44,"state_l1":"OFF"}'
```

When to report a state is up to the device itself, so you may find different behaviour over different switches since they can contain a different firmware.

In case you want to prevent the duplicate MQTT publishes you can use the debounce feature: https://www.zigbee2mqtt.io/information/configuration.html#all-devices

McGiverGim, 2020-08-07:
Ok. If it's expected behavior I will close this. Thanks for spending your time looking at this! I will use the debounce.

closed 2020-08-07 by McGiverGim