Koenkk/zigbee2mqtt #4048: [QOL] Format Home Assistant entity names without _

blakadder, 2020-08-05:
Currently Home Assistant entity names are generated with `_` which is perfectly functional but doesn't look that great in the UI and more importantly when using device triggers in automations.

[image]
[image]

While it is possible to change the names manually in Lovelace it would become quite a chore proportional to number of entities a device creates.

I'm proposing a simple change of `_` into a space when generating `name: ` value autodiscovery payload so the end result would be:
[image]
[image]

Koenkk, 2020-08-05:
I think this highly depends on how people name their entities. Personally I prefer to use snake_case for the `friendly_name` as I dislike the spaces, e.g. `kitchen_climate_sensor`, having this to appear as `kitchen_climate_sensor temperature` feels strange to me.

Something we can do is to use a space instead of `_` when the `friendly_name` contains a space. Does that sound like a good approach?

blakadder, 2020-08-06:
Excellent solution!

Koenkk, 2020-08-06:
Maybe even more fancy (not sure if this is desired):
- `Sonoff Motion` -> `Sonoff Motion Battery` (capital B)
- `sonoff motion` -> `sonoff motion battery`
- Default `_battery`

blakadder, 2020-08-06:
Perhaps, personally I'm not bothered by case...

Also what if it's "Sonoff motion" ?

Koenkk, 2020-08-08:
I've took that from your example in the OP.

I've implemented this now, assuming this can be closed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-08-08 by Koenkk

crowbarz, 2020-08-12:
I had been manually converting that `_` to a space in the `friendly_name` for every generated entity via `devices.yaml` overrides. This feature will eliminate the need to continue to do that - thanks!

However, a small pre-upgrade question, what happens with entities whose name is just a single word? Will these continue to need manual modification either in `devices.yaml`, HA entity settings, or Lovelace?

Koenkk, 2020-08-12:
> However, a small pre-upgrade question, what happens with entities whose name is just a single word? Will these continue to need manual modification either in devices.yaml, HA entity settings, or Lovelace?

It will only be done when the friendly name contains a space (so not in this case)