Koenkk/zigbee2mqtt #4049: Z2M Spam to Auto-Discovery topic (homeassistant)

to4ko, 2020-08-05:
# Bug Report
## What happened
Checking my mqtt broker (mosquitto installed on host machine) found that thare to many connections in minute (1200 send and 900 receive). Further investigation showed that autodiscovery topic "homeassistant" got republished every moment - around 200-250 messages in second.
My setup - CC2538 + Mosquitto on dedicated j3455 pc. 144 Zigbee devices (mainly Xiaomi)

[image]

## What did you expect to happen
do not such a republish in autodiscovery topic

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 0.14.2
Adapter hardware: CC2538
Adapter firmware version: {'type': 'zStack30x', 'meta': {'transportrev': 2, 'product': 2, 'majorrel': 2, 'minorrel': 7, 'maintrel': 2, 'revision': 20200211}}

Spirituss, 2020-08-05:
I would expect to have ability to customize update interval for each entity’s sensor/topic. For example, for plugs voltage or consumption there is no need to receive updates every second. It would be useful to use something like this:
homeassistant: 
  consumption: 
    update_interval: 60
meaning 60 seconds update interval.

Koenkk, 2020-08-05:
Zigbee2MQTT should only publish the discovery once (at the moment z2m starts or when a new device is connected which has not been discovered before). 

Do you also see that amount in the z2m log?

to4ko, 2020-08-06:
> Zigbee2MQTT should only publish the discovery once (at the moment z2m starts or when a new device is connected which has not been discovered before).
> 
> Do you also see that amount in the z2m log?

my bad....it was other instance reconnecting to broker...closing this one

closed 2020-08-06 by to4ko