Koenkk/zigbee2mqtt #4054: Gledopto RGB+CTT 2ID not publishing all data on mqtt when configured to publish attribute_and_json

ivanfmartinez, 2020-08-06:
# Bug Report
## What happened
I have connected the Gledopto RGB+CCT 2ID device, I can control the device (using rgb and cct) , but It does not publish all data on the mqtt.

When controlling device I can use topic /pqda/zigbee2mqtt/led_rgb_cct_002/set or /pqda/zigbee2mqtt/led_rgb_cct_002/rgb/set to control the rgb part, but the values in the json are not synchronized (I think this is another problem for checking later).

https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html

I'm using :
```
experimental:
  output: attribute_and_json
```

- The device is showing as detail only some values , but not the rgb and cct, appear to be stopping after the first object found "color", I did not have any other device with a complex data like this color attribute to make sure that this is the cause
``` 
/zigbee2mqtt/led_rgb_cct_002 {"state":"OFF","brightness":0,"color":{"x":0.1724,"y":0.7468},"color_temp":154,"state_rgb":"ON","brightness_rgb":100,"state_cct":"OFF","brightness_cct":0,"color_temp_cct":0,"color_rgb":{"x":0.5,"y":0.6},"color_temp_rgb":249,"last_seen":"2020-08-06T18:08:02.024Z"}
/zigbee2mqtt/led_rgb_cct_002/state OFF
/zigbee2mqtt/led_rgb_cct_002/brightness 0
/zigbee2mqtt/led_rgb_cct_002/color-x 0.1724
/zigbee2mqtt/led_rgb_cct_002/color-y 0.7468
```
- For reference is possible to see that on other device all attributes from json are show
```
/zigbee2mqtt/router_001 {"led":false,"linkquality":18,"elapsed":59983,"last_seen":"2020-08-06T18:08:30.355Z"}
/zigbee2mqtt/router_001/led false
/zigbee2mqtt/router_001/linkquality 18
/zigbee2mqtt/router_001/elapsed 59983
/zigbee2mqtt/router_001/last_seen 2020-08-06T18:08:30.355Z
```

## What did you expect to happen

The values for all attributes ended with _rgb and _cct should be published to mqtt

## How to reproduce it (minimal and precise)

use the GL-C-008-2ID and configure to publish attributes_and_json on mqtt

## Debug Info
Zigbee2mqtt version:1.14.2
Adapter hardware: CC2531
Adapter firmware version::20190619

even with log in debug mode there are no different message on log.

Koenkk, 2020-08-08:
Should be fixed in the latest dev branch now, please confirm.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ivanfmartinez, 2020-08-09:
Thanks,

The attributes are being published now, but its also publishing a "(null)"  value for each attribute that is an object.
because now it reach the publish command with a null message.

```
/zigbee2mqtt/led_rgb_cct_002 {"state":"OFF","brightness":0,"color":{"x":0.5,"y":0.6},"color_temp":249,"state_rgb":"OFF","brightness_rgb":0,"state_cct":"OFF","brightness_cct":0,"color_temp_cct":1000,"color_rgb":{"x":0.1724,"y":0.7468},"color_temp_rgb":154,"color_cct":{"x":0.652750055750174,"y":0.34446222719737},"last_seen":"2020-08-09T00:02:59.465Z"}
/zigbee2mqtt/led_rgb_cct_002/state OFF
/zigbee2mqtt/led_rgb_cct_002/brightness 0
/zigbee2mqtt/led_rgb_cct_002/color-x 0.5
/zigbee2mqtt/led_rgb_cct_002/color (null)
/zigbee2mqtt/led_rgb_cct_002/color-y 0.6
/zigbee2mqtt/led_rgb_cct_002/color_temp 249
/zigbee2mqtt/led_rgb_cct_002/state_rgb OFF
/zigbee2mqtt/led_rgb_cct_002/brightness_rgb 0
/zigbee2mqtt/led_rgb_cct_002/state_cct OFF
/zigbee2mqtt/led_rgb_cct_002/brightness_cct 0
/zigbee2mqtt/led_rgb_cct_002/color_temp_cct 1000
/zigbee2mqtt/led_rgb_cct_002/color_rgb-x 0.1724
/zigbee2mqtt/led_rgb_cct_002/color_rgb (null)
/zigbee2mqtt/led_rgb_cct_002/color_rgb-y 0.7468
/zigbee2mqtt/led_rgb_cct_002/color_temp_rgb 154
/zigbee2mqtt/led_rgb_cct_002/color_cct-x 0.652750055750174
/zigbee2mqtt/led_rgb_cct_002/color_cct (null)
/zigbee2mqtt/led_rgb_cct_002/color_cct-y 0.34446222719737
/zigbee2mqtt/led_rgb_cct_002/last_seen 2020-08-09T00:02:59.465Z
```

Koenkk, 2020-08-09:
Should also be fixed now.

ivanfmartinez, 2020-08-09:
Thanks, its working fine.

closed 2020-08-09 by ivanfmartinez