Koenkk/zigbee2mqtt #4055: SmartThings Arrival Sensor PGC410EU support?

liamstears, 2020-08-06:
Referring to the closed issue here: https://github.com/Koenkk/zigbee2mqtt/issues/3790

I have the same device and running into the same issue and hope you can resolve it?

I'm in the UK and I believe this model is the only SmartThings arrival sensor available in the UK so I'm sure I won't be the only person grateful if we get this working.

During testing I found that my device reports itself as "PGC410EU" instead of just "PGC410" so I made the necessary adjustments in devices.js

I read the previous and you mention keeping it awake, I have done this but also ran debug level logging and here are the results:

```zigbee2mqtt:info  2020-08-06 22:31:57: Device '0xd052a81e8ca00005' joined
zigbee2mqtt:info  2020-08-06 22:31:57: Starting interview of '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-06 22:31:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:info  2020-08-06 22:31:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-06 22:31:58: Device '0xd052a81e8ca00005' announced itself
zigbee2mqtt:info  2020-08-06 22:31:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-06 22:32:18: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"modelId":"PGC410EU"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:18: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"PGC410EU"}'
zigbee2mqtt:debug 2020-08-06 22:32:28: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"SmartThings"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:28: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"SmartThings"}'
zigbee2mqtt:debug 2020-08-06 22:32:38: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:38: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":0}'
zigbee2mqtt:debug 2020-08-06 22:32:48: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:48: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-08-06 22:32:58: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:58: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":4}'
zigbee2mqtt:debug 2020-08-06 22:33:08: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"stackVersion":71}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:08: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":71}'
zigbee2mqtt:debug 2020-08-06 22:33:18: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"hwVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:18: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":3}'
zigbee2mqtt:debug 2020-08-06 22:33:28: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"dateCode":"130925"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:28: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"130925"}'
zigbee2mqtt:debug 2020-08-06 22:33:38: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:38: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:error 2020-08-06 22:33:48: Failed to interview '0xd052a81e8ca00005', device has not successfully been paired
zigbee2mqtt:info  2020-08-06 22:33:48: Configuring '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-06 22:33:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
```

Hopefully you can help get this resolved?

Koenkk, 2020-08-08:
Please provide the herdsman debug logging when pairing this device.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

liamstears, 2020-08-08:
> Please provide the herdsman debug logging when pairing this device.
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

I can do this now but still new to all this so want to check what exactly I need to do with "zigbee_herdsman_debug": true

In my configuration would I have:

```
advanced:
  log_level: debug
  zigbee_herdsman_debug: true
  pan_id: 6754
  channel: 20

```

Or:

```
advanced:
  zigbee_herdsman_debug: true
  pan_id: 6754
  channel: 20

```

Or:

```
zigbee_herdsman_debug: true
advanced:
  log_level: debug
  pan_id: 6754
  channel: 20

```

Also did another test and just got this:

```
zigbee2mqtt:info  2020-08-08 15:50:32: Successfully interviewed '0xd052a81e8ca00005', device has successfully been paired
zigbee2mqtt:info  2020-08-08 15:50:32: Device '0xd052a81e8ca00005' is supported, identified as: SmartThings Arrival sensor (PGC410EU)
zigbee2mqtt:info  2020-08-08 15:50:32: Configuring '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-08 15:50:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xd052a81e8ca00005","model":"PGC410EU","vendor":"SmartThings","description":"Arrival sensor","supported":true}}'
zigbee2mqtt:error 2020-08-08 15:51:05: Failed to configure '0xd052a81e8ca00005', attempt 2 (Error: ConfigureReporting 0xd052a81e8ca00005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 45913 - 1 - 24 - 1 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

```

liamstears, 2020-08-08:
I've tried and tried to get extra debugging but have failed, I have only been able to achieve the above, really want to get this fixed though...

Koenkk, 2020-08-08:
Your "another test" looks better already! interview is succesfull, only configure fails, this can be due to the device falling a sleep when pairing, make sure to keep it awake by pressing a button on it.

liamstears, 2020-08-08:
> Your "another test" looks better already! interview is succesfull, only configure fails, this can be due to the device falling a sleep when pairing, make sure to keep it awake by pressing a button on it.

Tried again, keeping it awake, same problem:

```
zigbee2mqtt:info  2020-08-08 19:39:04: Device '0xd052a81e8ca00005' joined
zigbee2mqtt:info  2020-08-08 19:39:04: Starting interview of '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-08 19:39:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:info  2020-08-08 19:39:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-08 19:39:04: Device '0xd052a81e8ca00005' announced itself
zigbee2mqtt:info  2020-08-08 19:39:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-08 19:39:49: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"modelId":"PGC410EU"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:39:49: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"PGC410EU"}'
zigbee2mqtt:debug 2020-08-08 19:39:54: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"SmartThings"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:39:54: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"SmartThings"}'
zigbee2mqtt:debug 2020-08-08 19:39:59: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:39:59: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":0}'
zigbee2mqtt:debug 2020-08-08 19:40:04: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:04: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-08-08 19:40:09: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:09: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":4}'
zigbee2mqtt:debug 2020-08-08 19:40:14: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"stackVersion":71}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:14: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":71}'
zigbee2mqtt:debug 2020-08-08 19:40:19: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"hwVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:19: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":3}'
zigbee2mqtt:debug 2020-08-08 19:40:24: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"dateCode":"130925"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:24: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"130925"}'
zigbee2mqtt:debug 2020-08-08 19:40:29: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-08 19:40:29: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:info  2020-08-08 19:40:34: Successfully interviewed '0xd052a81e8ca00005', device has successfully been paired
zigbee2mqtt:info  2020-08-08 19:40:34: Device '0xd052a81e8ca00005' is supported, identified as: SmartThings Arrival sensor (PGC410EU)
zigbee2mqtt:info  2020-08-08 19:40:34: Configuring '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-08 19:40:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xd052a81e8ca00005","model":"PGC410EU","vendor":"SmartThings","description":"Arrival sensor","supported":true}}'
zigbee2mqtt:error 2020-08-08 19:41:08: Failed to configure '0xd052a81e8ca00005', attempt 1 (Error: ConfigureReporting 0xd052a81e8ca00005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 13518 - 1 - 20 - 1 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

```

liamstears, 2020-08-08:
I don't know if this will be of help from Samsung: https://github.com/SmartThingsCommunity/SmartThingsPublic/tree/master/devicetypes/smartthings/arrival-sensor.src

Koenkk, 2020-08-09:
- Can you share the `data/database.db` entry of this device?
- With the following entry do you regularly  see a `No converter available...` message for this device?

```js
{
    zigbeeModel: ['PGC410EU'],
    model: 'STS-PRS-251',
    vendor: 'SmartThings',
    description: 'Arrival sensor',
    supports: 'presence',
    fromZigbee: [],
    toZigbee: [],
},
```

liamstears, 2020-08-09:
> * Can you share the `data/database.db` entry of this device?
> * With the following entry do you regularly  see a `No converter available...` message for this device?
>
> ```js
> {
>     zigbeeModel: [...]

This is everything that showed up when paired again with the entry as above:

```
zigbee2mqtt:info  2020-08-09 20:53:33: Device '0xd052a81e8ca00005' joined
zigbee2mqtt:info  2020-08-09 20:53:33: Starting interview of '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-09 20:53:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:info  2020-08-09 20:53:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-09 20:53:33: Device '0xd052a81e8ca00005' announced itself
zigbee2mqtt:info  2020-08-09 20:53:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-09 20:54:32: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"modelId":"PGC410EU"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:32: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"PGC410EU"}'
zigbee2mqtt:info  2020-08-09 20:54:32: MQTT publish: topic 'homeassistant/binary_sensor/0xd052a81e8ca00005/presence/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.presence }}","device_class":"presence","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_presence","unique_id":"0xd052a81e8ca00005_presence_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2mqtt 1.14.2","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-09 20:54:32: MQTT publish: topic 'homeassistant/sensor/0xd052a81e8ca00005/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_battery","unique_id":"0xd052a81e8ca00005_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2mqtt 1.14.2","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-08-09 20:54:32: MQTT publish: topic 'homeassistant/sensor/0xd052a81e8ca00005/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_linkquality","unique_id":"0xd052a81e8ca00005_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2mqtt 1.14.2","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:debug 2020-08-09 20:54:37: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"SmartThings"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:37: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"SmartThings"}'
zigbee2mqtt:debug 2020-08-09 20:54:42: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:42: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":0}'
zigbee2mqtt:debug 2020-08-09 20:54:47: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:47: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-08-09 20:54:52: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:52: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":4}'
zigbee2mqtt:debug 2020-08-09 20:54:57: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"stackVersion":71}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-09 20:54:57: No converter available for 'STS-PRS-251' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":71}'
zigbee2mqtt:debug 2020-08-09 20:55:02: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"hwVersion":3}' from endpoint 1 with groupID 0
[... cut ...] ,"inClusterList":[0],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"PGC410EU","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925"}}},"binds":[]},"2":{"profId":64513,"epId":2,"devId":410,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1597002912112}
```

Koenkk, 2020-08-09:
We need to get the herdsman debug logging to get further, how are you running zigbee2mqtt? (hassio/docker/baremetal?)

liamstears, 2020-08-09:
> We need to get the herdsman debug logging to get further, how are you running zigbee2mqtt? (hassio/docker/baremetal?)

hassos/hassio

Koenkk, 2020-08-09:
In that case you need to switch to the edge addon version and set `"zigbee_herdsman_debug":true` in the configuration (on the root level)

liamstears, 2020-08-09:
> In that case you need to switch to the edge addon version and set `"zigbee_herdsman_debug":true` in the configuration (on the root level)

So I've managed to get the log working but its huge, it's too big for me to capture it live from the log on home assistant but the log on samba \share\zigbee2mqtt\log doesn't have the herdsman debug logging on it?

liamstears, 2020-08-09:
Ok so I tried what I can from the live log and got the following but bear in mind I may have missed bits, bits may be duplicated and there are bits in there from another device:

```
Zigbee2MQTT:debug 2020-08-09 22:26:04: Device '0xd052a81e8ca00005' announced itself
Zigbee2MQTT:info  2020-08-09 22:26:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
2020-08-09T21:26:10.950Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,93,191,1,165,229,39]
2020-08-09T21:26:10.950Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,93,191,1,165,229,39]
2020-08-09T21:26:10.951Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [93,191,1,165,229] - 39
2020-08-09T21:26:10.951Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":1,"relaylist":[58789]}
2020-08-09T21:26:10.951Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-09T21:26:11.008Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,93,191,18,13,0,26,0,64,12,120,0,0,12,1,127,1,0,20,16,2,0,9,0,0,0,165,229,8,27]
2020-08-09T21:26:11.009Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,93,191,18,13,0,26,0,64,12,120,0,0,12,1,127,1,0,20,16,2,0,9,0,0,0,165,229,8,27]
2020-08-09T21:26:11.009Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,93,191,18,13,0,26,0,64,12,120,0,0,12,1,127,1,0,20,16,2,0,9,0,0,0,165,229,8] - 27
2020-08-09T21:26:11.010Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48989,"srcendpoint":18,"dstendpoint":13,"wasbroadcast":0,"linkquality":26,"securityuse":0,"timestamp":7867456,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,127,1,0,20,16,2,0,9,0,0,0]}}
2020-08-09T21:26:11.013Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":127,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}},"address":48989,"endpoint":18,"linkquality":26,"groupID":0}'
2020-08-09T21:26:11.017Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-09T21:26:11.020Z zigbee-herdsman:controller:endpoint CommandResponse 0x0008890001d9b4a5/18 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
Zigbee2MQTT:info  2020-08-09 22:26:49: MQTT publish: topic 'homeassistant/binary_sensor/0xd052a81e8ca00005/presence/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.presence }}","device_class":"presence","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_presence","unique_id":"0xd052a81e8ca00005_presence_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2MQTT 1.14.2-dev","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Zigbee2MQTT:info  2020-08-09 22:26:49: MQTT publish: topic 'homeassistant/sensor/0xd052a81e8ca00005/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_battery","unique_id":"0xd052a81e8ca00005_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2MQTT 1.14.2-dev","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Zigbee2MQTT:info  2020-08-09 22:26:49: MQTT publish: topic 'homeassistant/sensor/0xd052a81e8ca00005/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0xd052a81e8ca00005","json_attributes_topic":"zigbee2mqtt/0xd052a81e8ca00005","name":"0xd052a81e8ca00005_linkquality","unique_id":"0xd052a81e8ca00005_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xd052a81e8ca00005"],"name":"0xd052a81e8ca00005","sw_version":"Zigbee2MQTT 1.14.2-dev","model":"Arrival sensor (STS-PRS-251)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2020-08-09T21:26:49.879Z zigbee-herdsman:controller:endpoint Read 0xd052a81e8ca00005/1 genBasic(["manufacturerName"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-08-09T21:26:49.880Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":19564,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":8,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,9,0,4,0]}}
2020-08-09T21:26:49.880Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,108,76,1,1,0,0,8,0,30,5,16,9,0,4,0,4]
[... cut ...]
2020-08-09T21:27:36.535Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":129,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}},"address":48989,"endpoint":18,"linkquality":28,"groupID":0}'
2020-08-09T21:27:36.539Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

```

liamstears, 2020-08-09:
Hope this is all the info you need and we are noq on a path to a fix :-D

Koenkk, 2020-08-10:
Can you post some more logging after the device have been paired, I'm looking for a regular message send by the device.

liamstears, 2020-08-10:
> Can you post some more logging after the device have been paired, I'm looking for a regular message send by the device.

I think I may have found what you are looking for

Firstly database is now:

```
{"id":11,"type":"EndDevice","ieeeAddr":"0xd052a81e8ca00005","nwkAddr":27174,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"PGC410EU","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":410,"inClusterList":[0],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"PGC410EU","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925"}}},"binds":[]},"2":{"profId":64513,"epId":2,"devId":410,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1597086143079}

```

I then got this error:

```2020-08-10T18:55:09.622Z zigbee-herdsman:controller:device:log Interview - failed for device '0xd052a81e8ca00005' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)'
2020-08-10T18:55:09.631Z zigbee-herdsman:controller:error Interview failed for '0xd052a81e8ca00005 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'

```

But then I repaired and got another error:

```
2020-08-10T19:01:53.399Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xd052a81e8ca00005', attempt 6
2020-08-10T19:01:53.400Z zigbee-herdsman:controller:device:log Interview - failed for device '0xd052a81e8ca00005' with error 'Error: Interview failed because can not get node descriptor ('0xd052a81e8ca00005')
    at Device.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)'
2020-08-10T19:01:53.406Z zigbee-herdsman:controller:error Interview failed for '0xd052a81e8ca00005 with error 'Error: Interview failed because can not get node descriptor ('0xd052a81e8ca00005')'
Zigbee2MQTT:error 2020-08-10 20:01:53: Failed to interview '0xd052a81e8ca00005', device has not successfully been paired

```

After this error though it said successfully paired so I kept an eye on the logs and found some bits which may be useful (this is not how the log printed it but snippets that could be useful posted together):

```
Zigbee2MQTT:info  2020-08-10 20:02:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xd052a81e8ca00005","model":"STS-PRS-251","vendor":"SmartThings","description":"Arrival sensor","supported":true}}'
2020-08-10T19:02:33.020Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:02:33.021Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:02:33.021Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [38,106,1,166,105] - 6
2020-08-10T19:02:33.022Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:02:33.022Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:02:50.983Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:02:50.983Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:02:50.984Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
{"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:02:53.003Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
{"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:03:33.035Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:03:13.020Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:03:13.020Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:03:22.643Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:03:22.644Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:03:22.644Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
2020-08-10T19:03:33.034Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [38,106,1,166,105] - 6
2020-08-10T19:03:33.035Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:03:33.035Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:03:53.049Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:03:53.050Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:03:53.050Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [38,106,1,166,105] - 6
[... cut ...]
2020-08-10T19:05:33.120Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:05:33.120Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:05:53.133Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:05:53.133Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:05:53.134Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [38,106,1,166,105] - 6

```

Koenkk, 2020-08-11:
Do you also see any of the following lines?

```
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg 
```

These indicate that the device has send a message.

liamstears, 2020-08-11:
> Do you also see any of the following lines?
> 
> ```
> zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg 
> ```
> 
> These indicate that the device has send a message.

Unfortunately not, just what I have put above is coming up regularly...

Koenkk, 2020-08-12:
To dive further I think we need to sniff the traffic of the device: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

liamstears, 2020-08-12:
> To dive further I think we need to sniff the traffic of the device: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
> 
> Could you do this?

Currently no as I don't have a sniffer or programmer

But

I've now ordered a CC2531 and programmer so hopefully should be set up and working soon so I can...

liamstears, 2020-08-25:
I have a sniffer!!

So here are the results so far, added the device using the following:

```
{
    zigbeeModel: ['PGC410EU'],
    model: 'STS-PRS-251',
    vendor: 'SmartThings',
    description: 'Arrival sensor',
    supports: 'presence',
    fromZigbee: [],
    toZigbee: [],
},

```

Database entry for the device is:

```
{"id":2,"type":"EndDevice","ieeeAddr":"0xd052a81e8ca00005","nwkAddr":42384,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"PGC410EU","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":410,"inClusterList":[0],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"PGC410EU","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925"}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":4,"stackVersion":71,"hwVersion":3,"dateCode":"130925","zclVersion":1,"interviewCompleted":false,"meta":{},"lastSeen":1598370371273}

```

I did get an error saying not successfully paired although I did see the device and it seemed to be functioning as expected at this point:

```
2020-08-25T15:46:18.213Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2020-08-25T15:46:18.213Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2020-08-25T15:46:18.214Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2020-08-25T15:46:18.214Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2020-08-25T15:46:18.214Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-25T15:46:18.215Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2020-08-25T15:46:18.215Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2020-08-25T15:46:18.215Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2020-08-25T15:46:18.215Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2020-08-25T15:46:18.215Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-25T15:46:18.219Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,24,45]
2020-08-25T15:46:18.219Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,24,45]
2020-08-25T15:46:18.220Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,24] - 45
2020-08-25T15:46:18.221Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":24}
2020-08-25T15:46:18.221Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-25T15:46:21.300Z zigbee-herdsman:controller:device:log Interview - failed for device '0xd052a81e8ca00005' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)'
2020-08-25T15:46:21.305Z zigbee-herdsman:controller:error Interview failed for '0xd052a81e8ca00005 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
Zigbee2MQTT:error 2020-08-25 16:46:21: Failed to interview '0xd052a81e8ca00005', device has not successfully been paired
Zigbee2MQTT:info  2020-08-25 16:46:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xd052a81e8ca00005"}}'

```

Then this is the output from the sniffer, it was turned on just before I added the device then left for a while after device had failed to pair to capture any messages, although I believe I turned off all other ZigBee devices in the area it may have picked up other stuff too:

```
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       0x0000                Broadcast             ZigBee   108    Data, Dst: Broadcast, Src: 0x0000

Frame 1: 108 bytes on wire (864 bits), 108 bytes captured (864 bits) on interface \\.\pipe\tiwspc_data, id 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 20, Length: 48
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x0000
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x0000

No.     Time           Source                Destination           Protocol Length Info
      2 0.017283       0x0000                Broadcast             ZigBee   111    Data, Dst: Broadcast, Src: 0x0000

Frame 2: 111 bytes on wire (888 bits), 111 bytes captured (888 bits) on interface \\.\pipe\tiwspc_data, id 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
[... 2557 lines cut ...]
ZigBee Cluster Library Frame, Mfr: Physical Graph Corporation (0x110a)
Data (5 bytes)

0000  1d 0a 11 3b 01                                    ...;.

No.     Time           Source                Destination           Protocol Length Info
    248 320.413607                                                 IEEE 802.15.4 65     Ack

Frame 248: 65 bytes on wire (520 bits), 65 bytes captured (520 bits) on interface \\.\pipe\tiwspc_data, id 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 20, Length: 5
IEEE 802.15.4 Ack, Sequence Number: 218

[... cut ...] h Info
    249 321.016035     0x0000                Broadcast             ZigBee   107    Link Status

Frame 249: 107 bytes on wire (856 bits), 107 bytes captured (856 bits) on interface \\.\pipe\tiwspc_data, id 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 20, Length: 47
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x0000
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x0000

```

Koenkk, 2020-08-25:
Could you provide the `.pcapng` file together with your network key? (you can set the key privately to me on telegram @koenkk)

closed 2020-10-03 by github-actions[bot]

FoxConsult, 2020-10-16:
I have a PGC410, US model (Not the EU model PGC410EU) SmartThings Presence sensor that will not pair:

#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: Device '0xd052a8005bc90005' joined
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: Starting interview of '0xd052a8005bc90005'
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd052a8005bc90005","ieee_address":"0xd052a8005bc90005"},"type":"device_joined"}'
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd052a8005bc90005","ieee_address":"0xd052a8005bc90005","status":"started"},"type":"device_interview"}'
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xd052a8005bc90005"},"type":"device_connected"}'
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xd052a8005bc90005"},"type":"pairing"}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:34:54: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data ''
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20190608","friendly_name":"Coordinator","ieeeAddr":"0x00124b0014d945f1","lastSeen":1602862494368,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"130401","description":"Multipurpose sensor","friendly_name":"0xd052a8006df70003","ieeeAddr":"0xd052a8006df70003","lastSeen":1602861930797,"manufacturerID":4362,"manufacturerName":"SmartThings","model":"STSS-MULT-001","modelID":"PGC313","networkAddress":59210,"powerSource":"Unknown","type":"EndDevice","vendor":"SmartThings"},{"description":"MiJia door & window contact sensor","friendly_name":"0x00158d0002bfc183","ieeeAddr":"0x00158d0002bfc183","lastSeen":1602859746626,"manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ01LM","modelID":"lumi.sensor_magnet","networkAddress":42429,"powerSource":"Battery","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"130401","description":"Motion sensor (2013 model)","friendly_name":"0xd052a80020710004","hardwareVersion":4,"ieeeAddr":"0xd052a80020710004","lastSeen":1602860552422,"manufacturerID":4362,"manufacturerName":"SmartThings","model":"STSS-IRM-001","modelID":"PGC314","networkAddress":4564,"powerSource":"Unknown","type":"EndDevice","vendor":"SmartThings"},{"dateCode":"20161129","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"0x00158d00036b2bff","hardwareVersion":30,"ieeeAddr":"0x00158d00036b2bff","lastSeen":1602861798553,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":44826,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"description":"-","friendly_name":"0xd052a8005bc90005","ieeeAddr":"0xd052a8005bc90005","lastSeen":1602862494331,"networkAddress":14746,"type":"Unknown","vendor":"-"}],"type":"devices"}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:01: Device '0xd052a8005bc90005' announced itself
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:35:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd052a8005bc90005","ieee_address":"0xd052a8005bc90005"},"type":"device_announce"}'
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:35:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xd052a8005bc90005"},"type":"device_announced"}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:06: Received Zigbee message from '0xd052a80020710004', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":49}' from endpoint 1 with groupID 0
#033[32mZigbee2MQTT:info #033[39m 2020-10-16 17:35:06: MQTT publish: topic 'zigbee2mqtt/0xd052a80020710004', payload '{"battery_low":false,"linkquality":102,"occupancy":true,"tamper":false}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:24: Saving state to file /opt/zigbee2mqtt/data/state.json
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:26: Received Zigbee message from '0xd052a8005bc90005', type 'readResponse', cluster 'genBasic', data '{"modelId":"PGC410"}' from endpoint 1 with groupID 0
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:26: No converter available for 'STSS-PRES-001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"PGC410"}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:36: Received Zigbee message from '0xd052a8005bc90005', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"SmartThings"}' from endpoint 1 with groupID 0
#033[34mZigbee2MQTT:debug#033[39m 2020-10-16 17:35:36: No converter available for 'STSS-PRES-001' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"SmartThings"}'
#033[31mZigbee2MQTT:error#033[39m 2020-10-16 17:35:41: Failed to interview '0xd052a8005bc90005', device has not successfully been paired

--------------------------------------------

Here is the zigbee-herdsman log:

Zigbee2MQTT:info  2020-10-16 16:38:58: Zigbee: allowing new devices to join.
  zigbee-herdsman:controller:log Permit joining +40s
[... cut ...] pe":"EndDevice","vendor":"Xiaomi"},{"dateCode":"130324","description":"Presence sensor","friendly_name":"0xd052a8005bc90005","hardwareVersion":3,"ieeeAddr":"0xd052a8005bc90005","lastSeen":1602859256197,"manufacturerID":4362,"manufacturerName":"SmartThings","model":"STSS-PRES-001","modelID":"PGC410","networkAddress":63260,"powerSource":"Unknown","type":"EndDevice","vendor":"SmartThings"}],"type":"devices"}'

------------------------------------------
Any help would be apreciated.

Thanks,

Tommie