Koenkk/zigbee2mqtt #4057: Ikea Tradfri 5 button remote (Triggering all switches)

quarcko, 2020-08-07:
Hi,

Probably this question was answered somewhere but i can't find anything on wiki or elsewhere.
So bought it recently, paired with my z2m setup okay. And when i press middle big power button
all my Xiaomi wall switches toggle! I did nothing, no binding, no grouping, anything.
Later i found out that i need a newer firmware on the stick to get the data about clicks - updated the stick,
and voila, i get now events in MQTT, but the big middle power button still triggers all the Xiaomi switches.
Tried the UNBIND procedure, but it fails:

zigbee2mqtt:error 2020-08-07 12:45:50: Failed to unbind cluster 'genOnOff' from '0x588e81fffe0ea366' to 'default_bind_group' (Error: Unbind 0x588e81fffe0ea366/1 genOnOff from '901' failed (AREQ - ZDO - unbindRsp after 10000ms))
zigbee2mqtt:info  2020-08-07 12:45:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"0x588e81fffe0ea366","to":"default_bind_group","cluster":"genOnOff"}}'
zigbee2mqtt:debug 2020-08-07 12:45:50: unbinding cluster 'genLevelCtrl' from '0x588e81fffe0ea366' to 'default_bind_group'
zigbee2mqtt:error 2020-08-07 12:46:00: Failed to unbind cluster 'genLevelCtrl' from '0x588e81fffe0ea366' to 'default_bind_group' (Error: Unbind 0x588e81fffe0ea366/1 genLevelCtrl from '901' failed (AREQ - ZDO - unbindRsp after 10000ms))
zigbee2mqtt:info  2020-08-07 12:46:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"0x588e81fffe0ea366","to":"default_bind_group","cluster":"genLevelCtrl"}}'
zigbee2mqtt:error 2020-08-07 12:46:00: Failed to unbind

How can i check the firmware version on the button?
Z2M tells me there is no OTA update available, so i assume version is newest?

Koenkk, 2020-08-08:
Try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure 

Right before doing this make sure to click abutton on the remote, otherwise it could be sleeping and miss the request.

quarcko, 2020-08-09:
Strange thing is, i returned now my old stick back with old firmware
and this device can succesfully be "unbound" from default_bind_group, at least the log says so.
Also configure with this old firmware runs ok:
zigbee2mqtt:info  2020-08-09 21:42:42: Configuring '0x588e81fffe0ea366'
zigbee2mqtt:info  2020-08-09 21:42:56: Successfully configured '0x588e81fffe0ea366'

But still pressing central button triggers all my Xiaomi switches.
Any ideas?

Koenkk, 2020-08-09:
You shouldn't unbind it from the default bind group, but keep it bound instead (which is by default after a configure)

quarcko, 2020-08-10:
Hell yeah, configure helped (without unbinding), and now this device does not trigger Xiaomi switches.

Now to the other thing: For this device to work, i need a latest firmware on the stick,
but it performs really bad for me, (i opened  abug report about it some time ago, but nothing seemed to help)
So, question: Can i somehow get the messages from this button, without updating stick firmware?

(Im still on a very old one, version probably: 20181119 that works for me like a champ)
and 20190619 firmwares are like missing 70% of data from the network, and whole system feels just not working)

Koenkk, 2020-08-10:
> i opened abug report about it some time ago, but nothing seemed to help

Can you link to this?

I assume this can be closed as the OP issue has been solved.

closed 2020-08-10 by Koenkk