Koenkk/zigbee2mqtt #4059: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) - Pressing reset doesn't help

JsBergbau, 2020-08-07:
I don't know if this is a bug at all, but I even try.
As described here https://github.com/Koenkk/zigbee2mqtt/issues/3688 I get `Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`
In the FAQ is written one should press the reset button after insert. But still that doesn't help. I think these devices are just not working. However because it occurs with 2 of them, I want to ask here. Perhaps someone knows whats wrong here.

husq101, 2020-08-09:
Same problem here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues/349#issuecomment-657978502

Unfortunately there is no known steady solution for this problem.

closed 2020-09-16 by github-actions[bot]

Stooovie, 2020-12-21:
To anyone getting the dreaded error "Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous>"

Make sure the Z2M config uses the actual physical port the adapter is inserted in. For whatever reason, my port has changed from /dev/ttyACM0 to /dev/ttyACM1 (without me physically touching the Pi for weeks!). After changing the Z2M configuration to reflect this, it started working again immediatelly.

JsBergbau, 2020-12-27:
Thanks for the hint. In my case devices were available at /dev/ttyUSB0 but I changed this in the config of course.
Trying to start with wrong serialport leads to immediately error
` Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'`