Koenkk/zigbee2mqtt #4060: Zigbee2mqtt stops responding randomly on CC26X2R1 stick

cvhoang, 2020-08-08:
# Bug Report

Zigbee2mqtt randomly stops every few days. Log file shows `SRSP - AF - dataRequest after 6000ms` error message.
The only way to get it working again is pressing the reset button new USB adapter and restart Z2M container.

## What did you expect to happen
No random crashes.

## How to reproduce it (minimal and precise)
Runs Z2M on CC26X2R1 with a large number of devices (~100) for a period of time.

## Debug Info
Zigbee2mqtt version: 1.14.1 
Adapter hardware: CC26X2R1
Adapter firmware version: 20200417

## More information

I'm currently running Z2M in Docker on Intel NUC with around 90 devices.

My devices are mostly Hue lights, sensors, Samsung Smart Plugs and Xiaomi sensors. I previously ran Conbee II stick. It was really stable but I migrated because I think Z2M is the future.

configuration.yaml:
```
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.13__Embed_with_CMSIS-DAP_L11008CR-if00
```

docker-compose.yaml:

```
  zigbee2mqtt:
    container_name: z2m
    volumes:
      - /opt/docker/zigbee2mqtt:/app/data
      - /opt/zigbee2mqtt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
      - /opt/zigbee2mqtt/homeassistant.js:/app/lib/extension/homeassistant.js
    devices:
      - /dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.13__Embed_with_CMSIS-DAP_L11008CR-if00
    environment:
      - TZ
    image: koenkk/zigbee2mqtt
    restart: always
```

Koenkk, 2020-08-08:
Can you check with the new firmware? `20200805`: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

cvhoang, 2020-08-09:
Thanks @Koenkk. Will try the new firmware soon. I'll keep you updated.

cvhoang, 2020-08-23:
Unfortunately after a few days running the latest version of the firmware, Z2M just crashed. @Koenkk do you think that buying a different stick (zzh one for example) would help? Thanks.

Koenkk, 2020-08-24:
I'm not sure, note that the Conbee II is also supported by Zigbee2MQTT.

Can you catch a herdsman debug log of the crash?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

cvhoang, 2020-08-24:
Thank you. I assume switching from TI board to Conbee II doesn't require repairing all the devices? It's a dreadful task with 100s of devices and automations.

Koenkk, 2020-08-25:
@cvhoang officially it does, but I heard users reporting that just plugging the conbee and starting works (but it can result in some values not being reported).

cvhoang, 2020-08-25:
Thanks @Koenkk. I can confirm that devices gradually connect to Conbee stick. However I repaired them anyway to prevent any issues. I'll update this thread so that we know that if there are indeed issues with TI board.

cvhoang, 2020-09-05:
@Koenkk I've been running Conbee II stick for the last ten days or so and there have been no crashes. Since the list of devices are the same, it's reasonable to conclude that TI board firmware has issues dealing with a large number of nodes. Unfortunately I didn't capture Herdsman log before switching to Conbee. Hopefully someone will be able to do it in the future. For now I hope Conbee II integration improves as it seems to be a more stable option for large deployment.

closed 2020-10-14 by github-actions[bot]