Koenkk/zigbee2mqtt #4061: Aqara D1 wall switch double rocker no neutral

VanQz, 2020-08-08:
# Bug Report
## What happened
I used mqtt discovery on zigbee2mqtt as recommended settings followed on the documentation webpage. On home assistant i managed to see the switches and i changed the switches entity name and id. Im able to receive the state as home assistant is subscribed to zigbee2mqtt however when i used the switch that was auto discovered, it sends the publish to my broker and forwarded it to zigbee 2 mqtt but nothing happened.
## What did you expect to happen
Lights to turn on however an error occurred in zigbee2mqtt logs
```
[31mzigbee2mqtt:error[39m 2020-08-08 16:13:48: Publish 'set' 'state' to '0x00158d0004a03c73' failed: 'Error: Command 0x00158d0004a03c73/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34438 - 1 - 3 - 6 - 11 after 10000ms)'

[32mzigbee2mqtt:info [39m 2020-08-08 16:13:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d0004a03c73' failed: 'Error: Command 0x00158d0004a03c73/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 34438 - 1 - 3 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00158d0004a03c73"}}'
```
## How to reproduce it (minimal and precise)
turn on auto discovery on both ends and pair the device with zigbee2mqtt and turn on the switch.
## Debug Info
Zigbee2mqtt version: latest
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-08-08:
It looks that the CC2531 is not able to reach the switch, make sure to connect your CC2531 through a USB extension cable to improve its range. If that doesn't help check if it works when putting the switch close to the CC2531.

VanQz, 2020-08-08:
The link state shows that its 47 at current and im able to receive update when the light switch is triggered

VanQz, 2020-08-11:
> It looks that the CC2531 is not able to reach the switch, make sure to connect your CC2531 through a USB extension cable to improve its range. If that doesn't help check if it works when putting the switch close to the CC2531.

are u able to give me a breakdown on the error so i could understand what is actually happening and possibly for me to find a fix with try and error. thanks

Koenkk, 2020-08-11:
Can you try if it works when repaired and keeping it close to the coordinator? (so that it has a direction connection)

VanQz, 2020-08-16:
I believe the documentation has been updated and its working indivisually now. Would like to ask how would i know if its a router device. Im assuming it is cause its a light switch even if its non neutral. If it is how can i verify it and how can i check to know if other devices is connected to it

Koenkk, 2020-08-17:
You can check it with https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget (`type` value)

closed 2020-09-24 by github-actions[bot]