Koenkk/zigbee2mqtt #4062: Some Tradfri dimmable bulbs sometimes incorrectly report brightness=0

emccreigh, 2020-08-08:
When I on/off toggle the only dimmable Tradfri bulbs I have, a LED1545G12 and a LED1842G3, the status report from that /set or a /get a second later reports a brightness of 0, even when it's obvious from looking at the bulb that that's wrong.  Subsequent automatic periodic status reports from the bulb show the correct brightness value.

My zigbee2mqtt is at v 1.14.2, the firmware in my CC2531 is recent, I've asked both bulbs whether they have an OTA update available, and they don't.  My testing used bash driving mosquitto_sub/mosquitto_pub.

I'm inclined to believe that this is a Tradfri firmware bug in the bulbs.

My current  workaround is never to set a brightness value of 0 in a Tradfri bulb and to ignore brightness reports of 0 from those bulbs and retain in my database the last nonzero brightness set or heard.

I can share a _sub and a _pub bash script that show the problem, but I'm guessing this problem must be well-known ... ?

Koenkk, 2020-08-08:
Please share the debug logging of this issue.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

emccreigh, 2020-08-10:
OK, today I did a "sudo systemctl restart zigbee2mqtt.service" to start a clean log, then sent a series of /set commands, four seconds apart, to my Tradfri LED1545G12 bulb (with /get's in between, just to see if anything was reported differently):
  turn off
  brightness 100
  toggle
  toggle
  toggle

Attached is a copy of the zigbee2mqtt log.txt through all of that...

[tradfri_brightness_zigbee2mqtt_log_10aug2020.txt](https://github.com/Koenkk/zigbee2mqtt/files/5049990/tradfri_brightness_zigbee2mqtt_log_10aug2020.txt)

Very interestingly, after I had grabbed that copy of the log, and about 1.3 minutes after the last log entry in that copy, a new periodic status message from that bulb appeared in log.txt, in which brightness had been corrected to 100.  Further experimentation shows that a TOGGLE (and perhaps an OFF or ON as well) sets the reported brightness to 0, but following periodic automatic status messages report the truth.

Years ago a friend commented to me that Siemens was a bank that dabbled in electronics.  I keep forgetting that IKEA is a furniture maker that dabbles in electronics.

emccreigh, 2020-08-10:
Confirming today's earlier observations, it appears that after a Tradfri bulb toggles, it reports a brightness of 0 until its next automated report a few minutes later.  Thereafter it appears that brightness is reported correctly until the next toggle.  Here's the log of all that.  This experiment sets the brightness to 100, waits for four seconds, toggles, waits five minutes, toggles, waits five minutes, toggles, and finally waits a bit more than five minutes.

[tradfri_brightness_2020aug10-1730.txt](https://github.com/Koenkk/zigbee2mqtt/files/5051818/tradfri_brightness_2020aug10-1730.txt)

This bug/feature seems to affect only brightness, and not color temperature, a similar but different bulb attribute.

Koenkk, 2020-08-10:
I assume you mean that the following is wrong (bulb is on but brightness is 0).

```
info  2020-08-10 11:22:59: MQTT publish: topic 'zigbee2mqtt_1/0x90fd9ffffed477a0', payload '{"state":"ON","linkquality":136,"brightness":0,"color_temp":370,"color":{"x":0.4599,"y":0.4106},"update_available":false}'
```

This may be the same bug as https://github.com/Koenkk/zigbee2mqtt/issues/3948 which already has been fixed but is  only in the dev branch yet.

See https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html how to switch.

emccreigh, 2020-08-11:
Thanks for making me aware of issue #3948.  This issue #4062 has similarities, but I'm not sure it's the same. On the other hand, issues #3948 and #4062 might be different symptoms of the same underlying problem.

I'm working at the MQTT level; imagine I'm only using mosquitto_sub and mosquitto_pub.  (I'm really using Perl's Net::MQTT::Simple)  I've not had any trouble *controlling* the IKEA bulbs. Turning on/off, toggling, changing brightness all works fine for me.  (I don't change brightness and state in the same message, and I know that when I set its brightness a Tradfri bulb will turn on as a side effect.)  While a Tradfri bulb is off it clearly remembers the last brightness I set, because when I turn it on again, it shines at the correct brightness.

My problem is that for up to three minutes after a state change, the bulb incorrectly reports back a brightness value of 0.  Thereafter, in its next periodic automatic status report, the bulb will begin reporting the correct brightness value again (even if its state is off).  As I mentioned earlier, my current work-around is to ignore a reported Tradfri brightness value of 0.

It's certainly possible that a fix for #3948 could also fix #4062; I'll check.

emccreigh, 2020-08-11:
Today I installed the zigbee2mqtt  dev branch and now the Tradfri bulb's brightness property seems correct immediately after a toggle to "on".  After a toggle to "off" the  brightness goes to 0 until the next automated periodic status report, after which it correctly shows the internal brightness register of the bulb.  I assume the bulb has an internal brightness register whose meaning is "how bright would I be if I were on?", and it's a mystery to me why the bulb doesn't *always* report the value in that register.  For bulbs that have color temperatures, they don't temporarily report a value of zero when you turn the bulb off.  The temporary brightness zero makes no sense.  Home automation software should be able to answer the question, "if I turned this bulb on now, how bright would it be?"  The bulb's internal brightness register shouldn't be hidden just because the bulb is off.

closed 2020-09-18 by github-actions[bot]