Koenkk/zigbee2mqtt #4066: philips bulb does not support power outrage convertor

daksh1305, 2020-08-09:
Hey @koenkk,

I am using an philips hue bulb model no 9290012574 the bulb gets connected to the stick and work flawlessly. 
The only issue which i am facing is that when there is an power outrage the bulb turns on with full brightness with its default colour (warm white).I tried passing the json ("power_on_behaviour_1": "recover") as addressed in the issue #746 but the log message says that no convertors are available. I am currently on the latest version on zigbee2mqtt (1.14.2).

Please can you help me with that.

Thanks,

Koenkk, 2020-08-09:
`power_on_behaviour_1` is not the correct payload to set this, please read the docs: https://www.zigbee2mqtt.io/devices/7146060PH.html#power-on-behavior

daksh1305, 2020-08-11:
Hey @Koenkk,

I figured it out what the issue was my bulb wasn’t updated to the latest firmware i updated it using ota update and tested it again and now the bulb retains its state after power outrage.

closed 2020-08-11 by daksh1305