Koenkk/zigbee2mqtt #4067: Xiaomi Mijia temperature/humidity sensors need occasional repairing

AivarasAukselis, 2020-08-09:
## What happened
One year and a half ago I have bought 3 Xiaomi Mijia temperature/humidity sensors from aliexpress. 1 of these sensors works flawlessly for all the time I own it. It reports temperature every few seconds (location where it is place has large temperature variation during day and night) and it still runs on the same original battery. However, other two sensors drop out of the network every few days and need repairing to get them reporting again. Furthermore, both of these sensors went through two batteries already. There are two differences between these two sensors and the one that works very well:
1. The misbehaving sensors are located in rooms where temperature variation is much smaller
2. These sensors do not connect to the gateway directly but are relayed through IKEA TRÅDFRI smart socket.
I was thinking to throw away these sensors, but now that I bought TuYa temperature and humidity sensor and had some pairing issues (see  bug #4040) I started wondering perhaps the issues with Mijia sensors can be resolved.

Perhaps someone could help me debug the issue I am having and hopefully resolve it?

Network map directly after repairing:
[image]

Network map after couple of days:
[image]

Note that Sensor (Hall) has very poor connection, linkquality 1, but works with zero issues.

## What did you expect to happen
2 of the Mijia sensors behaving like the other one.

## How to reproduce it (minimal and precise)
Pair Mijia temperature sensor so it is relayed through IKEA TRÅDFRI smart socket.

## Debug Info
Zigbee2mqtt version: various versions throughout 1.5 years including current latest development branch
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-08-09:
Make sure to check https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network and the linked post (https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623)

AivarasAukselis, 2020-08-09:
Yes I have checked those, the link quality of the disconnecting Mijia devices are always above 100. I have one Mijia device that has link quality of 1 and never disconnects. I have tried various batteries. Today I popped new VARTA batteries also and one device disconnected after half an hour another took 6h before it disconnected. As for the second link they state that relaying over tradfri plugs should work fine:
[image]

Koenkk, 2020-08-10:
Does the device keep connected when pairing and leaving it close to the coordinator?

Stoatwblr, 2020-08-12:
I've just written a fairly long posting about zigbee channelling issues - and why they matter - at https://community.openhab.org/t/zigbee-channelling-considerations-when-installing-your-zigbee-network/103455/

Is it possible you're suffering from zigbee vs wifi interference issues? (this is a particular risk if you're running on wifi ch1 and zigibee 11)

AivarasAukselis, 2020-08-13:
I looked into this more, so here it's an update.

I bought ConBee II adapter to replace my CC2531, when I replaced my adapter I had to re-pair all of the devices. After repairing I noticed that the network has much stronger connections now. And if I flip switches in homeassistant devices react almost instantly, before there was 1-2s delay. I also replaced positions of Plug 1 and Plug 3. Plug 1 is Xiaomi plug, Plug 3 is IKEA. because the not working sensors were connecting to plug 3 before, I thought maybe if it will be Xiaomi plug it will work better. The new network map is as follows:
[image]

Sensor (Hall) shows as disconnected, but it is working fine (network map bug?).

After several hours Livingroom temperature sensor disconnected again and stopped reporting. However, the bathroom sensor is still working fine. Livingroom sensor is in direct line of sight with all the plugs and the coordinator. It is also located approx 4m away from the coordinator. However, it chooses to connect to Plug 1 (previously plug 3) that is the furthest at 6m. It is the only sensor that is in direct line of sight with all of the plugs. Other sensors have line of sight with 2 plugs each.

[image]

From the graph above you see that living room sensor did report some values after being disconnected but that happened on rare occasion. Furthermore, if I would click the button on the sensor after it lost connection it would blink with two slow blinks instead of blinking rapidly like the other ones when sending data.

In upcoming days I will try out @Stoatwblr and @Koenkk  proposals and keep you updated.

closed 2020-10-01 by github-actions[bot]