Koenkk/zigbee2mqtt #4068: Problem setting color of RGB Gledopto using RGB values

ivanfmartinez, 2020-08-09:
# Bug Report
## What happened
I have tried to set the color of my led strip using rgb values  (GLEDOPTO RGB+CTT 2ID):
``` 
/zigbee2mqtt/led_rgb_cct_002/rgb/set -m '{ "brightness" : 20, "color" : {"rgb":"0,255,0"}}'
/zigbee2mqtt/led_rgb_cct_002/rgb/set -m '{ "brightness" : 20, "color" : {"rgb":"255,0,0"}}'
```

Using the first value as 255 the STRIPE becomes GREEN, and when using second value 255 if becomes RED.

When using third value as 255 it becomes blue as expected.

same behaviour if I use hex : "#rrggbb", or sending separate r,g,b values

## What did you expect to happen

Using the first value as 255 I expect to see the stripe as RED, and second value 255 to become GREEN.

## How to reproduce it (minimal and precise)

Set the color using RGB values.

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531

ivanfmartinez, 2020-08-09:
I have tested with some hsb values and appear that green and red is swapped also.

Koenkk, 2020-08-09:
@ivanfmartinez I've seen this earlier, it seems that the Gledopto firmware has some bug swapping the R and G ports. I would recommend switching the R and G wires. (we don't have individual control of the channels from zigbee2mqtt, zigbee2mqtt just send a color value in x/y, the led controller itself is responsible for translating this to the channels so this cannot be fixed form zigbee2mqtt)

ivanfmartinez, 2020-08-09:
Thanks, I will do this.

closed 2020-08-09 by ivanfmartinez