Koenkk/zigbee2mqtt #4072: Distance Osram Plug

BremerButjer, 2020-08-10:
This is my first post, i hope this is the right place for that...

Here's the situation:

Zigbee2mqtt runs successfully on a Raspberry4 with CC2531 Coordinator Stick and the current FW.

The following power socket has been put into operation:
https://www.zigbee2mqtt.io/devices/AC10691.html
(Osram Smart+ Plug)

The socket is quite far from the coordinator, but 1.5m away from the nearest Zigbee lamp (supported Tradfri Bulb).
The Bulb can be switched without problems, but the switching commands to the Osram Plug do not get through regularly. It seems that the switching commands are not forwarded through the routers (e.g. the Tradfri Bulb), but only pass through when directly connected to the coordinator.
Is there a possibility to check this or to change it if necessary?

Furthermore I have once addressed the power socket directly via Zigbee protocol (without MQTT) and it still returns the value of the current power consumption. Is there a chance to get this value transmitted in the MQTT?

thx a lot for response

Koenkk, 2020-08-10:
Probably same issue as https://github.com/Koenkk/zigbee2mqtt/issues/3910

Switching to the source routing firmware has solved it in that case: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

BremerButjer, 2020-08-11:
Thanks for the quick answer.
Unfortunately I do not have a flasher for the stick. There is no other way to exchange the firmware on the stick?

Is there a quick way to view the network map?  I would have to see with whom the plug has a connection.

Koenkk, 2020-08-11:
There is no other way to flash.

For the networkmap see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap

closed 2020-09-18 by github-actions[bot]