Koenkk/zigbee2mqtt #4073: Network map comes up without routes but used to work

ayavilevich, 2020-08-10:
# Bug Report
## What happened

Asked to generate a network map using: zigbee2mqtt/bridge/networkmap/routes and graphviz

Got back data without routes and lqi. Just individual components.

```
Aug 10 12:57:48 hassbian npm[7092]: zigbee2mqtt:info  2020-08-10 12:57:48: Starting network scan (includeRoutes 'true')
Aug 10 12:57:49 hassbian npm[7092]: zigbee2mqtt:info  2020-08-10 12:57:49: Network scan finished
Aug 10 12:57:49 hassbian npm[7092]: zigbee2mqtt:info  2020-08-10 12:57:49: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
Aug 10 12:57:49 hassbian npm[7092]: node[shape=record];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00124b000e896a31" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b000e896a31 (0)|0 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00124b000e896a31" -> "0x00124b0018559a2e" [penwidth=0.5, weight=0, color="#994444", label="1"]
Aug 10 12:57:49 hassbian npm[7092]:   "0x00158d0001de86d7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{door_main|0x00158d0001de86d7 (6195)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|26 minutes, 25 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00158d00016c45a0" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{button_a|0x00158d00016c45a0 (48627)|Xiaomi Aqara wireless switch (WXKG11LM)|14 minutes, 25 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00158d0001e160e1" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{pir_office|0x00158d0001e160e1 (44052)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|7 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00158d0001718b49" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{switch_a|0x00158d0001718b49 (31603)|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|12 minutes, 47 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00124b0018559a2e" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{zigbee_router|0x00124b0018559a2e (30568)|Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)|2 seconds ago}"];
Aug 10 12:57:49 hassbian npm[7092]:   "0x00124b0018559a2e" -> "0x00124b000e896a31" [penwidth=0.5, weight=0, color="#994444", label="1"]
Aug 10 12:57:49 hassbian npm[7092]: }'
```

In the past I have gotten complete maps with the missing data. It is possible that at that time I was using an older firmware (2018) on the coordinator adapter and that once I updated to latest then it stopped working. There was no change to the devices in the network from the last time it worked.
Should 20190608 adapter firmware support network scans with routes?
Should the scan complete so fast?

## What did you expect to happen

Expected to get a map with routes and lqi information like I used to get in the past.

## How to reproduce it (minimal and precise)

AFAIK just request zigbee2mqtt/bridge/networkmap/routes and graphviz

## Debug Info
Zigbee2mqtt version:  1.14.2
Adapter hardware: CC2531
Adapter firmware version: 20190608

'{"version":"1.14.2","commit":"b91b456","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'

ayavilevich, 2020-08-10:
I just re-paired a device and for that device there is now a route in the network map.

lolorc, 2020-09-10:
same here, repairing 40+ devices just to get a working network map is quite of a hassle  :-)

closed 2020-09-18 by github-actions[bot]

Snoozy01, 2021-08-24:
> I just re-paired a device and for that device there is now a route in the network map.

Hello, same for me. I did it with 3 different test devices with same positive result.