Koenkk/zigbee2mqtt #4076: Sage by Hughes Doorbell Sensor Issues

liamstears, 2020-08-10:
I seem to be having some issues with this device in particular, it will not show battery level and it constantly seems to be doing something as it shows up in logs a lot which I think is unusual I would expect it only to come up in logs when the doorbell is pushed?

So first here's my database entry:

```
{"id":7,"type":"EndDevice","ieeeAddr":"0x0008890001d9b4a5","nwkAddr":48989,"manufId":4116,"manufName":" Echostar","powerSource":"DC Source","modelId":"   Bell","epList":[18],"endpoints":{"18":{"profId":260,"epId":18,"devId":260,"inClusterList":[0,3,9,1],"outClusterList":[3,6,8,25],"clusters":{"genBasic":{"attributes":{"modelId":"   Bell","manufacturerName":" Echostar","powerSource":4,"zclVersion":1,"appVersion":2,"stackVersion":3,"hwVersion":4,"dateCode":" 05/21/15","swBuildId":"   3.2.2"}}},"binds":[]}},"appVersion":2,"stackVersion":3,"hwVersion":4,"dateCode":" 05/21/15","swBuildId":"   3.2.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1596751055206}

```

What I notice in this is that it shows DC Source as power instead of Battery?

And here is some debug logging:

```
2020-08-10T19:17:06.935Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":2,"relaylist":[58789,27046]}
2020-08-10T19:17:06.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:06.958Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:06.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:06.959Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
2020-08-10T19:17:06.959Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":2,"relaylist":[58789,27046]}
2020-08-10T19:17:06.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:07.934Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:07.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:07.935Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
2020-08-10T19:17:07.935Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":2,"relaylist":[58789,27046]}
2020-08-10T19:17:07.936Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:11.824Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:11.825Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:11.825Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
2020-08-10T19:17:11.825Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":2,"relaylist":[58789,27046]}
2020-08-10T19:17:11.826Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:11.884Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,93,191,18,13,0,120,0,81,13,107,0,0,12,1,221,1,0,20,16,2,0,9,0,0,0,166,105,7,88]
2020-08-10T19:17:11.885Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,93,191,18,13,0,120,0,81,13,107,0,0,12,1,221,1,0,20,16,2,0,9,0,0,0,166,105,7,88]
2020-08-10T19:17:11.885Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,93,191,18,13,0,120,0,81,13,107,0,0,12,1,221,1,0,20,16,2,0,9,0,0,0,166,105,7] - 88
2020-08-10T19:17:11.886Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48989,"srcendpoint":18,"dstendpoint":13,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":7015761,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,221,1,0,20,16,2,0,9,0,0,0]}}
2020-08-10T19:17:11.889Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":221,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}},"address":48989,"endpoint":18,"linkquality":120,"groupID":0}'
Zigbee2MQTT:debug 2020-08-10 20:17:11: Received Zigbee message from '0x0008890001d9b4a5', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}' from endpoint 18 with groupID 0
2020-08-10T19:17:11.892Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:11.893Z zigbee-herdsman:controller:endpoint CommandResponse 0x0008890001d9b4a5/18 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-08-10T19:17:11.895Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48989,"destendpoint":18,"srcendpoint":1,"clusterid":25,"transid":131,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,100,2,152]}}
2020-08-10T19:17:11.895Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,93,191,18,1,25,0,131,0,30,4,25,100,2,152,189]
2020-08-10T19:17:11.902Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-08-10T19:17:11.902Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-08-10T19:17:11.903Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-08-10T19:17:11.903Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-08-10T19:17:11.903Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:11.915Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,131,69]
2020-08-10T19:17:11.915Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,131,69]
2020-08-10T19:17:11.916Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,131] - 69
2020-08-10T19:17:11.916Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":131}
2020-08-10T19:17:11.916Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:13.587Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:17:13.587Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,38,106,1,166,105,6]
2020-08-10T19:17:13.588Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [38,106,1,166,105] - 6
2020-08-10T19:17:13.588Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":27174,"relaycount":1,"relaylist":[27046]}
2020-08-10T19:17:13.588Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:17.136Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:17.137Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,93,191,2,165,229,166,105,233]
2020-08-10T19:17:17.137Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [93,191,2,165,229,166,105] - 233
2020-08-10T19:17:17.137Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48989,"relaycount":2,"relaylist":[58789,27046]}
2020-08-10T19:17:17.138Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-08-10T19:17:17.197Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,93,191,18,13,0,118,0,46,78,107,0,0,12,1,222,1,0,20,16,2,0,9,0,0,0,166,105,7,105]
2020-08-10T19:17:17.197Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,93,191,18,13,0,118,0,46,78,107,0,0,12,1,222,1,0,20,16,2,0,9,0,0,0,166,105,7,105]
2020-08-10T19:17:17.197Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,93,191,18,13,0,118,0,46,78,107,0,0,12,1,222,1,0,20,16,2,0,9,0,0,0,166,105,7] - 105
2020-08-10T19:17:17.198Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":48989,"srcendpoint":18,"dstendpoint":13,"wasbroadcast":0,"linkquality":118,"securityuse":0,"timestamp":703236 [... cut ...]
2020-08-10T19:17:26.291Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":133}
2020-08-10T19:17:26.291Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

```

The sensor does work so I'm happy about that but it would be nice to get the battery to update correctly and get rid of all these annoying logs from the device if possible as it makes it a lot harder when trying to fix another device (like we are trying to fix smartthings sensor now but I have trouble sorting through these logs to find the right ones) Thanks in advance for any help with this

Koenkk, 2020-08-11:
I'm not sure if we can get the device to silence itself, it seems to be constantly sending OTA request (which zigbee2mqtt responds to not being available).

For the battery try changing the entry to:

```js
{
    zigbeeModel: ['   Bell'],
    model: 'SAGE206612',
    vendor: 'EchoStar',
    description: 'SAGE by Hughes doorbell sensor',
    supports: 'action',
    fromZigbee: [fz.SAGE206612_state, fz.battery_3V],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        await bind(device.getEndpoint(18), coordinatorEndpoint, ['genPowerCfg']);
    },
},
```

liamstears, 2020-08-11:
Have tried this the same way as I have been to debug the arrival sensor and it doesn't seem to have worked

Battery is still not showing a percentage, whats also odd is before changing anything I had over 100lqi and now we the change its only 28lqi

liamstears, 2020-08-12:
Scrap that lqi seems fine

Looks line battery isn't sent at all, when its activated it sends an on signal and lqi only, see logs:

```
info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5', payload '{"action":"on","linkquality":115}'
info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5', payload '{"linkquality":115,"action":""}'
info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5/action', payload 'on'

2020-08-12T00:11:43.154Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":40167,"srcendpoint":18,"dstendpoint":13,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":15647053,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,25,1,0,20,16,2,0,9,0,0,0]}}
2020-08-12T00:11:43.155Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":25,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}},"address":40167,"endpoint":18,"linkquality":115,"groupID":0}'

```

slippern, 2020-08-16:
> Scrap that lqi seems fine
> 
> Looks like battery isn't sent at all, when its activated it sends an "on" signal and lqi only, see logs:
> 
> ```
> info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5', payload '{"action":"on","linkquality":115}'
> info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5', payload '{"linkquality":115,"action":""}'
> info  2020-08-12 01:10:19: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9b4a5/action', payload 'on'
> 
> 
> 2020-08-12T00:11:43.154Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":40167,"srcendpoint":18,"dstendpoint":13,"wasbroadcast":0,"linkquality":115,"securityuse":0,"timestamp":15647053,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,25,1,0,20,16,2,0,9,0,0,0]}}
> 2020-08-12T00:11:43.155Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":25,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4116,"imageType":2,"fileVersion":9}},"address":40167,"endpoint":18,"linkquality":115,"groupID":0}'
> ```

I cant se the battery status in my logs either.. 

igbee2MQTT:info  2020-08-16 11:44:53: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9aec6', payload '{"linkquality":39,"action":"on"}'
Zigbee2MQTT:info  2020-08-16 11:44:53: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9aec6', payload '{"linkquality":39,"action":""}'
Zigbee2MQTT:info  2020-08-16 11:44:53: MQTT publish: topic 'homeassistant/device_automation/0x0008890001d9aec6/action_on/config', payload '{"automation_type":"trigger","type":"action","subtype":"on","payload":"on","topic":"zigbee2mqtt/0x0008890001d9aec6/action","device":{"identifiers":["zigbee2mqtt_0x0008890001d9aec6"],"name":"0x0008890001d9aec6","sw_version":"Zigbee2MQTT 1.14.3","model":"SAGE by Hughes doorbell sensor (SAGE206612)","manufacturer":"EchoStar"}}'
Zigbee2MQTT:info  2020-08-16 11:44:53: MQTT publish: topic 'zigbee2mqtt/0x0008890001d9aec6/action', payload 'on'

closed 2020-09-23 by github-actions[bot]