Koenkk/zigbee2mqtt #4077: Is it possible to add a new Paul Neuhaus Q-REMOTE device

fritzlc, 2020-08-10:
Remote controler
Neuhaus Lighting Group
Type: E0040007

[https://zigbeealliance.org/zigbee_products/q-cct-remote-controller/](url)
[https://www.paul-neuhaus.de/q-anleitungen/](url)
[https://www.paul-neuhaus.de/fileadmin/user_upload/paul-neuhaus/Anleitungen_Q-Leuchten/Anleitung_1.2.pdf](url)
[https://www.youtube.com/watch?v=2sEVUHe1OA8](url)

sie 10 14:51:30 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:30: Successfully interviewed '0x00158d0002f3112b', device has successfully been paired
sie 10 14:51:30 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:30: Device '0x00158d0002f3112b' is supported, identified as: Paul Neuhaus Q-REMOTE (100.462.31)
sie 10 14:51:30 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0002f3112b","model":"100.462.31","vendor":"Paul Neuhaus","description":"Q-REMOTE","supported":true}}'
sie 10 14:51:30 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0014d96ac5","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1597063890307},{"ieeeAddr":"0x00158d0002f3112b","type":"EndDevice","networkAddress":10339,"model":"100.462.31","vendor":"Paul Neuhaus","description":"Q-REMOTE","friendly_name":"0x00158d0002f3112b","manufacturerID":4151,"manufacturerName":"Neuhaus Lighting Group ","powerSource":"Battery","modelID":"NLG-remote ","hardwareVersion":1,"softwareBuildID":"0170-0103","dateCode":"20150901","lastSeen":1597063890134}]}'
sie 10 14:51:32 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:32: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":65,"action":"on","action_group":2}'
sie 10 14:51:44 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:51:44: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":73,"action":"off","action_group":2}'
sie 10 14:52:23 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:52:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0002f3112b"}}'
sie 10 14:52:23 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:52:23: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":78,"action":"color_temperature_move","action_color_temperature":286,"action_transition_time":5,"action_group":2}'
sie 10 14:52:25 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:52:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":76,"action":"color_temperature_move","action_color_temperature":206,"action_transition_time":5,"action_group":2}'
sie 10 14:52:25 raspberrypi npm[6399]: zigbee2mqtt:info  2020-08-10 14:52:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":84,"action":"color_temperature_move","action_color_temperature":229,"action_transition_time":5,"action_group":2}'

```

**MQTT messages from device**
info  2020-08-10 14:58:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":84,"action":"off","action_group":1}'
info  2020-08-10 14:58:43: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":81,"action":"on","action_group":2}'
info  2020-08-10 14:58:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":78,"action":"off","action_group":3}'
info  2020-08-10 14:58:46: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":84,"action":"color_temperature_move","action_color_temperature":286,"action_transition_time":5,"action_group":3}'
info  2020-08-10 14:58:50: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":84,"action":"color_temperature_move","action_color_temperature":286,"action_transition_time":5,"action_group":1}'
info  2020-08-10 14:58:52: MQTT publish: topic 'zigbee2mqtt/0x00158d0002f3112b', payload '{"linkquality":84,"action":"color_temperature_move","action_color_temperature":286,"action_transition_time":5,"action_group":2}'

Koenkk, 2020-08-11:
sie 10 14:51:30 raspberrypi npm[6399]: zigbee2mqtt:info 2020-08-10 14:51:30: Device '0x00158d0002f3112b' is **supported, identified as: Paul Neuhaus Q-REMOTE (100.462.31)**

It is already?

fritzlc, 2020-08-16:
ohh i see 
zigbee_to_mqtt "is supported, identified as: Paul Neuhaus Q-REMOTE (100.462.31)"
but
domoticz-zigbee2mqtt-plugin does not support zigbee device with model "100.462.31" yet
sorry ;)

closed 2020-09-23 by github-actions[bot]