Koenkk/zigbee2mqtt #4079: Unable to pair Sonoff motion sensor SNZB-03

Gertjuh, 2020-08-11:
# Bug Report
I have tried for 2 weeks now to pair the Sonoff SNZB-3 motion sensor with no succes. Thant why I am seeking for help.

## What happened
I can see in the logfiles the sensor is detected, but after a while it 'failed to interview <mac address>":

zigbee2mqtt:info  2020-08-05 14:18:08: MQTT publish: topic 'zigbee2mqtt/0x00124b001faf4b06', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":110,"battery":80,"voltage":2900}'
zigbee2mqtt:info  2020-08-05 14:18:11: Starting interview of '0x00124b001faf4b06'
zigbee2mqtt:info  2020-08-05 14:18:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001faf4b06"}}'
zigbee2mqtt:error 2020-08-05 14:20:30: Failed to interview '0x00124b001faf4b06', device has not successfully been paired
zigbee2mqtt:info  2020-08-05 14:20:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001faf4b06"}}'

In database.db:
{"id":54,"type":"EndDevice","ieeeAddr":"0x00124b001faf4b06","nwkAddr":44471,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"ms01","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":3,"hwVersion":1,"dateCode":"20191107","zclVersion":1,"interviewCompleted":false,"meta":{"configured":1},"lastSeen":1596788367251}

I also have deleted the entry in the database.db, devices.yaml and state.json and after a reboot tried it again with the sam result.

I am using a CC2531.

What other things can I try?

Ciubas, 2020-08-11:
I have pared it, but the motion detection don't work. 
[image]

Koenkk, 2020-08-11:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

Ciubas, 2020-08-11:
The LQ, voltage and battery works well. Worsk well on Tuya and Sonoff Zigbee Bridge. 

Yes i pared few times, there is only problem with that devices (its 50cm from cc), battery is ok.

Gertjuh, 2020-08-12:
> Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ?

Yes, I did all that is mentioned in that FAQ, Koen.

Koenkk, 2020-08-12:
@Gertjuh could you share the zigbee herdsman debug logging when pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Gertjuh, 2020-08-17:
Hi,
It took me a few days to figure out why pairing this device was not functioning at all anymore. With debugging on I never saw the device anymore and with debugging off I managed to have one unsuccesful attempt.

Now I think I have so much data to handle with 54 devices that the SNZB-03 was not able to give a proper signal-sequence. Does that make sense?
Just 15 minutes ago I held the device 10cm from the (external) antenna and now pairing has succeeded!

Here is the log information:

info  2020-08-17 14:30:36: Device '0x00124b001faf4b06' joined
info  2020-08-17 14:30:36: Starting interview of '0x00124b001faf4b06'
info  2020-08-17 14:30:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001faf4b06"}}'
info  2020-08-17 14:30:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001faf4b06"}}'
debug 2020-08-17 14:30:36: Received Zigbee message from '0x04cf8cdf3c77614a', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":21173}' from endpoint 1 with groupID 0
info  2020-08-17 14:30:36: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77614a', payload '{"illuminance":21173,"illuminance_lux":130,"linkquality":102,"battery":100,"voltage":3100}'
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Received Zigbee message from '0x00124b001faf4b06', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":1,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-08-17 14:30:37: Skipping message, definition is undefined and still interviewing
debug 2020-08-17 14:30:37: Device '0x00124b001faf4b06' announced itself
info  2020-08-17 14:30:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001faf4b06"}}'

debug 2020-08-17 14:31:05: Received Zigbee message from '0x00124b001faf4b06', type 'readResponse', cluster 'genBasic', data '{"modelId":"ms01"}' from endpoint 1 with groupID 0
debug 2020-08-17 14:31:05: No converter available for 'SNZB-03' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"ms01"}'

debug 2020-08-17 14:31:05: Received Zigbee message from '0x00124b001faf4b06', type 'readResponse', cluster 'genBasic', data '{"modelId":"ms01"}' from endpoint 1 with groupID 0
debug 2020-08-17 14:31:05: No converter available for 'SNZB-03' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"ms01"}'
info  2020-08-17 14:31:05: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b001faf4b06/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x00124b001faf4b06","json_attributes_topic":"zigbee2mqtt/0x00124b001faf4b06","name":"0x00124b001faf4b06_occupancy","unique_id":"0x00124b001faf4b06_occupancy_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001faf4b06"],"name":"0x00124b001faf4b06","sw_version":"Zigbee2MQTT 1.14.3","model":"Motion sensor (SNZB-03)","manufacturer":"SONOFF"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-08-17 14:31:05: MQTT publish: topic 'homeassistant/sensor/0x00124b001faf4b06/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00124b001faf4b06","json_attributes_topic":"zigbee2mqtt/0x00124b001faf4b06","name":"0x00124b001faf4b06_battery","unique_id":"0x00124b001faf4b06_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001faf4b06"],"name":"0x00124b001faf4b06","sw_version":"Zigbee2MQTT 1.14.3","model":"Motion sensor (SNZB-03)","manufacturer":"SONOFF"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...]
info  2020-08-17 14:31:12: Device '0x00124b001faf4b06' is supported, identified as: SONOFF Motion sensor (SNZB-03)
info  2020-08-17 14:31:12: Configuring '0x00124b001faf4b06'
info  2020-08-17 14:31:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001faf4b06","model":"SNZB-03","vendor":"SONOFF","description":"Motion sensor","supported":true}}'

info  2020-08-17 14:31:13: Successfully configured '0x00124b001faf4b06'

Gertjuh, 2020-09-10:
I have an update on this issue.

I frequenly had some problems with the responsiveness of devices. I always had a lag between pressing a button and switching on a light. I already added a cc2531 with antenna as a router in my zigbee network. This helped a little bit. 
Now I know I wasn't far of with the suspection that it had to do something with a busy network.
This weekend I saw some error messages in the Zigbee2mqtt logfiles.

Zigbee2MQTT:error 2020-09-08 11:25:34: Publish 'set' 'state' to '0x00158d000224a23e' failed: 'Error: Command 0x00158d000224a23e/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 37185 - 1 - 40 - 6 - 11 after 10000ms)'

It seems related to a busy network. After searching on the Internet for a solution I saw this [picture](https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence) about the coexistance wifi and Zigbee. 

Then saw a potential issue. After the problem described in this issue my neighbor solved his Wifi problems by boosting up the signal last week. After that the problems a had went worse. I saw his wifi was on channel 1. My Zigbee network was on channel 11. The same frequency part. I the decided to change my Zigbee network to channel 26. Yes, I was aware of the fact that I had to re-pair all my devices. And that was the solution for me. Pairing of the 54 devices went very good. Anf now the whole Zigbee network is very responsive. No problems anymore. I wish I had done this in the beginning (2 years ago). That would have saved me many hours of troubleshooting end frustration.

closed 2020-10-18 by github-actions[bot]