Koenkk/zigbee2mqtt #4081: SNZB-04 wrong identified

mrslvd, 2020-08-11:
Hello,
I got SNZB-04 (window/door sensor) device and after pairing it to zigbee2mqtt it is identified as SNZB-02 (humidity and temperature sensor). I am using zigbee2mqtt dongle connected to Home Assistant. 
Can somebody give me an advice how to proceed?

Best regards

Koenkk, 2020-08-11:
Possible duplicate of https://github.com/Koenkk/zigbee-herdsman-converters/issues/1449

Should already be fixed in latest dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

mrslvd, 2020-08-12:
Thanks!
I found those threads later yesterday and forgot to tell you. Sorry for that. I install zigbee2mqtt-edge and now the SNZB-04 is recognized.
I am not sure whether this is the right way but for me it was the easiest. Thanks for your wonderful support!
Best regards,
Miro

closed 2020-08-12 by Koenkk