Koenkk/zigbee2mqtt #4082: Xiaomi RTCGQ11LM motion sensor does not report battery

bartek75, 2020-08-11:
# Bug Report
## What happened
No battery information in messages:
info  2020-08-11 20:47:25: MQTT publish: topic 'zigbee2mqtt/0x00158d00053ef89e', payload '{"illuminance":3,"illuminance_lux":3,"linkquality":86,"occupancy":true}' 
## What did you expect to happen
Battery information to be in payload
## How to reproduce it (minimal and precise)
Normal use: trigger motion
## Debug Info
Zigbee2mqtt version:  1.14.2 (commit #faaf3e4)
Adapter hardware: CC2531
Adapter firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"main
trel":3,"revision":20190608}}'

juliend20, 2020-08-11:
Hello, I have two xiaomi aqara. I confirm i have the battery information. But the sensor doesn't send this information all the time. Wait a few message from this sensor you should see the battery information. 
Best

bartek75, 2020-08-12:
juliend20 is right, thx.

closed 2020-08-12 by bartek75