Koenkk/zigbee2mqtt #4086: CC2531 - ZB 3.0, panid/epanid issues

Stoatwblr, 2020-08-12:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Stoatwblr, 2020-08-12:
After about 10 minutes of operation, this gets logged and at this point everything stops working entirely

I'm assuming the stick is "bad", but what I don't know is whether that's firmware or hardware

```
zigbee2mqtt:warn  2020-08-12 21:18:34: Soft reset timeout triggered
zigbee2mqtt:warn  2020-08-12 21:18:35: Soft resetted ZNP due to timeout
zigbee2mqtt:debug 2020-08-12 21:19:41: Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:debug 2020-08-12 21:19:46: Non-pingable device 'sensor_bathroom_quarterlight' was last seen '565.057' seconds ago.
(node:2936072) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0xc2: NWK_INVALID_REQUEST)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:290:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:2936072) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2936072) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

Stoatwblr, 2020-08-12:
other notes:    0x00124B001CD48A18 is the MAC address of the device (why would the epanID be forced to this?)

Changing back to 1.2HA seems to work but the panid is still broken

Koenkk, 2020-08-13:
Can you provide the zigbee2mqtt debug startup logging and your configuration.yaml?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Stoatwblr, 2020-08-13:
I've flashed the device back to 1.2HA. Will have to reflash and then hope it breaks again.... :) This may take a while

Stoatwblr, 2020-08-13:
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: zigbee2mqtt
  password: xxxxxxxx
  include_device_information: true
  keepalive: 60
  version: 5
serial:
  port: /dev/ttyUSB.CC2531-01
  disable_led: false
advanced:
  baudrate: 115200
  rtscts: false
  soft_reset_timeout: 300
  last_seen: 'epoch'
  channel: 15
  cache_state: true
  cache_state_persistent: true
  log_level: debug
#  log_output: syslog
  pan_id: 0x1a62
#### 6754
  ext_pan_id: [0xC3, 0x8D, 0xF4, 0x34, 0xD1, 0xC8, 0xC3, 0x8B]
###C3 8D F4 34 D1 C8 C3 8B
  network_key:
    - 156
    - 161
    - 36
    - 139
    - 219
    - 66
    - 182
    - 200
    - 210
    - 248
    - 84
    - 56
    - 205
    - 81
    - 17
    - 133
###9c a1 24 8b db 42 b6 c8 d2 f8 54 38 cd 51 11 85
  availability_timeout: 300
  report: true

Stoatwblr, 2020-08-13:
```

zigbee2mqtt:info  2020-08-13 20:15:27: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-13.20-15-27' filename: log.txt
zigbee2mqtt:debug 2020-08-13 20:15:27: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-08-12.02-29-24'
zigbee2mqtt:debug 2020-08-13 20:15:27: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-08-13 20:15:27: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
zigbee2mqtt:info  2020-08-13 20:15:27: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-08-13 20:15:27: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[195,141,244,52,209,200,195,139],"channelList":[15],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB.CC2531-01"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-08-13 20:15:33: zigbee-herdsman started
zigbee2mqtt:info  2020-08-13 20:15:33: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:debug 2020-08-13 20:15:33: Zigbee network parameters: {"panID":6754,"extendedPanID":"0x8bc3c8d134f48dc3","channel":15}
zigbee2mqtt:info  2020-08-13 20:15:33: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-08-13 20:15:33: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-08-13 20:15:33: Allowing new devices to join.
zigbee2mqtt:warn  2020-08-13 20:15:33: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-08-13 20:15:33: Zigbee: allowing new devices to join.
zigbee2mqtt:debug 2020-08-13 20:15:33: Soft reset timeout set to 300 seconds
zigbee2mqtt:info  2020-08-13 20:15:33: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:debug 2020-08-13 20:15:33: Using MQTT keepalive: 60
zigbee2mqtt:info  2020-08-13 20:15:33: Connected to MQTT server
zigbee2mqtt:info  2020-08-13 20:15:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-08-13 20:15:33: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.2","commit":"b91b456","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":true}'
```

Stoatwblr, 2020-08-13:
This is an ikea tradfri light joining from reset

```

zigbee2mqtt:info  2020-08-13 20:20:13: Device '0x680ae2fffe9adecc' joined
zigbee2mqtt:info  2020-08-13 20:20:13: Starting interview of '0x680ae2fffe9adecc'
zigbee2mqtt:info  2020-08-13 20:20:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x680ae2fffe9adecc"}}'
zigbee2mqtt:info  2020-08-13 20:20:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x680ae2fffe9adecc"}}'
zigbee2mqtt:info  2020-08-13 20:20:13: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe9adecc/availability', payload 'online'
zigbee2mqtt:debug 2020-08-13 20:20:27: Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:error 2020-08-13 20:22:31: Failed to interview '0x680ae2fffe9adecc', device has not successfully been paired
zigbee2mqtt:info  2020-08-13 20:22:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x680ae2fffe9adecc"}}'
zigbee2mqtt:debug 2020-08-13 20:25:27: Saving state to file /opt/zigbee2mqtt/data/state.json

zigbee2mqtt:warn  2020-08-13 20:27:31: Soft reset timeout triggered
zigbee2mqtt:warn  2020-08-13 20:27:32: Soft resetted ZNP due to timeout
```

Stoatwblr, 2020-08-13:
I'm assuming to some extent that the dongle itself is the problem, but without another one to compare with, it's impossible to tell. There's one on order but it'll take a while to show up

Stoatwblr, 2020-08-13:
Doh, I've just realised that was 1.2. Reflashing to 3.0 again

Stoatwblr, 2020-08-13:
```
zigbee2mqtt:info  2020-08-13 23:36:38: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-13.23-36-38' filename: log.txt
zigbee2mqtt:debug 2020-08-13 23:36:38: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-08-13 23:36:38: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
zigbee2mqtt:info  2020-08-13 23:36:38: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-08-13 23:36:38: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[195,141,244,52,209,200,195,139],"channelList":[15],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB.CC2531-01"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-08-13 23:36:53: zigbee-herdsman started
zigbee2mqtt:info  2020-08-13 23:36:53: Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}}'
zigbee2mqtt:debug 2020-08-13 23:36:53: Zigbee network parameters: {"panID":6754,"extendedPanID":"0x00124b001cd48a18","channel":15}
zigbee2mqtt:info  2020-08-13 23:36:53: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-08-13 23:36:53: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-08-13 23:36:53: Allowing new devices to join.
zigbee2mqtt:warn  2020-08-13 23:36:53: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-08-13 23:36:53: Zigbee: allowing new devices to join.
zigbee2mqtt:debug 2020-08-13 23:36:53: Soft reset timeout set to 300 seconds
zigbee2mqtt:info  2020-08-13 23:36:53: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:debug 2020-08-13 23:36:53: Using MQTT keepalive: 60
zigbee2mqtt:info  2020-08-13 23:36:54: Connected to MQTT server
zigbee2mqtt:info  2020-08-13 23:36:54: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-08-13 23:36:54: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.2","commit":"b91b456","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":true}'
```

As you can see above, the epanid isn't what it should be and matches the mac of the dongle

pairing a device:

```
zigbee2mqtt:info  2020-08-13 23:38:04: Device '0x00158d00049fd8e1' joined
zigbee2mqtt:info  2020-08-13 23:38:04: Starting interview of '0x00158d00049fd8e1'
zigbee2mqtt:info  2020-08-13 23:38:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00049fd8e1"}}'
zigbee2mqtt:info  2020-08-13 23:38:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00049fd8e1"}}'
zigbee2mqtt:info  2020-08-13 23:38:04: MQTT publish: topic 'zigbee2mqtt/0x00158d00049fd8e1/availability', payload 'online'
zigbee2mqtt:debug 2020-08-13 23:38:05: Device '0x00158d00049fd8e1' announced itself
zigbee2mqtt:info  2020-08-13 23:38:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00049fd8e1"}}'
zigbee2mqtt:debug 2020-08-13 23:38:05: Received Zigbee message from '0x00158d00049fd8e1', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_magnet.aq2","appVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-13 23:38:05: Received Zigbee message from '0x00158d00049fd8e1', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2995,"3":35,"4":424,"5":97,"6":[0,1],"10":0,"100":1}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-13 23:38:05: MQTT publish: topic 'zigbee2mqtt/0x00158d00049fd8e1', payload '{"battery":97,"voltage":2995,"contact":false,"linkquality":28,"device":{"friendlyName":"0x00158d00049fd8e1","model":"MCCGQ11LM","ieeeAddr":"0x00158d00049fd8e1","networkAddress":15081,"applicationVersion":3},"last_seen":1597358285953}'
zigbee2mqtt:debug 2020-08-13 23:38:22: Received Zigbee message from '0x00158d00049fd8e1', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2995,"3":35,"4":424,"5":98,"6":[0,0],"10":0,"100":1},"modelId":"lumi.sensor_magnet.aq2"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-13 23:38:22: MQTT publish: topic 'zigbee2mqtt/0x00158d00049fd8e1', payload '{"battery":97,"voltage":2995,"contact":false,"linkquality":21,"device":{"friendlyName":"0x00158d00049fd8e1","model":"MCCGQ11LM","ieeeAddr":"0x00158d00049fd8e1","networkAddress":15081,"type":"EndDevice","manufacturerID":4151,"applicationVersion":3},"last_seen":1597358302861}'
zigbee2mqtt:debug 2020-08-13 23:38:25: Received Zigbee message from '0x00158d00049fd8e1', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2995,"3":35,"4":424,"5":98,"6":[0,0],"10":0,"100":1},"modelId":"lumi.sensor_magnet.aq2"}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-08-13 23:38:25: MQTT publish: topic 'zigbee2mqtt/0x00158d00049fd8e1', payload '{"battery":97,"voltage":2995,"contact":false,"linkquality":44,"device":{"friendlyName":"0x00158d00049fd8e1","model":"MCCGQ11LM","ieeeAddr":"0x00158d00049fd8e1","networkAddress":15081,"type":"EndDevice","manufacturerID":4151,"applicationVersion":3},"last_seen":15973 [... cut ...]
zigbee2mqtt:info  2020-08-13 23:56:22: Successfully setup reporting for '0x680ae2fffe9adecc' - 1 - genOnOff
zigbee2mqtt:debug 2020-08-13 23:56:22: Setup reporting for '0x680ae2fffe9adecc' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-08-13 23:56:22: Successfully setup reporting for '0x680ae2fffe9adecc' - 1 - genLevelCtrl
zigbee2mqtt:debug 2020-08-13 23:56:22: Setup reporting for '0x680ae2fffe9adecc' - 1 - lightingColorCtrl
zigbee2mqtt:info  2020-08-13 23:56:22: Successfully setup reporting for '0x680ae2fffe9adecc' - 1 - lightingColorCtrl
```

Koenkk, 2020-08-14:
Can you provide the herdsman debug logging when starting zigbee2mqtt?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Stoatwblr, 2020-08-14:
```
openhab@Riffraff:/opt/zigbee2mqtt$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.14.2 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-08-14 13:18:19: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-14.13-18-19' filename: log.txt
zigbee2mqtt:debug 2020-08-14 13:18:19: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-08-14 13:18:20: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
zigbee2mqtt:info  2020-08-14 13:18:20: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-08-14 13:18:20: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[195,141,244,52,209,200,195,139],"channelList":[15],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB.CC2531-01"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyUSB.CC2531-01' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[156,161,36,139,219,66,182,200,210,248,84,56,205,81,17,133],"panID":6754,"extendedPanID":[195,141,244,52,209,200,195,139],"channelList":[15]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB.CC2531-01"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB.CC2531-01 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,80,159,1,1,0,0,0,109,192,55,0,0,7,8,22,10,0,0,16,1,80,159,29,93] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,80,159,1,1,0,0,0,109,192,55,0,0,7,8,22,10,0,0,16,1,80,159,29,93] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,80,159,1,1,0,0,0,109,192,55,0,0,7,8,22,10,0,0,16,1,80,159,29] - 93 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40784,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3653741,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,22,10,0,0,16,1]}} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +463ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,7,28] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,7,28] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,7] - 28 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1913} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +6ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,2,2,7,2,217,20,52,1,2,0,0,0,0] - 144 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425} +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +7ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +7ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +0ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
^\Quit (core dumped)
```

Koenkk, 2020-08-17:
Looks like a bug in the z-stack firmware, as can be seen the returned extended pan id is correct:

```
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":8,"value":{"type":"Buffer","data":[195,141,244,52,209,200,195,139]}} +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'extendedPanID' is valid +24ms
```

But the returned value is:

```
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extNwkInfo - {"shortaddr":0,"devstate":9,"panid":6754,"parentaddr":0,"extendedpanid":"0x00124b001cd48a18","parentextaddr":"0x0000000000000000","channel":15} +9ms
```

I'm wondering what extendedpanid is actually used, can you try to sniff this?

Stoatwblr, 2020-08-20:
I can try if you give instructions. Right now I have no extra zigbee equipment onhand though (extra USB stuff is in transit)

Stoatwblr, 2020-08-20:
as a FWIW, I believe the epanid on air is the returned one, as I had to repair all the equipment to make things work

Stoatwblr, 2020-08-25:
New device has shown up (CC2531 and CC2540 both with external antennas)

falmanna, 2020-09-21:
Hi @Koenkk , I am experiencing the same issue with `CC2538_CC2592_MODKAMRU_V3_20200211`
My coordinator `expanid` and `channel` are getting rest to `00124B0018F1FC4A` and `Channel 11` every time I try to change them.

falmanna, 2020-10-23:
unstale

falmanna, 2020-11-24:
Unstale, issue still present

Koenkk, 2020-11-24:
@falmanna I understand the problem but I don't think it's very likely that this problem will be fixed because zstack 3.0.x is not developed by TI anymore

falmanna, 2020-11-24:
@Koenkk I am not sure if it's present in the default 3.0.x zstack as well or only on this version. 
If it's in the default then I agree it doesn't make since to track it unless someone has the skills and willing to fix it.

Koenkk, 2020-11-24:
@falmanna given that it happens on both CC2531 and CC2538 I think it is. Looking at the patches of these firmwares I don't see anything what could possibly cause this issue: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/firmware_CC2538_CC2592_MODKAMRU_V3.patch https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/firmware_CC2531_CC2530.patch

falmanna, 2020-11-24:
@Koenkk I agree since all changes are only for the configs part, but we can't be sure until we test the original. 
I think we can try to consult TI forums for a help and patch it here if it's possible.

closed 2021-01-01 by github-actions[bot]