Koenkk/zigbee2mqtt #4087: Wrong identification of Lonsonho x701 1 gang switch with N line

vbudennyi, 2020-08-12:
# Bug Report
## What happened
I bought 1 gang switches from Lonsonho:
https://aliexpress.com/item/4000298926256.html?spm=a2g0s.9042311.0.0.64fb5c0fmcx845
I successfully connected them to zigbee2mqtt
{code}
2020-08-08.15-07-14/log.txt:info  2020-08-08 15:20:10: Device '0xbc33acfffe655dbb' is supported, identified as: TuYa Valve control (TS0001)
2020-08-08.15-07-14/log.txt:info  2020-08-09 13:17:01: Device '0xbc33acfffe655ca5' is supported, identified as: TuYa Valve control (TS0001)
{code}
They have identified as TuYa Valve control (TS0001)
DB entries:
{code}
{"id":9,"type":"Router","ieeeAddr":"0xbc33acfffe655dbb","nwkAddr":6787,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1596892810580}
{"id":10,"type":"Router","ieeeAddr":"0xbc33acfffe655ca5","nwkAddr":39767,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1596971821581}
{code}

## What did you expect to happen
I expect that it should identify as x701 switch instead of valve control similar to 
https://www.zigbee2mqtt.io/devices/TS0002.html
And if this switch has N line is it possible to get info about power management? As I understand Xioami switches with N line has such option. 
Thanks in advance

## How to reproduce it (minimal and precise)
Connect such switch to zigbee2mqtt and validate logs

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version: zStack12

Koenkk, 2020-08-12:
Fixed!

> And if this switch has N line is it possible to get info about power management? As I understand Xioami switches with N line has such option.

You should check with the original gateway if the device supports this.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

vbudennyi, 2020-08-13:
Hi @Koenkk 
I have tested using edge version **1.14.3-dev**
and no success
after pairing 1st switch i got:
```log
2020-08-13.12-11-15/log.txt:info  2020-08-13 12:19:37: Successfully interviewed '0xbc33acfffe655dbb', device has successfully been paired
2020-08-13.12-11-15/log.txt:info  2020-08-13 12:19:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe655dbb","supported":false}}'
```
and it's not supported and i don't see it in the list
DB record:
```json
{"id":10,"type":"Router","ieeeAddr":"0xbc33acfffe655dbb","nwkAddr":50776,"manufId":0,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{},"binds":[]}},"interviewCompleted":true,"meta":{},"lastSeen":1597313756982}
```
Second switch has paired without problem but model remains the same
```log
2020-08-13.11-18-22/log.txt:info  2020-08-13 11:20:49: Starting interview of '0xbc33acfffe655ca5'
2020-08-13.11-18-22/log.txt:info  2020-08-13 11:20:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xbc33acfffe655ca5"}}'
2020-08-13.11-18-22/log.txt:info  2020-08-13 11:20:50: Successfully interviewed '0xbc33acfffe655ca5', device has successfully been paired
2020-08-13.11-18-22/log.txt:info  2020-08-13 11:20:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe655ca5","model":"TS0001","vendor":"TuYa","description":"Valve control","supported":true}}'

```
DB record:
```json
{"id":9,"type":"Router","ieeeAddr":"0xbc33acfffe655ca5","nwkAddr":23188,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597310450913}
```

After switching on **1.14.3 release** branch first switch paired successfully
logs after success:
```log
020-08-13.12-05-16/log.txt:info  2020-08-13 12:08:41: Starting interview of '0xbc33acfffe655dbb'
2020-08-13.12-05-16/log.txt:info  2020-08-13 12:08:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xbc33acfffe655dbb"}}'
2020-08-13.12-05-16/log.txt:info  2020-08-13 12:08:43: Successfully interviewed '0xbc33acfffe655dbb', device has successfully been paired
2020-08-13.12-05-16/log.txt:info  2020-08-13 12:08:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe655dbb","model":"TS0001","vendor":"TuYa","description":"Valve control","supported":true}}'
```
DB record:
```json
{"id":10,"type":"Router","ieeeAddr":"0xbc33acfffe655dbb","nwkAddr":883,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597314713907}
```

Koenkk, 2020-08-13:
Forgot to update, sorry.

Changes will be in tomorrow.

wiktorvip, 2020-09-09:
I'm using version: 1.14.4 and switches X701 are till showing as TuYa Valve control (TS0001)
[image]

Koenkk, 2020-09-09:
@wiktorvip please share your `data/database.db`

wiktorvip, 2020-09-09:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/5195650/database.zip)

vbudennyi, 2020-09-09:
Hi @Koenkk 
I'm also have the same situation after upgrade to 1.14.4
my db records:

{"id":17,"type":"Router","ieeeAddr":"0xbc33acfffe655e75","nwkAddr":3697,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1599232254280}

{"id":12,"type":"Router","ieeeAddr":"0xbc33acfffe655caf","nwkAddr":60067,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1598547950558}

{"id":10,"type":"Router","ieeeAddr":"0xbc33acfffe655dbb","nwkAddr":8921,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1598547953162}

{"id":9,"type":"Router","ieeeAddr":"0xbc33acfffe655ca5","nwkAddr":23188,"manufId":0,"manufName":"_TYZB01_seqwasot","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TYZB01_seqwasot","powerSource":1,"zclVersion":3,"appVersion":66}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[]}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1598548138827}

Koenkk, 2020-09-09:
@vbudennyi @wiktorvip when starting zigbee2mqtt, how is the device listed? It should be listed as `TuYa 1 gang switch`. I've also updated the picture

wiktorvip, 2020-09-09:
@Koenkk it is listed as` TuYa 1 gang switch (Router)` as you wrote
`Zigbee2MQTT:info  2020-09-09 20:48:30: Light_Hallway (0xbc33acfffe655d91): TS0001 - TuYa 1 gang switch (Router)`

closed 2020-10-17 by github-actions[bot]