Koenkk/zigbee2mqtt #4088: Lonsonho X712A works wrong

Machaeon, 2020-08-12:
Changing state from zigbee2mqtt not always triggers switch. E.g. 
Z2m: off, Switch: off
Z2m: on, Switch off
Z2m: off, Switch on

cheshic, 2020-08-29:
Same problem.

rkiller2006, 2020-09-02:
I have 3 switches X711A, which always toggle instead of turn_on or turn_off. Who knows how to solve it?

Machaeon, 2020-10-08:
There some difference betweet Lonsonho and other devices log.
X712A - Message repeats 4 times:
INFO: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"linkquality":99,"state_l1":"ON","state_l2":"OFF"}'
INFO: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"linkquality":97,"state_l1":"ON","state_l2":"OFF"}'
INFO: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"linkquality":97,"state_l1":"ON","state_l2":"OFF"}'
INFO: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"linkquality":97,"state_l1":"ON","state_l2":"OFF"}'

eWeLink SA-003-Zigbee - message repets 2 times:
INFO: MQTT publish: topic 'zigbee/Amp Switch', payload '{"linkquality":162,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee/Amp Switch', payload '{"linkquality":162,"state":"ON"}'

rkiller2006, 2020-10-08:
Hello Machaeon:
MY X711A switches send 3 times:
INFO: MQTT publish: topic 'zigbee2mqtt/Hallway light', payload '{"linkquality":73,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Hallway light', payload '{"linkquality":44,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Hallway light', payload '{"linkquality":44,"state":"OFF"}'

Sometimes with different messages:
INFO: MQTT publish: topic 'zigbee2mqtt/Bathroom ventilation', payload '{"linkquality":44,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Bathroom ventilation', payload '{"linkquality":44,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Bathroom ventilation', payload '{"linkquality":44,"state":"ON"}'

But always the second message is correct state:

Machaeon, 2020-10-09:
In my case second and third message shows right state.
Sometimes there only one message - in that cases switch doesn't reacts on command.

Machaeon, 2020-10-13:
It seems that z2m uses last message for state detection. Is it posible to make it to use another (e.g. second) message?

mgrom, 2020-11-08:
@Machaeon @rkiller2006  does this switch work good or not? Does it behave as router or end-device? Thanks for answers :)

Pete118, 2020-11-26:
> @Machaeon @rkiller2006 does this switch work good or not? Does it behave as router or end-device? Thanks for answers :)

I would appreciate this information,too :)

rkiller2006, 2020-11-27:
Hello, the switches work wrong. It's is router devices.

UndeRus, 2020-12-17:
Same problem with X713A

mgrom, 2020-12-17:
@rkiller2006 @UndeRus could you please be more specific, please? It'd be great to know what are possible problems with those devices.

UndeRus, 2020-12-17:
@mgrom 
when i try to set state "on" or "off" on X713A it behaves like toggle, but it also works not every call

UndeRus, 2020-12-18:
Today i tried to send into mosquitto 
```
mosquitto_pub -h localhost -t zigbee2mqtt/cabinet_3_wall_switch/set -m '{"state_l1": "ON", "state_l2": "OFF", "state_l3": "OFF"}'
```
It works like toggle, not like turn on the switch

closed 2021-02-27 by github-actions[bot]

densavage, 2021-04-16:
Same problem with X711A

Machaeon, 2022-03-30:
Still no news?