Koenkk/zigbee2mqtt #4090: Errors after updating to 1.14.3

janjbrand, 2020-08-12:
# Bug Report
## What happened

After upgrading to 1.14.3, I see errors in the logs and de link with Home Assistant is broken.

I see the following errors:

`zigbee2mqtt      | Zigbee2MQTT:error 2020-08-12 18:27:45: Publish 'set' 'state' to '0x000d6f001374797a' failed: 'Error: Command 0x000d6f001374797a/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'
zigbee2mqtt      | Zigbee2MQTT:info  2020-08-12 18:27:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6f001374797a' failed: 'Error: Command 0x000d6f001374797a/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x000d6f001374797a"}}'`

`zigbee2mqtt      | Zigbee2MQTT:error 2020-08-12 18:47:56: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: CommandResponse 0x000d6f0010677141/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))
zigbee2mqtt      |     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:657:27)
zigbee2mqtt      |     at Generator.next (<anonymous>)
zigbee2mqtt      |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58))`

## How to reproduce it (minimal and precise)
Brought down the Docker containers, removed then and brought them up; the same errors appeared. Going back to 1.14.2. solves the problem.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-08-13:
Could you provide the herdsman debug logging when controlling a device on both 1.14.2 and 1.14.3?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

GuGu927, 2020-08-15:
Here's same error.
Error from Xiaomi Plug.

Zigbee2MQTT:error 2020-08-15 21:03:50: Publish 'set' 'state' to 'plug' failed: 'Error: Command 0x00158d0001828978/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 38527 - 1 - 35 - 6 - 11 after 10000ms)'

ChessSpider, 2020-08-15:
Just wanted to comment I also experienced problems on 1.14.3. I reverted manually to 1.14.2 and problems have not returned yet.

SVhuMMer, 2020-08-16:
Have issues with 1.14.3 as well with Aqara wireless switches. Pls advise how to manually revert to 14.2?

janjbrand, 2020-08-19:
I tried to reproduce the errors, but after downgrading to 1.14.2 and upgrading to 1.14.3 I do not have errors any more. I repositioned the antenna, so maybe that solved the problem. Does an upgrade to 1.14.3 rebuild the network?

yevgenb, 2020-08-27:
Same error with sonoff plug:
Zigbee2MQTT:info  2020-08-27 13:42:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b001e7200a0' failed: 'Error: Command 0x00124b001e7200a0/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x00124b001e7200a0"}}'

Koenkk, 2020-08-28:
@yevgenb this happens when there is a lot of interference, make sure your wifi doesn't interfere (https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html) and that your adapter is connected through a USB extension cable positioned 30cm + from your PC.

reinerho, 2020-08-29:
I'm not sure if I have the same problem. My zigbee2mqtt suddenly lost connection to all my devices (xiaomi and ikea). I tried to reconfigure but without success. I tried to setup a new zigbee2mqtt server then on several Raspberry Pi versions (1B, 3B , zero, 4) but it always ended with the same result: With "npm start" the program starts properly, but when trying to connect any device, it crashes with the following messages:
Zigbee2MQTT:error 2020-08-29 19:49:19: Adapter disconnected, stopping
Zigbee2MQTT:info  2020-08-29 19:49:19: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2020-08-29 19:49:19: Disconnecting from MQTT server
Zigbee2MQTT:error 2020-08-29 19:49:19: Failed to stop zigbee
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-08-29T17_49_19_208Z-debug.log

In the logfile I found no additional information.

Koenkk, 2020-08-30:
@reinerho could you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

reinerho, 2020-08-30:
[2020-08-30T11_03_06_638Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/5146697/2020-08-30T11_03_06_638Z-debug.log)
Here is my log. Hope it helps.

reinerho, 2020-08-30:
Obviously the problem is in line 12: "Failed to exec start script". But I can't see what is wrong, as the problem occurres on four different Raspberry Models. For installation I always followed https://www.zigbee2mqtt.io/.

Koenkk, 2020-08-30:
This is not the zigbee2mqtt logging. But before diving further, what adapter/firmware are you using? If using a CC2531 I would recommend switching to the source routing firmware first: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

xX-Nexus-Xx, 2020-08-30:
I have the similar issues since I upgraded to 1.14.3

- I was running the previous versions for a long time (over 1year) and had no issue at all.
- i did the FW upgrade for the  CC2652R as well
- ever since I'm getting the below issue on my zigbee lights
- I don't think its interference as I already have selected the channel for zigbee and Wifi so there is no conflict (as said I had no issue before the last update
- it looks like my battery powered devices are not impacted

**See below some error messages from my lights:**
` 

08:55:13 | Publish 'set' 'state' to 'Light.Hallway1' failed: 'Error: Command 0x086bd7fffe52b41b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))' to device Original message:
-- | --
08:44:39 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 43801 - 1 - 218 - 6 - 11 after 10000ms)' to device Original message:
08:44:19 | Publish 'set' 'state' to 'Light.Hallway1' failed: 'Error: Command 0x086bd7fffe52b41b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))' to device Original message:
08:39:28 | Publish 'set' 'state' to 'Light.Toilett' failed: 'Error: Command 0x90fd9ffffe34ed9f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))' to device Original message:
08:35:48 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))' to device Original message:
08:34:12 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 43801 - 1 - 210 - 6 - 11 after 10000ms)' to device Original message:
08:33:52 | Publish 'set' 'state' to 'Light.Hallway1' failed: 'Error: Command 0x086bd7fffe52b41b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))' to device Original message:
08:26:58 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 43801 - 1 - 206 - 6 - 11 after 10000ms)' to device Original message:
08:26:38 | Publish 'set' 'state' to 'Light.Hallway1' failed: 'Error: Command 0x086bd7fffe52b41b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))' to device Original message:

`

How can I downgrade to the previous version?  I would like to test if this is really release dependent

regards

reinerho, 2020-08-31:
I also hoped to get a hint how to downgrade the zigbee2mqtt. My system was without any errors before upgrading. Nevertheless I ordered the hardware to flash the firmware. I'll give it a try as soon as I got it.
Btw. where would I find the correct logfile? I added the "advanced: log_level: debug" to my configuration.yaml and the file I sent was the only I found. 
As you may see - I am not really an expert with zigbee2mqtt. I started some month ago and after everything was ok I never touched it until I had this update. Unfortunately I forgot to have a backup before.

Koenkk, 2020-08-31:
@xX-Nexus-Xx it depends on how you are running Zigbee2MQTT. If running bare metal and following https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#6-for-later-update-zigbee2mqtt-to-the-latest-version between `git pull` and `npm ci` execute e.g. `git checkout tags/1.14.1`

xX-Nexus-Xx, 2020-09-01:
@Koenkk Thanks, the downgrade did work but did not solve my above problems

I'm running Z-Stack_3.x.0 with CC26X2R1 on a RPI 3b and have now started zigbe2mqtt with the herdsman log activated

I only switched ON/OFF the lights and got below errors:

`

Time | Text
-- | --
10:48:15 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:47:37 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:47:10 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:47:08 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:46:52 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:46:50 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:46:39 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:45:55 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:28:40 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:28:32 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:28:24 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:28:15 | Publish 'set' 'state' to 'Light.Hallway2' failed: 'Error: Command 0x086bd7fffe5a8c05/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
10:26:08 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:
[... cut ...] e' (205))' to device Original message:
10:25:55 | Publish 'set' 'state' to 'Light.Study' failed: 'Error: Command 0x14b457fffe48a02f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'No network route' (205))' to device Original message:

I have as well the herdsman logfile. How can I share it with you ?

Koenkk, 2020-09-01:
@xX-Nexus-Xx 

- Does the bulb change it's state?
- This could be related to interference, is your adapter positioned away from the PI3 (30cm+)?
- Can you try to re plug power of the bulbs?

xX-Nexus-Xx, 2020-09-01:
@Koenkk the issue is solved ( I hope)
- i monitored the bulbs more closely and occasionally they did => I thought your interference theory was correct
- my adapter is more then 30cm from the PI away and Pi is in a shielded box (not the adapter)
- I used a WIFI analyzer to check all the Wifi Channels used by my devices and none overlay with my Zigbee channel 25
- I had a spare router (the cheap ones from Ikea) and added em to my network ==> **_This solved the problem!!_**
 
There were no other changes or settings required, just the router. Which is strange as I thought the bulbs do act as router too and should have been enough to cover my (small) area (max distance to router is 50m). Again, I did not had this issue before the FW and Zigbee release update!! .. it was running stable for months.

@Koenkk :  Thanks for your help and quick responses

xX-Nexus-Xx, 2020-09-03:
@Koenkk .. I celebrated too early .... after 1day without issues ... the issue is back!   I don't know if it is related but I did generate a network-map (zigbee2mqtt/bridge/networkmap) and shortly afterwards the issues were back.

I also found that on zigbee2mqtt start the below errors

`Sep 03 15:52:18 NodeRed systemd[1]: Stopped zigbee2mqtt.
Sep 03 15:52:24 NodeRed systemd[1]: Started zigbee2mqtt.
Sep 03 15:52:26 NodeRed npm[1235]: > zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
Sep 03 15:52:26 NodeRed npm[1235]: > node index.js
Sep 03 15:52:44 NodeRed npm[1235]: (node:1250) UnhandledPromiseRejectionWarning: Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms
Sep 03 15:52:44 NodeRed npm[1235]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
Sep 03 15:52:44 NodeRed npm[1235]:     at listOnTimeout (internal/timers.js:549:17)
Sep 03 15:52:44 NodeRed npm[1235]:     at processTimers (internal/timers.js:492:7)
Sep 03 15:52:44 NodeRed npm[1235]: (node:1250) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
Sep 03 15:52:44 NodeRed npm[1235]: (node:1250) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Sep 03 15:52:44 NodeRed npm[1235]: zigbee2mqtt:error 2020-09-03 15:52:44: Failed to setup reporting for '0x001788010356f861' - Error: Bind 0x001788010356f861/11 genOnOff from '0x00124b001caabb97/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Sep 03 15:52:44 NodeRed npm[1235]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
Sep 03 15:52:44 NodeRed npm[1235]:     at Generator.throw (<anonymous>)
Sep 03 15:52:44 NodeRed npm[1235]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Sep 03 15:52:54 NodeRed npm[1235]: zigbee2mqtt:error 2020-09-03 15:52:54: Failed to setup reporting for '0x0017880104eb4335' - Error: Bind 0x0017880104eb4335/11 genOnOff from '0x00124b001caabb97/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Sep 03 15:52:54 NodeRed npm[1235]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
Sep 03 15:52:54 NodeRed npm[1235]:     at Generator.throw (<anonymous>)
Sep 03 15:52:54 NodeRed npm[1235]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
`

When generating the network map, the below errors are listed
`Sep 03 15:56:16 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:56:16: Failed to execute LQI for 'Light.Toilett'
Sep 03 15:56:26 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:56:26: Failed to execute LQI for 'Light.Hallway2'
Sep 03 15:56:36 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:56:36: Failed to execute LQI for 'Light.Bed2'
Sep 03 15:56:46 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:56:46: Failed to execute LQI for 'Light.Bed1'
Sep 03 15:57:27 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:57:27: Failed to execute LQI for 'Light.S1'
Sep 03 15:57:37 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:57:37: Failed to execute LQI for 'Light.S2'
Sep 03 15:57:47 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:57:47: Failed to execute LQI for 'Light.Study'
Sep 03 15:57:57 NodeRed npm[1321]: zigbee2mqtt:error 2020-09-03 15:57:57: Failed to execute LQI for 'router1'`

oxivanisher, 2020-09-04:
> I also hoped to get a hint how to downgrade the zigbee2mqtt. My system was without any errors before upgrading. Nevertheless I ordered the hardware to flash the firmware. I'll give it a try as soon as I got it.
@reinerho :
You can flash it by only using a RPI: https://notenoughtech.com/home-automation/flashing-cc2531-without-cc-debugger/
Attention: if using a RPI4b, ensure wiringpi >= 2.52 http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/ or it will not work.

oxivanisher, 2020-09-04:
I tried also power-cycling them and repositioning of the antenna without success. The bulbs no longer respond to any commands, but they might still repeat messages for other devices since my sensors still report temps and such.

Here is a herdsman debug output:
`020-09-04T21:26:49.026Z zigbee-herdsman:controller:endpoint Command 0xccccccfffebcef80/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCod
e":null,"transactionSequenceNumber":null})
2020-09-04T21:26:49.027Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xccccccfffebcef80:22142/1 (0,0)
2020-09-04T21:26:49.030Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":22142,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":20,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,6,0]}}
2020-09-04T21:26:49.032Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,126,86,1,1,6,0,20,0,30,3,1,6,0,8]
2020-09-04T21:26:49.050Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-09-04T21:26:49.051Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-09-04T21:26:49.052Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-09-04T21:26:49.053Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-09-04T21:26:49.054Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-04T21:26:49.842Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,20,31]
2020-09-04T21:26:49.843Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,20,31]
2020-09-04T21:26:49.843Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,20] - 31
2020-09-04T21:26:49.844Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":20}
2020-09-04T21:26:49.845Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-04T21:26:49.850Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xccccccfffebcef80:22142,205,0)
2020-09-04T21:26:49.852Z zigbee-herdsman:adapter:zStack:adapter Request network address of '0xccccccfffebcef80'
2020-09-04T21:26:49.855Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nwkAddrReq - {"ieeeaddr":"0xccccccfffebcef80","reqtype":0,"startindex":0}
2020-09-04T21:26:49.857Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,10,37,0,128,239,188,254,255,204,204,204,0,0,49]
2020-09-04T21:26:49.873Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,0,0,100]
2020-09-04T21:26:49.874Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,0,0,100]
2020-09-04T21:26:49.875Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 0 - [0] - 100
2020-09-04T21:26:49.875Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nwkAddrReq - {"status":0}
2020-09-04T21:26:49.877Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-04T21:26:59.886Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 22142
2020-09-04T21:26:59.888Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":22142,"options":0,"radius":30}
2020-09-04T21:26:59.890Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,126,86,0,30,82]
2020-09-04T21:26:59.905Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2020-09-04T21:26:59.906Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2020-09-04T21:26:59.907Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2020-09-04T21:26:59.908Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2020-09-04T21:26:59.910Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-04T21:27:02.917Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xccccccfffebcef80:22142/1 (0,1)
2020-09-04T21:27:02.920Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":22142,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":21,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,6,0]}}[0/9180]
2020-09-04T21:27:02.923Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,126,86,1,1,6,0,21,0,30,3,1,6,0,9]
2020-09-04T21:27:02.935Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-09-04T21:27:02.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-09-04T21:27:02.936Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-09-04T21:27:02.936Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-09-04T21:27:02.937Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-04T21:27:03.590Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,21,30]
2020-09-04T21:27:03.590Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,21,30]
2020-09-04T21:27:03.591Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,21] - 30
2020-09-04T21:27:03.592Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":21}
2020-09-04T21:27:03.592Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
Zigbee2MQTT:info  2020-09-04 23:27:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'TradfriBulp_workshop' failed: 'Error: Command 0xccccccfffebcef80/1 genOnOff.off({}, {\"timeout\":10000,\"disableR
esponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"
friendly_name":"TradfriBulp_workshop"},"type":"zigbee_publish_error"}'
`

oxivanisher, 2020-09-05:
So, I got it working again. But I can not tell, what exactly did the trick, or if the problem will return. I did the following things over several hours trial and error:

* Flashed back the "normal" firmware, not source routing. Maybe the Ikea Bulbs (repeaters themselves) are not able to join over another repeater?
* Removed the devices with Zigbee2MqttAssistant (checked "Use Force Remove mode.").
* And then started binding again.

With my non existent knowledge of Zigbee, my "findings" are something like this:
* Since the errors where 'No network route', somewhere during hacking on [other things](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1528) with zigbee2mqtt, they lost the network, or some key changed (although all other devices did not need to be reconnected).
* Re-binding did not work for a extended time during my testing, since i flashed the "source routing" firmware.
* Somewhere in zigbee2mqtt or zigbee2mqttassistant, something of the devices was leftover. I often hit "reset local state" in zigbee2mqttassistant, and even deleted the database, so it was probably in zigbee2mqtt.
* Only after the "old entries" of the devices in `configuration.yaml` where removed (done trough zigbee2mqttassistant), I was able to let the devices re-join. > Is the devices stored somewhere else with another identifier than the device id?

**This is all guess work, so please take this with a large grain of salt!**

GuGu927, 2020-09-05:
I have same problem with  type zigbee_publish_error
and this error solved after   "repairing"
When turn off router for some time(don't know how long), this error happen.
Is this mean mesh network broken? or something else?

xX-Nexus-Xx, 2020-09-07:
My system was not really able to "repair" the devices, it failed often. (As said: All was running fine before the upgrade)

I ended up deleting the database and start from scratch!

I'm now on latest version with FW and Zigbe2MQTT and all is working for the last few days, but I will keep a watching eye and not change anything for a while again.

neoc83, 2020-09-25:
Same problem here with HA Add-on after upgrading (not sure from what version it started but I'm now at 1.14.4.2). Some details that could help:
- Tradfri lights are reporting status initially at restart but no updates after that.
- Tradfri lights cannot be controlled.
- Osram smart plug is reporting status initially at restart but no updates after that (no update after using physical switch).
- Osram smart plug cannot be controlled.
- Battery powered devices (Aqara sensors) are reporting status initially at restart but no updates after that.

I have not had any of these kind of issues with this setup before.

Tradfri light log:
```
Zigbee2MQTT:info  2020-09-25 10:15:23: MQTT publish: topic 'zigbee2mqtt/tradfri_basic_spot_3', payload '{"brightness":0,"linkquality":55,"state":"OFF","update_available":true}'
Zigbee2MQTT:error 2020-09-25 10:38:44: Publish 'set' 'state' to 'tradfri_basic_spot_3' failed: 'Error: Command 0x680ae2fffe14fccd/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":35382,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":14,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,11,1]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'
```

Osram smart plug log:
```
Zigbee2MQTT:info  2020-09-25 10:15:22: MQTT publish: topic 'zigbee2mqtt/osram_smart_plug_1', payload '{"linkquality":55,"state":"OFF","update_available":true}'
Zigbee2MQTT:error 2020-09-25 10:40:37: Publish 'set' 'state' to 'osram_smart_plug_1' failed: 'Error: Command 0x7cb03eaa0a025ecd/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no resources' (26))'
```

Aqara sensor log:
```
Zigbee2MQTT:info  2020-09-25 10:15:22: MQTT publish: topic 'zigbee2mqtt/aqara_temperature_sensor_2', payload '{"battery":86,"humidity":51.06,"linkquality":21,"pressure":1001.1,"temperature":20.74,"voltage":2975}'
```

neoc83, 2020-09-29:
> Same problem here with HA Add-on after upgrading (not sure from what version it started but I'm now at 1.14.4.2). Some details that could help:

As suggested by many, I started looking at optimizing the Zigbee connection by using a USB extension cable for the CC2531. Although nothing changed in my home setup, it worked out well and all devices are connected and communicating as before. Perhaps a new wifi network in the neighborhood was causing interference?

closed 2020-11-06 by github-actions[bot]