Koenkk/zigbee2mqtt #4092: Socket ZNCZ02LM is not sending voltage

Gooman-rus, 2020-08-13:
# Bug Report
## What happened
I have updated to 1.14.3 and socket ZNCZ02LM is not sending the voltage.
## What did you expect to happen
I expecting that ZNCZ02LM will send the voltage as it said in the improvements section of the 1.14.3 release.

## How to reproduce it (minimal and precise)
Update to 1.14.3 and check the voltage field of any ZNCZ02LM socket.
[image]

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2538
Adapter firmware version: 2020.02.11

IronButterfly, 2020-08-13:
The same story. Voltage doesn't work.

SAOPP, 2020-08-13:
Same over here guys.

Koenkk, 2020-08-13:
@RodriBus  when is the voltage updated in your case, is it updated constantly?

RodriBus, 2020-08-14:
My devices are reporting.

I can see their values at the zigbee2mqtt log:
`Zigbee2MQTT:info  2020-08-14 07:28:26: MQTT publish: topic 'zigbee2mqtt/Mi power plug 1', payload '{"state":"OFF","power":0,"voltage":217.6,"consumption":76.93,"temperature":29,"linkquality":115}'`

And are carried over to their entities:
[image]

Also, the values don't update that often, but they do:
[image]

Is there any more info I can provide?

Koenkk, 2020-08-14:
For me the voltage is send directly after pairing: 

[image]

@Gooman-rus @IronButterfly @SAOPP can you try to remove the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove , pair and provide the debug logging of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Gooman-rus, 2020-08-14:
Maybe my sockets have too old firmwares (09-04-2018)?

```
Zigbee2MQTT:warn  2020-08-14 13:55:08: Device '0x00158d00039fa6af' left the network
Zigbee2MQTT:info  2020-08-14 13:55:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00158d00039fa6af"}}'
Zigbee2MQTT:info  2020-08-14 13:55:15: Device 'Socket_tablet' joined
Zigbee2MQTT:info  2020-08-14 13:55:15: Starting interview of 'Socket_tablet'
Zigbee2MQTT:info  2020-08-14 13:55:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Socket_tablet"}}'
Zigbee2MQTT:info  2020-08-14 13:55:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Socket_tablet"}}'
Zigbee2MQTT:debug 2020-08-14 13:55:15: Device 'Socket_tablet' announced itself
Zigbee2MQTT:info  2020-08-14 13:55:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Socket_tablet"}}'
Zigbee2MQTT:debug 2020-08-14 13:55:16: Received Zigbee message from 'Socket_tablet', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.plug","appVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:16: Received Zigbee message from 'Socket_tablet', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":25,"5":34,"7":"0x0000000000000000","8":4898,"9":2048,"100":0,"149":4.977133274078369,"152":0,"154":16}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-14 13:55:17: MQTT publish: topic 'zigbee2mqtt/Socket_tablet', payload '{"state":"OFF","power":0,"voltage":null,"consumption":4.98,"temperature":25,"linkquality":39}'
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"hwVersion":18}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:18: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{"dateCode":"09-04-2018"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-14 13:55:18: Received Zigbee message from 'Socket_tablet', type 'attributeReport', cluster 'genOnOff', data '{"61440":50331392,"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-14 13:55:18: MQTT publish: topic 'zigbee2mqtt/Socket_tablet', payload '{"state":"OFF","power":0,"voltage":null,"consumption":4.98,"temperature":25,"linkquality":94}'
Zigbee2MQTT:debug 2020-08-14 13:55:18: Received Zigbee message from 'Socket_tablet', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-14 13:55:18: Successfully interviewed 'Socket_tablet', device has successfully been paired
Zigbee2MQTT:info  2020-08-14 13:55:18: Device 'Socket_tablet' is supported, identified as: Xiaomi Mi power plug ZigBee (ZNCZ02LM)
Zigbee2MQTT:info  2020-08-14 13:55:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Socket_tablet","model":"ZNCZ02LM","vendor":"Xiaomi","description":"Mi power plug ZigBee","supported":true}}'
Zigbee2MQTT:debug 2020-08-14 13:55:18: Received Zigbee message from 'Socket_tablet', type 'attributeReport', cluster 'genOnOff', data '{"61440":50331392,"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-14 13:55:18: MQTT publish: topic 'zigbee2mqtt/Socket_tablet', payload '{"state":"OFF","power":0,"voltage":null,"consumption":4.98,"temperature":25,"linkquality":84}'
Zigbee2MQTT:debug 2020-08-14 13:55:21: Received Zigbee message from 'Socket_tablet', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
```

Koenkk, 2020-08-14:
The plug indeed doesn't report the voltage, it should be in the following message in the `150` property which is missing:

```
Zigbee2MQTT:debug 2020-08-14 13:55:17: Received Zigbee message from 'Socket_tablet', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":25,"5":34,"7":"0x0000000000000000","8":4898,"9":2048,"100":0,"149":4.977133274078369,"152":0,"154":16}}' from endpoint 1 with groupID 0
```

I think this has been removed in later revisions of the plug, mine has `"dateCode":"02-28-2017"` (this can be checked in `data/databse.db`). @RodriBus what do you see here?

RodriBus, 2020-08-14:
@Koenkk Checking the file I can see that, indeed, my plugs have `"dateCode":"02-28-2017"`.

poisondima, 2020-08-14:
I have z2m v1.14.3-dev, but variable "voltage" is null
[image]
[image]

to4ko, 2020-08-15:
using in total 29 lumi plugs with datecodes 2017 and 2018 - no voltage reported for any single one of them

Koenkk, 2020-08-17:
@to4ko what is the exact `dateCode`?

to4ko, 2020-08-17:
> @to4ko what is the exact `dateCode`?

attached
[lumi_plug.txt](https://github.com/Koenkk/zigbee2mqtt/files/5084593/lumi_plug.txt)

Koenkk, 2020-08-17:
@to4ko I don't see a dateCode `02-28-2017`, so probably only those support it. 

I've fixed the `voltage: null` values (required manual removal from `data/state.json` first and added a note to the docs (https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html#voltage).

Assuming this can be closed since we cannot fix this.

closed 2020-08-17 by Koenkk

jusxon, 2021-12-05:
Would this also apply to the US version of this plug (ZNCZ12LM)?

I am also having this issue on a ZNCZ12LM with dateCode of "05-02-2018".

Creo2005, 2022-01-09:
Voltage is present:
[image]
But in Domoticz - Devices not present!