Koenkk/zigbee2mqtt #4094: CC2652RB problem in Combination with Zwave-Binding, OpenHAb, RPi4b+ (4Gb)

bifi2090, 2020-08-13:
# Bug Report
## What happened

CC2652RB (https://slae.sh/projects/cc2652) quits its service after some seconds (blue LED off, Debug Firmware CC2652RB) when runningzigbee2mqtt (bare-metal as well as docker) when it runs parallel to the ZWave-Binding on RPi4 b+ (4Gb) within openHAB. If the Zwave-Binding is stopped, the CC2652RB works as expected (blue LED remains on).

## What did you expect to happen

Use the CC2652RB with zigbee2mqtt (bare-metal or docker, no matter) in parallel use with the ZWave-Binding. 

## How to reproduce it (minimal and precise)

Use an Aeotec-Gen5 Zwave-Controller (USB) and the CC2652RB-z2m-Stick on a Raspberry Pi 4b+ (4GB) using the ZWave-Binding and zigbee2mqtt. 

## Debug Info
Zigbee2MQTT version: 1.14.2 
Adapter hardware: CC2652RB
Adapter firmware version: http://slae.sh/downloads/znp_CC2652RB_20200715.bin (debug, also tested with newest firmware CC2652RB_20200805.zip)

Actually I use the CC2531-Coordinator with z2m bare-metal and the Aeotec-Zstick with the Zwave-Binding on the above mentioned machine. The System runs completely stable, except for a few connection problems with the CC2531, which is why I want to switch to the CC2652RB. Apparently i thought the Zwave-Binding blocks the /dev/ttyUSB* port to which the stick registers. I've set up a test environment for the CC2652RB with z2m in Docker. When connected to the Mqtt_Broker, the blue Led leaves on for some seconds (that is the length of time in which I can pair devices and get some Data of them) an turns subsequently off. At this point I can´t pair devices anymore, or get some Data of them.

An associated ongoing discussion and several logs (hoping not to post all again here :) ) regarding the z2m can be understood/viewed in the OpenHAB-Community Forum:

[https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245](url)

I opened this issue here, to check whether there is problem with zigbee2mqtt instead of the Zwave-Binding mentioned in the OpenHab-Forum.

I hope someone can help me
best regards

DrTron, 2020-08-14:
Now that's fascinating. I received my CC2652RB from slaesh today and tried to get it working. No dice, really strange error messages, until I stumbled on this issue. What a coincidence.
Guess what I use within openHAB? Correct, a Aeotec Gen5 zstick with the zwave binding. And as with you, zigbee2mqtt works flawlessly when I disable that.

I'm sorry that I don't have a solution, but maybe you get some solace in that you're not the only one. I'd expect that the CC2652RB are not all that common, but the Aeotec sticks sure are.

bifi2090, 2020-08-14:
> Now that's fascinating. I received my CC2652RB from slaesh today and tried to get it working. No dice, really strange error messages, until I stumbled on this issue. What a coincidence.
> Guess what I [...]

ok that calms me down. So there must really be a problem in Combination with the Zwave-Binding. 
The Aeotec Stick is known to be used with a HUB. So too with me. But I can leave it stuck. Merely terminating the Zwave binding leads to success. Hope that someone can help us :)

slaesh, 2020-08-14:
I guess the problem is, that the z-wave software is accessing the same serial-port for any unknown reason?!

As soon as I am home, I will investigate this!

bifi2090, 2020-08-14:
So below is the Full Debug-log (docker-run parameter -e DEBUG=zigbee-herdsman*) when starting z2m with CC2652RB. 
I forced to stop ("STRG + C") the Container at about 09:09:41 :

https://pastebin.com/abQ8qUj5

It is also mentioned in the openHab-Forum 
(https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245)

DrTron, 2020-08-14:
Yes, I found the openhab-thread as well. We might want to move the discussion there as zigbee2mqtt clearly is not at fault here.

bifi2090, 2020-08-18:
I close this issue now, because the problem depends definitely from the Zwave-Binding within OpenHAB. 
For everybody who's interested: remove the actuall Zwave-Binding (2.5.7, 23.07.2020) and install Zwave-Binding 2.5.0 (org.openhab.binding.zwave-2.5.0.jar) manually via Addons-Folder. That´s the actuall solution, because since Zwave 2.5.1 there was an serial reconnection workaround implemented. If the workaround get´s removed at actuall bindings, then the problem is away.

bifi2090, 2020-08-19:
CC2652RB is now working with z2m-bare-metal ind parallel with the actual Zwave-Binding on Raspi 4 OpenHAB.

Solution:

https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245/63?u=bifi2090

closed 2020-08-19 by bifi2090