Koenkk/zigbee2mqtt #4095: Unable to start Zigbee2MQTT

Th0mC, 2020-08-13:
# Bug Report
## What happened
Unable to start Zigbee2MQTT
## What did you expect to happen
## How to reproduce it (minimal and precise)
I tried many things but each time I try to launch zigbee2MQTT it fails.
I verified all topics from the FAQ (port, port access :
- port AMA0. the only thing is that I can't launch the 'ls -l /dev/serial/by-id' command
- access to the port ok
- no voltage issue
- no USB issue
- I uninstalled modemmanager on my raspberry pi 4
- When I plug my CC3531 stick, the green light is on. So I pushed the reset button
- I tried to reflash my CC2531 stick with the last routing firmware CC2531_DEFAULT_20190608.zip --> CC2531ZNP-Prod.hex

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608.zip --> CC2531ZNP-Prod.hex

## Logs
https://pastebin.com/e4nWk0i7
## Config File
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyAMA0

Koenkk, 2020-08-13:
Instead of `/dev/ttyAMA0` can you try `/dev/ttyACM0`?

Th0mC, 2020-08-13:
> Instead of `/dev/ttyAMA0` can you try `/dev/ttyACM0`?

Yes I already tried that. But checking 'dmesg | grep "tty" ' I see that ttyACM0 now appears. Maybe since I reflashed the stick...
And Tadam! now it works!

If it can help someone in the future. Here is the output of 'dmesg | grep "tty"' after reflashing:

> pi@raspberrypi:/opt/zigbee2mqtt $ dmesg | grep "tty"
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 cma=256M  smsc95xx.macaddr=DC:A6:32:2D:2F:3F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 consoletty1 root=PARTUUID=37b74fa4-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000721] console [tty1] enabled
[    0.435298] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 34, base_baud = 0) is a PL011 rev2
[19216.825132] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

Before, only ttyAMA0 was appearing

closed 2020-08-13 by Th0mC