Koenkk/zigbee2mqtt #4096: Advice on how to use with home assistant and naming things

hvanoch, 2020-08-13:
Hi,

Just starting with integrating the zigbee2mqtt in my home assistant setup.
Everything in home assistant has to be referenced with an enity_id. 

When using home assistant and zigbee2mqtt with discovery all entities get a name like `binary_sensor.0x00158d00053f84dc_contact`

What are the possible ways to change this entity ID. Is this only through the UI or is there also a way to do it through some configuration (either in the home assistant or in the zigbee2mqtt)? That way I can put it my source control. It would also be a lot easier to search.

What would you advice?

Koenkk, 2020-08-13:
The `0x00158d00053f84dc` part of `binary_sensor.0x00158d00053f84dc_contact` is taken from the `friendly_name` in the Zigbee2MQTT `configuration.yaml`. However this is only set once! When the `friendly_name` is updated and Zigbee2mqtt resend the discovery payload, Home Assistant does not update the entity_id.

So the options are:
- Change via UI in Home Assistant
- Prevent Home Assistant from discovering the id containing the "wrong" `friendly_name` by either:
  - Stopping Home Assistant when pairing new devices
  - Stop Zigbee2MQTT, set `homeassistant: false` in Zigbee2MQTT `configuration.yaml`, start, pair device, stop and set `homeassistant: true` again.

hvanoch, 2020-08-13:
Thank you for the clear explanation. Guess it makes sense it doesn't update the entity_id by itself.

If I get annoyed by this then I guess I might make a script that changes the entity_id in the `config/.storage/core.entity_registry` with some regex whenever I do a rename of a zigbee device through ha.

felixstorm, 2020-08-14:
I was trying to find an easy way to also change HA's entity_id for some time as well and reading your issue, I think I found another option: If you actively delete all of the device's entities from Home Assistant before you rename, HA will accept the new name also as the entity_id (and not the friendly name only).

Deletion should be done via MQTT also (publish an empty payload to all config topics) and I found a relatively easy way to do this [here](https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466/2) (picture taken from there also):
[image]

So the complete procedure would be:
1. Pair the new device with zigbee2mqtt.
2. Remove *all* of the device's HA entities (e.g. `homeassistant/sensor/<friendly_name>/action/config`, `homeassistant/sensor/<friendly_name>/battery/config`, `homeassistant/sensor/<friendly_name>/linkquality/config`, `homeassistant/binary_sensor/<friendly_name>/update_available/config` etc. - you can get this information from the z2m log right after pairing) using MQTT as shown above.
3. Rename the device in zigbee2mqtt.
4. HA should see the device again as new and will also pickup the new name as the entity_id.

@Koenkk Maybe z2m could delete the entities from HA first, wait a few seconds and then publish the new name when renaming a device? This way it could work seamlessly at least as long as Home Assistant is running while renaming...

Koenkk, 2020-08-17:
@felixstorm good idea! implemented

to4ko, 2020-09-02:
> @felixstorm good idea! implemented

can we have some kind of a switch on that? entity_id and friendly name little bit different and this kind of "enhancements" can 
 and would ruin all the **existing** logic in HA

depen84, 2020-09-02:
> can we have some kind of a switch on that?
+1
I like my 0x entities!

Spirituss, 2020-09-02:
@Koenkk Technically, recreation of ha entities is a possible solution. But one should have choices either to recreate entities automatically or do it manually (current deconz logic, for example), since automatic deletion of entities could be very damaging for existing legacy in the ha, such as automations, lovelace, etc.

septillion-git, 2020-09-03:
What's the status on this? Does it only change entity id when I change the friendly name? Or will it also change entity id of devices I already renamed in the past?

Because although I hate the 0x entity id's and manually renamed most, the key is most... I would hate it even more if current automations en cards stop working with this update... 🙄

septillion-git, 2020-09-04:
And if you rename through the devices.yaml? (Which I used to do...)

Koenkk, 2020-09-04:
That is the same as: 
> When renaming to configuration.yaml (stop/start z2m) -> no rediscover, was already the case

septillion-git, 2020-09-04:
Thanks!

Good and a shame at the same time. 😛 So what would be the easiest way to rename the friendly names and entities? Or should I just wait for something like Z2M Assistant to support it?

Koenkk, 2020-09-04:
Yes, or the recently integrated frontend (https://github.com/Koenkk/zigbee2mqtt/issues/4266) (`homeassistant_rename` option will be added soon there).

septillion-git, 2020-09-04:
Sounds great! Thanks!

to4ko, 2020-09-04:
> I've updated the behaviour:
> 
> * When renaming to `configuration.yaml` (stop/start z2m) -> no rediscover, was already the case
> * When renaming through the old api (`zigbee2mqtt/bridge/config/rename`) -> no rediscover anymore
> * When renaming through the new api (#3281) (`zigbee2mqtt/bridge/request/device/rename`) -> by default no rediscover anymore but it can be enabled by publishing: `{"from": "oldname", "to": "newname", "homeassistant_rename": true}`

thanks a lot, Koen! will wait for the next update to install.

closed 2020-10-12 by github-actions[bot]

MDSDM, 2021-01-27:
This works almost...

_Living Room - door sensor_ in ZigBee2MQTT

Change name in Home Assistant to _Door Sensor_

Then I change in ZigBee2MQTT to _Sensor Door_ and disable change in Home Assistant

I get link quality, contact and battery to change their names.

RezzZ, 2023-05-15:
any chance we can have a setting to auto tick the "Update Home Assistant entity ID" option when renaming devices?
I'm migrating from deconz to z2m and It's pretty anoying that I have to check this option every time I rename a devices friendly name :).
Otherwise I'm very happy with Z2M so far 😃